MichaIng/DietPi #3166: Unable to locate package grub-efi-amd64grub-efi-amd64-signed

timocapa, 2019-10-17:
Converting Debian 9.9.0 (x86_64) to DietPi Buster (beta)

#### Steps to reproduce

1. Install Debian 9.9.0 (I used netinstall without any packages) in a VM
2. Run PREP, choose Beta and Buster
3. Wait

#### Expected behaviour

- Installs fine

#### Actual behaviour

- 
[image]

#### Extra details

- This is a VM from FreeNAS, which uses bhyve as hypervisor.
- I believe this isn't a system issue, but a source or script issue.

MichaIng, 2019-10-17:
@timocapa 
Jep indeed, I just fixed it: https://github.com/MichaIng/DietPi/commit/5bd0d7735e34050e78b3448e5e532b058e477b9a

Many thanks  for reporting!

MichaIng, 2019-10-18:
I mark this as closed since solved for dev and master branch. Feel free to reopen if required.

closed 2019-10-18 by MichaIng