MichaIng/DietPi #3170: dietpi-PREP | wget: unable to resolve host address ‘github.com’ | can access and download master.zip

ralban, 2019-10-18:
#### Details:
- Date           | Wed 16 Oct 23:22:16 EDT 2019
- Bug report     |
- DietPi version | v6.26.3 (MichaIng/master)
- Image creator  | N/A
- Pre-image      | N/A
- SBC device     | NULL (index=20)
- Kernel version | Linux dietpihole 4.19.0-6-amd64 #1 SMP Debian 4.19.67-2+deb10u1 (2019-09-20) x86_64 GNU/Linux
- Distro         | buster (index=5)
- Command        | Connection test: https://github.com/MichaIng/DietPi/archive/master.zip
- Exit code      | 4
- Software title | DietPi-PREP

#### Steps to reproduce:

1. run dietpi-PREP script on fresh Debian 10.1 install

#### Expected behaviour:

- The script should execute and complete.

#### Actual behaviour:

- Script fails to resolve github.com, cannot download master.zip, and errors out.

#### Extra details:

- I am able to use wget to download master.zip outside of the dietpi-PREP scrip, see logs below.

#### Additional logs:
```
Spider mode enabled. Check if remote file exists.
--2019-10-16 23:22:00--  https://github.com/MichaIng/DietPi/archive/master.zip
Resolving github.com (github.com)... failed: Connection timed out.
wget: unable to resolve host address ‘github.com’
```

---------------------------------------------------------------------

[FAILED] DietPi-PREP | Unable to continue, DietPi-PREP will now terminate.

root@dietpihole:~# wget https://github.com/MichaIng/DietPi/archive/master.zip
--2019-10-17 21:41:14--  https://github.com/MichaIng/DietPi/archive/master.zip
Resolving github.com (github.com)... 192.30.253.112
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/MichaIng/DietPi/zip/master [following]
--2019-10-17 21:41:19--  https://codeload.github.com/MichaIng/DietPi/zip/master
Resolving codeload.github.com (codeload.github.com)... 140.82.113.9
Connecting to codeload.github.com (codeload.github.com)|140.82.113.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2696948 (2.6M) [application/zip]
Saving to: ‘master.zip.1’

master.zip.1                         100%[=====================================================================>]   2.57M  12.9MB/s    in 0.2s

MichaIng, 2019-10-18:
@ralban 
Many thanks for your report.

> Resolving github.com (github.com)... failed: Connection timed out.

Indeed your DNS resolving seems to take more then 5 seconds (slightly more only it seems), which is our default timeout. Note that usually its 0.5 seconds.

Please try the following before running the script:
```
export G_CHECK_URL_TIMEOUT=10
```
The same setting can be found in dietpi.txt and dietpi-config.

Also you might want to play with `/etc/resolv.conf` if another DNS works faster for you.
__________________
Task for us:
- When running into any timeout during URL check, allow to raise timeout and retry attempts directly from error prompt.

ralban, 2019-10-19:
Thanks, @MichaIng 
"export G_CHECK_URL_TIMEOUT=10" resolved the issue. You may close this Issue. Thanks!

MichaIng, 2019-11-13:
I'll reopen as of enhancement to allow raising timeout/attempts directly from error prompt.

MichaIng, 2020-03-20:
Closing in favour of: #3185

closed 2020-03-20 by MichaIng