MichaIng/DietPi #3173: DietPi-Software | Kodi: Segfaults and won't start

glogiotatidis, 2019-10-18:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.26.3
- Distro version | 10.1
- Kernel version | 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
- SBC device | RPi3 A+
- Power supply used | 5V 2A
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | Kodi
- Was the software title installed freshly or updated/migrated? Freshly installed
- Can this issue be replicated on a fresh installation of DietPi? Yes, used to be OK in 6.25.3

#### Steps to reproduce
1. Install Kodi via dietpi-software
2. `$ startkodi`

#### Expected behaviour

- Kodi should start

#### Actual behaviour

- Kodi segfaults

#### Extra details
Kodi crash log
```
############## Kodi CRASH LOG ###############

################ SYSTEM INFO ################
 Date: Fri 18 Oct 2019 02:11:54 PM EEST
 Kodi Options: 
 Arch: armv7l
 Kernel: Linux 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019
 Release: Raspbian GNU/Linux 10 (buster)
############## END SYSTEM INFO ##############

############### STACK TRACE #################
gdb not installed, can't get stack trace.
############# END STACK TRACE ###############

################# LOG FILE ##################

﻿2019-10-18 14:11:52.822 T:1899628016  NOTICE: -----------------------------------------------------------------------
2019-10-18 14:11:52.822 T:1899628016  NOTICE: Starting Kodi (18.4 Git:20190906-f5ea6a9b04). Platform: Linux ARM 32-bit
2019-10-18 14:11:52.822 T:1899628016  NOTICE: Using Release Kodi x32 build (version for Raspberry Pi)
2019-10-18 14:11:52.822 T:1899628016  NOTICE: Kodi compiled 2019-09-08 by GCC 8.3.0 for Linux ARM 32-bit version 4.18.20 (266772)
2019-10-18 14:11:52.822 T:1899628016  NOTICE: Running on Raspbian GNU/Linux 10 (buster), kernel: Linux ARM 32-bit version 4.19.75-v7+
2019-10-18 14:11:52.834 T:1899628016  NOTICE: FFmpeg version/source: 4.0.4-Kodi
2019-10-18 14:11:52.834 T:1899628016  NOTICE: Host CPU: ARMv7 Processor rev 4 (v7l), 4 cores available
2019-10-18 14:11:52.834 T:1899628016  NOTICE: ARM Features: Neon enabled
2019-10-18 14:11:52.834 T:1899628016  NOTICE: special://xbmc/ is mapped to: /usr/share/kodi
2019-10-18 14:11:52.834 T:1899628016  NOTICE: special://xbmcbin/ is mapped to: /usr/lib/arm-linux-gnueabihf/kodi
2019-10-18 14:11:52.834 T:1899628016  NOTICE: special://xbmcbinaddons/ is mapped to: /usr/lib/arm-linux-gnueabihf/kodi/addons
2019-10-18 14:11:52.834 T:1899628016  NOTICE: special://masterprofile/ is mapped to: /root/.kodi/userdata
2019-10-18 14:11:52.834 T:1899628016  NOTICE: special://envhome/ is mapped to: /root
2019-10-18 14:11:52.835 T:1899628016  NOTICE: special://home/ is mapped to: /root/.kodi
2019-10-18 14:11:52.835 T:1899628016  NOTICE: special://temp/ is mapped to: /root/.kodi/temp
2019-10-18 14:11:52.835 T:1899628016  NOTICE: special://logpath/ is mapped to: /root/.kodi/temp
2019-10-18 14:11:52.835 T:1899628016  NOTICE: The executable running is: /usr/lib/arm-linux-gnueabihf/kodi/kodi-rbpi_v7
2019-10-18 14:11:52.835 T:1899628016  NOTICE: Local hostname: flix
2019-10-18 14:11:52.835 T:1899628016  NOTICE: Log File is located: /root/.kodi/temp/kodi.log
2019-10-18 14:11:52.840 T:1899628016  NOTICE: -----------------------------------------------------------------------
2019-10-18 14:11:52.860 T:1899628016    INFO: loading settings
2019-10-18 14:11:52.860 T:1899628016  NOTICE: special://profile/ is mapped to: special://masterprofile/
2019-10-18 14:11:52.861 T:1899628016   ERROR: CSettings: unable to load settings from special://masterprofile/guisettings.xml, creating new default settings
2019-10-18 14:11:52.891 T:1899628016   DEBUG: CSkinSettings: no <skinsettings> tag found
[... 180 lines cut ...]
2019-10-18 14:11:53.544 T:1899628016  NOTICE: Found resolution 640 x 480 with 640 x 480 @ 75.000000 Hz
2019-10-18 14:11:53.544 T:1899628016  NOTICE: Found resolution 800 x 600 with 800 x 600 @ 56.000000 Hz
2019-10-18 14:11:53.544 T:1899628016  NOTICE: Found resolution 800 x 600 with 800 x 600 @ 60.000000 Hz
2019-10-18 14:11:53.544 T:1899628016  NOTICE: Found resolution 800 x 600 with 800 x 600 @ 59.940063 Hz
2019-10-18 14:11:53.544 T:1899628016  NOTICE: Found resolution 800 x 600 with 800 x 600 @ 75.000000 Hz
2019-10-18 14:11:53.544 T:1899628016  NOTICE: Found resolution 1024 x 768 with 1024 x 768 @ 60.000000 Hz
2019-10-18 14:11:53.544 T:1899628016  NOTICE: Found resolution 1024 x 768 with 1024 x 768 @ 59.940063 Hz
2019-10-18 14:11:53.544 T:1899628016  NOTICE: Found resolution 1024 x 768 with 1024 x 768 @ 75.000000 Hz
2019-10-18 14:11:53.544 T:1899628016  NOTICE: Found resolution 1152 x 864 with 1152 x 864 @ 75.000000 Hz
2019-10-18 14:11:53.544 T:1899628016  NOTICE: Found resolution 1280 x 960 with 1280 x 960 @ 60.000000 Hz
2019-10-18 14:11:53.544 T:1899628016  NOTICE: Found resolution 1280 x 960 with 1280 x 960 @ 59.940063 Hz
2019-10-18 14:11:53.544 T:1899628016  NOTICE: Found resolution 1280 x 1024 with 1280 x 1024 @ 60.000000 Hz
2019-10-18 14:11:53.544 T:1899628016  NOTICE: Found resolution 1280 x 1024 with 1280 x 1024 @ 59.940063 Hz
2019-10-18 14:11:53.544 T:1899628016  NOTICE: Found resolution 1280 x 1024 with 1280 x 1024 @ 75.000000 Hz
2019-10-18 14:11:53.544 T:1899628016  NOTICE: Found resolution 1440 x 900 with 1440 x 900 @ 60.000000 Hz
2019-10-18 14:11:53.544 T:1899628016  NOTICE: Found resolution 1440 x 900 with 1440 x 900 @ 59.940063 Hz
2019-10-18 14:11:53.545 T:1899628016  NOTICE: Found resolution 1680 x 1050 with 1680 x 1050 @ 60.000000 Hz
2019-10-18 14:11:53.545 T:1899628016  NOTICE: Found resolution 1680 x 1050 with 1680 x 1050 @ 59.940063 Hz
2019-10-18 14:11:53.545 T:1899628016  NOTICE: Found (1680x1050@60.000000) at 34, setting to RES_DESKTOP at 16

############### END LOG FILE ################

############ END Kodi CRASH LOG #############

```

MichaIng, 2019-10-18:
@glogiotatidis 
Many thanks for your report.

You as well used the RPi Buster image with v6.25, right? Or was it Stretch?
Since there was no real RPi-compatible Kodi release available before on RPi Buster.

However from the error log I am not 100% sure, which finally caused the crash, at least looks like it has some audio issues.
Do you use RPi onboard audio? HDMI or 3.5mm jack in case? Any error messages when you enable/chose one of those via dietpi-config?

glogiotatidis, 2019-10-18:
> You as well used the RPi Buster image with v6.25, right? Or was it Stretch?

Yes it's `10.1`. Actually I downloaded a fresh image a couple of days ago that worked (except for the polkit issue) and then today another fresh image that doesn't.

Same HW, same setup.

I don't have altered any audio settings. It defaults to `rpi-bcm2835-auto` and it has HDMI plugged. 

Forcing HDMI audio, or disabling audio altogether do not fix the issue.

Enabled Kodi debug logs and tried installing a few packages mentioned like udisks2 and upower, nothing

glogiotatidis, 2019-10-18:
Fix: It was the default GPU memory split! 

Kodi was installed non-interactivelly with `AUTO_SETUP_AUTOMATED=1` which doesn't give the option to setup the memory split. Setting the mem split via the `dietpi-config` tool, fixes the issue

Can we have memory split conf values in `dietpi.txt`?

MichaIng, 2019-10-18:
@glogiotatidis 
Ah interesting, actually it should apply 320 MiB GPU memory automatically. Was it still 16 MiB in your case? Let me recheck the code.

MichaIng, 2019-10-18:
Found the issue: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L12192-L12197
- On noninteractive install, the inceased memory split is not applied automatically. User would be asked, but these G_WHIP_YESNO dialogs are answered with "No" by default.

Hmm since Kodi requires AFAIK min 128 MiB, we should force this (if lower value found) in every case, then ask user if recommended value 320 MiB should be applied. Although on RPi4 256 MiB should be more then sufficient since several GPU features have been moved over to system memory with that model. However this is another topic.

glogiotatidis, 2019-10-18:
Neat, thanks for the quick debug session @MichaIng

MichaIng, 2020-03-19:
Okay, the minimum 128 MiB, required for Kodi to start, is now applied automatically: https://github.com/MichaIng/DietPi/commit/309688a1eb4da856e103a0ca6260be9ca593d6f0

closed 2020-05-03 by MichaIng