MichaIng/DietPi #3174: Error fresh install v6.25 Buster on RPi-zero-w when upgrading to v6.26

maartenlangeveld, 2019-10-18:
I tried a fresh install of DietPi in order to upgrade from Stretch Buster on my RPi-zero-w. Downloaded the DietPi v6.25 Buster image, wrote it to sd card and after first startup the system automatically initiates upgrade from DietPi v6.25 top v6.26. The process halts half way with following error (have re-tried many times but that does not help).

```
 ┌───────────────────────────────┤ DietPi Error Handler: ├────────────────────────────────┐
                                   │ DietPi-Update: G_AGUG
                                   │  - Exit code: 100
                                   │  - DietPi version: v6.25.3 (MichaIng/master) | HW_MODEL:1 | HW_ARCH:1 | DISTRO:5
                                   │  - Image creator: DietPi Core Team
                                   │  - Pre-image: Raspbian Lite
                                   │
                                   │ Log file contents:
                                   │ Setting up libss2:armhf (1.44.5-1+deb10u2) ...^M
                                   │ Setting up usb.ids (2019.07.27-0+deb10u1) ...^M
                                   │ Setting up sudo (1.8.27-1+deb10u1) ...^M
                                   │ Failed to connect to bus: No such file or directory^M
                                   │ Failed to connect to bus: No such file or directory^M
                                   │ Failed to connect to bus: No such file or directory^M
                                   │ Failed to connect to bus: No such file or directory^M
                                   │ invoke-rc.d: initscript sudo, action "restart" failed.^M

```

MichaIng, 2019-10-18:
@maartenlangeveld 
Many thanks for your report.

I am confused now, did time sync initially fail or APT installs? Or did you ignore time sync error and then ran into second error during APT installs?

I guess currently systemd-udevd is down, isn't it?
```
systemctl status systemd-udevd
```

It is already strange that it uses sysvinit helper to try restarting the services (invoke-rc.d) since it is a systemd init where systemctl should be used.
And then it looks for dbus which is not required for anything.
I will check bugreport.

maartenlangeveld, 2019-10-18:
@MichaIng 

Meanwhile I have done a new fresh install and same happens. I do not recall there was a time sync fail or error. I think, but I am not sure, it was APT installs.

Have ingnored error many times and rebooted, trying to get it working.

Have sent a couple of bug reports along the way. You should find them.

P.S. I also saw haveged as service whule in the release notes of v6.26 i read that haveged had been replaced by a similar service....

MichaIng, 2019-10-18:
@maartenlangeveld 
Could you please paste: `ls -ld /tmp`

Jep haveged has been replaced by rngd (rng-tools) however this should not play any role. It must have something to do with the systemd/udev update. I ran into an issue myself where /tmp had wrong permissions (mode=1777 node applied as it should), however this was on Bullseye. And on VM even with 0755 permissions there are no issues like yours.

I checked if "Failed to connect to bus" messages are due to failed systemd-journald start, however on VM again this does not matter at all. Something deeper leads to all `systemctl` calls failing with this error, however dbus actually cannot be the reason.

maartenlangeveld, 2019-10-18:
@MichaIng

ls -ld /tmp
drwxrwxrwt 10 root root 300 Oct 18 23:19 /tmp

In my 6.26 install there's still haveged.... How can that be?

BTW upgrade from 6.25 stretch went completely hassle free. These problems all relate to fresh install using v6.25 Buster image.

maartenlangeveld, 2019-10-18:
@Michael,

Can force a re-install of 6.26 update?

maartenlangeveld, 2019-10-18:
@MichaIng 

Have started again since system was not stable. Indeed got a error message which I might have forgotten (sorry), if i remember well was somehting with rsync?? With the recommendation to issue a sysctl command. I have aborted the update (crtl+c) and issued the command. Now trying again, will keep you informed.

maartenlangeveld, 2019-10-18:
@MichaIng 

Error message during upgrade to v6.26 (Buster)
...
Unpacking raspberrypi-sys-mods (20190730) over (20190429) ...

Broadcast message from systemd-journald@DietPi (Sat 2019-10-19 00:01:10 CEST):

systemd[1]: Caught <SEGV>, dumped core as pid 2374.

Failed to reload daemon: Connection reset by peer

Broadcast message from systemd-journald@DietPi (Sat 2019-10-19 00:01:14 CEST):

systemd[1]: Freezing execution.

Failed to connect to bus: No such file or directory
Failed to connect to bus: No such file or directory
Failed to connect to bus: No such file or directory
...

maartenlangeveld, 2019-10-18:
@MichaIng,

Is it possible to disable automatic update after fresh install, i.e. can I change a setting in dietpi.txt on the sd-card that prevents it will automatically update after installation?

maartenlangeveld, 2019-10-18:
@MichaIng 

> Another question: Where you connected via local screen + keyboard or SSH or even serial console?
> Because all those non-virtual (non-SSH) console commands seems to have been lost:

Connected through SSH (headless install).

MichaIng, 2019-10-20:
@maartenlangeveld 
Sorry for delay. I tried to replicate via systemd-nspawn boot and did some research + your analysed bugreport, but finally this APT-internal systemd (and/or udev) upgrade error is hard to trace. All I know is that the systemd init was completely redone as postinst script (as part of APT upgrade) and this step crashed for some reason so that systemctl is not able to connect to any running init/service part anymore. Not sure why even reboot does not solve it, probably some important postinst (--configure) step is missing then, which breaks ordinary systemd init on boot.

Bad was then that other important installs failed as well with systemd/systemctl failure in the first place, e.g. sudo, and as well shutdown is not done well without graceful service stop, which again can be a reason for boot problems then.

However I could have tested on real RPi and try to debug and fix things, however since most likely a new image would have been the result anyway, I simply redone the image directly and skipped wasting time on systemd upgrade issue 😉. If this re-appears, I have do dig deeper, but for now, please test this new image: https://dietpi.com/downloads/images/testing/DietPi_RPi-ARMv6-Buster.7z

I used a new method to create it via `systemd-nspawn` (container) on a virtual machine + `qemu-static` to allow running ARM binaries. This works VERY well, allows to even "boot" the system from image file and run our image installer/preparation there. Finally we can do the whole process with a single automated script by this, from pre-image download until ready 7z archive upload to download/testing dir, all from a VM or any other machine without even the need to own the target SBC 😃. However at least one final boot test on real SBC should be done.

maartenlangeveld, 2019-10-20:
@MichaIng 

Thanks a lot ! New image works flawless on my RPi-zero-w 😉

Maarten

MichaIng, 2019-10-20:
@maartenlangeveld 
Many thanks for testing ❤️, will move it into regular download place.

MichaIng, 2019-12-11:
Due to some bugs in v6.26 that I don't want to ship as stable image, I'll redo one after v6.27 release and upload it to stable. However I mark this issue as closed now. A bunch of new images will be done after v6.27 release.

closed 2019-12-11 by MichaIng