MichaIng/DietPi #3177: Xserver package fails to install on dietpi-software - Odroid C2

YouveGotRedOnYou, 2019-10-19:
# Details:
- Date           | Sat 19 Oct 14:42:36 BST 2019
- Bug report     | 4483972a-4e9e-40ae-b3ec-3bd01d4d93ac
- DietPi version | v6.26.3 (Fourdee/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid C2 (aarch64) (index=12)
- Kernel version | Linux DietPi 3.16.57+ #1 SMP PREEMPT Sun Aug 19 15:27:56 CEST 2018 aarch64 GNU/Linux
- Distro         | stretch (index=4)
- Command        | G_AGI libegl1/stretch-backports xserver-xorg xinit xcompmgr xterm dbus-x11 xfonts-base x11-xserver-utils x11-utils libgl1-mesa-dri mesa-utils mesa-utils-extra
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ... using diet-Software to install
2. ...

#### Expected behaviour:

- ...Software should install

#### Actual behaviour:

- ...install fails

#### Extra details:

- ...

#### Additional logs:
```
E: Unable to correct problems, you have held broken packages.
```

MichaIng, 2019-10-19:
@YouveGotRedOnYou 
Many thanks for your report. Please paste the full output of:
```
apt install libegl1/stretch-backports xserver-xorg xinit xcompmgr xterm dbus-x11 xfonts-base x11-xserver-utils x11-utils libgl1-mesa-dri mesa-utils mesa-utils-extra
```

YouveGotRedOnYou, 2019-10-19:
Yessir! Here you go...

pt install libegl1/stretch-backports xserver-xorg xinit xcompmgr                         xterm dbus-x11 xfonts-base x11-xserver-utils x11-utils libgl1-mesa-dri mesa-uti                        ls mesa-utils-extra
Reading package lists... Done
Building dependency tree
Reading state information... Done
dbus-x11 is already the newest version (1.10.28-0+deb9u1).
libgl1-mesa-dri is already the newest version (13.0.6-1+b2).
mesa-utils is already the newest version (8.3.0-3).
mesa-utils-extra is already the newest version (8.3.0-3).
x11-utils is already the newest version (7.7+3+b1).
x11-xserver-utils is already the newest version (7.7+7+b1).
xcompmgr is already the newest version (1.1.7-1+b1).
xfonts-base is already the newest version (1:1.0.4+nmu1).
xinit is already the newest version (1.3.4-3+b1).
xserver-xorg is already the newest version (1:7.7+19).
xterm is already the newest version (327-2).
Selected version '1.1.0-1~bpo9+1' (Debian Backports:stretch-backports [arm64]) f                        or 'libegl1'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libegl1 : Depends: libegl-mesa0 but it is not going to be installed
 mesa-utils-extra : Depends: libegl1-mesa (>= 7.8.1) or
                             libegl1-x11
E: Unable to correct problems, you have held broken packages.

YouveGotRedOnYou, 2019-10-19:
I was able to manually install most of the apps, but it seems to be stuck on libegl1/stretch-backports.

MichaIng, 2019-10-19:
@YouveGotRedOnYou 
Found the issues... silly bug, please do:
```
sed -i 's/G_AGI $libegl1/G_AGI/' /DietPi/dietpi/dietpi-software
```

MichaIng, 2019-10-19:
Fixed with: https://github.com/MichaIng/DietPi/commit/8eae2950048e4c480f3aa236be7590d1aa146407
Backport via: https://github.com/MichaIng/DietPi/commit/6609192aa824f0626af20cb3dc73e215b231f1e3

YouveGotRedOnYou, 2019-10-20:
Lovely, that worked. Thank you.

closed 2019-10-20 by YouveGotRedOnYou

Haylerd, 2021-02-12:
I have this error.

Joulinar, 2021-02-12:
Pls can you share the exact error you have. The original issue has been closed a year ago.