MichaIng/DietPi #3178: dietpi-updat tells me that V6.25.3 is up-to-date although v.6.26.3 is out there

BerndKohl, 2019-10-19:
### Creating a bug report/issue

#### Required Information
- DietPi version |  v6.25.3
- Distro version | bullseye/sid
- Kernel version | Linux two 4.19.50-v7+ #896 SMP Thu Jun 20 16:11:44 BST 2019 armv7l GNU/Linux
- SBC device | RPi 3 Model B (armv7l)
- Power supply used | 5V 2A
- SDcard used | SanDisk ultra

#### Steps to reproduce
1. dietpi-update

 ─────────────────────────────────────────────────────
 DietPi v6.25.3 (jessie-support) : 18:49 - Sa 19.10.2019
 ─────────────────────────────────────────────────────
 - Device model : RPi 3 Model B (armv7l)
 - Uptime : up 15 weeks, 3 days, 21 hours, 27 minutes
 - CPU temp : 48'C : 118'F (Optimal temperature)
 - FQDN/hostname : XXX
 - LAN IP : 192.XXX.XXX.XXX (eth0)
 - Freespace (RootFS) : 106G
 - MOTD : WARNING: On Raspberry Pi the new available firmware upgrade, that adds                                                                                                                                                              support for RPi 4, uses nearly all space in /boot. Please check your /boot dire                                                                                                                                                             ctory and in case remove junk from it.
          See this GitHub issue about how a clean /boot directory should look li                                                                                                                                                             ke (including new RPi 4 files): https://github.com/MichaIng/DietPi/issues/2935
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder)
 Image           : DietPi Core Team (pre-image: Raspbian Lite)
 Web             : https://DietPi.com | https://twitter.com/dietpi_
 Patreon Legends : PINE64 community
 Donate          : https://DietPi.com/#donate
 DietPi Hosting  : Powered by https://MyVirtualServer.com

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@XXX:~# dietpi-update
[  OK  ] DietPi-Update | Root access verified.
[  OK  ] DietPi-Update | RootFS R/W access verified.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/MichaIng/DietPi/jessie-support/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/MichaIng/DietPi/jessie-support/dietpi/server_version-6
[  OK  ] DietPi-Update | No update required, your DietPi installation is already up to date:
[ INFO ] DietPi-Update | Current version : v6.25.3
[ INFO ] DietPi-Update | Latest version  : v6.25.3

MichaIng, 2019-10-19:
@BerndKohl 
Many thanks for your report.

Somehow your system was migrated to the "jessie-support" branch, for which no further updates are available. However your header says it's "Bullseye"? Is that intended? Did you dist-upgrade? Especially on RPi (Raspbian) I do absolutely not recommend to run Bullseye on a production server. I run into A LOT of problems, since Raspbian does not take care the testing branch very good (inconsistencies and such).

However, just to be sure, can you paste: `cat /etc/os-release`

To move over to normal master branch and apply the update:
```
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /DietPi/dietpi.txt
dietpi-update 1
```

BerndKohl, 2019-10-20:
Here you are:
root@XXX:~# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux bullseye/sid"
NAME="Raspbian GNU/Linux"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

BerndKohl, 2019-10-20:
`G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /DietPi/dietpi.txt
dietpi-update 1` did the trick. Now I am running the latest version.
Many thanks!

closed 2019-10-20 by BerndKohl