MichaIng/DietPi #3180: SyncThing not available after install

ralban, 2019-10-20:
#### Details:
- Date           | Wed 19 Oct  2019
- Bug report     |
- DietPi version | v6.26.3 (MichaIng/master)
- Image creator  | N/A
- Pre-image      | N/A
- SBC device     | NULL (index=20)
- Kernel version | Linux dietpihole 4.19.0-6-amd64 #1 SMP Debian 4.19.67-2+deb10u1 (2019-09-20) x86_64 GNU/Linux
- Distro         | buster (index=5)
- Command        | Connection test: https://github.com/MichaIng/DietPi/archive/master.zip
- Exit code      | ?
- Software title | SyncThing

#### Steps to reproduce:

1. run dietpi-PREP script on fresh Debian 10.1 install
2. install SyncThing
3. Attempt to access http://IP-ADDRESS:8384

#### Expected behaviour:

- SyncThing should load in my browser

#### Actual behaviour:

- SyncThing does not load

#### Extra details:

- BugReport submitted

#### Additional logs:
- BugReport submitted

MichaIng, 2019-10-20:
@ralban 
Many thanks for your report.

If you submitted a bugreport, please paste the ID here: `mawk 'NR==5' /DietPi/dietpi/.hw_model`

ralban, 2019-10-21:
b9746b36-f05f-42ca-bc73-e65d756dda33

mowestusa, 2019-10-24:
I have the same issue, just installed syncthing yesterday, and I have been unable to access the web interface. I do have the additional information below that I gathered from dietpi-services logs.

I'm a bit of a noob, but it looks like it might have to do with directory permissions. I do have my "dietpi-userdata" mounted on an external hard drive attached by USB.

Here is the log notes from attempting to restart the syncthing service today:

 ● syncthing.service - Syncthing (DietPi)
 │    Loaded: loaded (/etc/systemd/system/syncthing.service; disabled; vendor preset:         
 │ enabled)                                                                                   
 │    Active: failed (Result: exit-code) since Thu 2019-10-24 11:39:16 EDT; 15s ago          
 │   Process: 9382 ExecStart=/etc/syncthing/syncthing                                         
 │ -logfile=/var/log/syncthing/syncthing.log -logflags=3                                     
 │ -home=/mnt/dietpi_userdata/syncthing (code=exited, status=1/FAILURE)                      
 │  Main PID: 9382 (code=exited, status=1/FAILURE)                                            
 │                                                                                            
 │ Oct 24 11:39:14 Wolfberry syncthing[9382]: [monitor] 2019/10/24 11:39:14 INFO:             
 │ Starting syncthing                                                                         
 │ Oct 24 11:39:14 Wolfberry syncthing[9382]: [start] 2019/10/24 11:39:14 INFO:               
 │ syncthing v1.3.0 "Fermium Flea" (go1.13.1 linux-arm) teamcity@build.syncthing.net          
 │ 2019-10-01 05:34:58 UTC                                                                    
 │ Oct 24 11:39:14 Wolfberry syncthing[9382]: [start] 2019/10/24 11:39:14 INFO:               
 │ Generating ECDSA key and certificate for syncthing...                                      
 │ Oct 24 11:39:15 Wolfberry syncthing[9382]: [start] 2019/10/24 11:39:15 WARNING:            
 │ Failed to load/generate certificate: save cert: open
 │ /mnt/dietpi_userdata/syncthing/cert.pem: permission denied                                 
 │ Oct 24 11:39:15 Wolfberry syncthing[9382]: [monitor] 2019/10/24 11:39:15 INFO:             
 │ Syncthing exited: exit status 1                                                            
 │ Oct 24 11:39:16 Wolfberry syncthing[9382]: [monitor] 2019/10/24 11:39:16 INFO:             
 │ Default folder created and/or linked to new config                                         
 │ Oct 24 11:39:16 Wolfberry syncthing[9382]: [monitor] 2019/10/24 11:39:16 WARNING: 4        
 │ restarts in 9.011826s; not retrying further                                                
 │ Oct 24 11:39:16 Wolfberry systemd[1]: syncthing.service: Main process exited,              
 │ code=exited, status=1/FAILURE                                                              
 │ Oct 24 11:39:16 Wolfberry systemd[1]: syncthing.service: Unit entered failed state.        
 │ Oct 24 11:39:16 Wolfberry systemd[1]: syncthing.service: Failed with result                
 │ 'exit-code'.

When I cd into the directory this is the permissions that I see for the directories that syncthing is creating and attempting to use when it starts up:

drwx------ 2 root   root   4096 Oct 23 14:51 syncthing
drwxr-xr-x 2 root   root   4096 Oct 23 14:51 syncthing_data
drwxrwxr-x 2 dietpi dietpi 4096 Jan 23  2019 Video

I included the "Video" directory, not because syncthing created it, but it was created in that same userdata directory and it has very different permissions from the two folders created by syncthing.

I also sent the following dietpi-bugreport:
Bug report sent, reference code: 9a39253a-499a-4358-ac3b-8bbeb55ee5c8

I hope this helps.

g7kse, 2019-10-24:
Me three. Same install setup as @mowestusa

g7kse, 2019-10-25:
Ok Thanks @MichaIng all ok on gui which is now displayed but we still have a permission problem with folders

 2019-10-25 08:24:18: Failed to create folder marker: mkdir /mnt/dietpi_userdata/syncthing_data/.stfolder: permission denied

2019-10-25 08:24:18: Error on folder "Syncthing Data" (0000-0000): folder marker missing

MichaIng, 2019-10-25:
@g7kse 
Ah that's true:
```
chown -R dietpi:dietpi /mnt/dietpi_userdata/syncthing*
systemctl restart syncthing
```

g7kse, 2019-10-25:
All sorted. tnx 

Alex

closed 2020-01-01 by MichaIng