MichaIng/DietPi #3181: Amiberry libpnh12 problem

wasu111, 2019-10-20:
- DietPi version | 6.26.3 master/Fourdee
- Distro version | 9.11
- Kernel version | Linux DietPi 4.14.66+ #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018 armv7l GNU/Linux
- SBC device | Odroid XU3/XU4/HC1/HC2 (armv7l)
- Power supply used | 5V 3A 
- SDcard used | SanDisk extreeme)

- Software title | DietPi only for Amiberry emulation
- FRESH instalation of DietPi from homepage

#### Steps to reproduce
After clean install of DietPi and Amiberry from home page there is a problem with running it.
systemctl status amiberry.service  says libpng12.so.0 cannot open shared object file: No such file or directory. There is no way to install libpng12.so.0 after
sudo apt-get install libpng12.so.0
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libpng12.so.0
E: Couldn't find any package by glob 'libpng12.so.0'
E: Couldn't find any package by regex 'libpng12.so.0'

#### Expected behaviour
Amiberry should start

#### Actual behaviour
Amiberry do not load

#### Extra details
[image]

MichaIng, 2019-10-20:
@wasu111 
Many thanks for your report.

Seems we were too quick to enable Odroid XU4 support for Amiberry since the binary was obviously compiled against Debian Jessie. We'll rebuild all of them for v6.27 anyway, however meanwhile you can solve the missing (wrong) libpng version by installing the package from Jessie manually:
```
cd /tmp
wget https://deb.debian.org/debian/pool/main/libp/libpng/libpng12-0_1.2.50-2+deb8u3_armhf.deb -O libpng.deb
dpkg -i libpng.deb
rm libpng.deb
```
Lets see if this is the only Jessie vs Stretch library mismatch.

wasu111, 2019-10-21:
Thanks for fast answer but unfortunately still got problem. This time with EGL window

[image]

MichaIng, 2019-10-21:
@wasu111 
Hmm X.org (and GPU drivers) have been successfully installed before?
```
G_AGI --reinstall xf86-video-armsoc-odroid malit628-odroid
```

Otherwise I am afraid we first need to recompile the XU4 amiberry binary against Stretch library versions.
Will keep you informed, once done.

wasu111, 2019-10-21:
I've tried to reinstall G_AGI and I've tried to write image again still the same problem.

MichaIng, 2019-12-03:
@wasu111 
I compiled a new XU4 binary based on new Amiberry 3.0.4. I cannot test it on a real machine, run into some segmentation fault within qemu container, but I hope this is only due to missing SoC graphics device. If you want to test it:
```
mv /mnt/dietpi_userdata/amiberry /mnt/dietpi_userdata/amiberry_bak
cd /tmp
wget https://dietpi.com/downloads/binaries/stretch/amiberry_xu4.tar.bz2
tar -xf amiberry_xu4.tar.bz2
mv amiberry /mnt/dietpi_userdata/amiberry
LD_LIBRARY_PATH='/mnt/dietpi_userdata/amiberry/lib' /mnt/dietpi_userdata/amiberry/amiberry
```
If this works fine, you can copy your old configs in place, although it is recommended to use the new amiberry.conf, respectively have Amiberry recreating it freshly after update.
```
cp -a /mnt/dietpi_userdata/amiberry_bak/{conf,kickstarts,savestates,screenshots} /mnt/dietpi_userdata/amiberry/amiberry/
```
Since we packed the SDL2 lib now right into the Amiberry dir, this needs to be added to the library path environment, but the files from the shared system dir can be removed instead:
```
rm -v /usr/local/lib/libSDL2*
cat << _EOF_ > /etc/systemd/system/amiberry.service
[Unit]
Description=Amiberry Amiga Emulator (DietPi)

[Service]
WorkingDirectory=$G_FP_DIETPI_USERDATA/amiberry
Environment=LD_LIBRARY_PATH=$G_FP_DIETPI_USERDATA/amiberry/lib
ExecStart=$G_FP_DIETPI_USERDATA/amiberry/amiberry

[Install]
WantedBy=local-fs.target
_EOF_
systemctl daemon-reload
systemctl restart amiberry
```

To be true I have not much experience with those .uae files.  We had one shipped with the previous version, but the content looks like auto-generated for a RPi1 with file paths that do not match. However if you need a .uae config auto-loaded, easiest is to create `/mnt/dietpi_userdata/amiberry/conf/default.uae`, which is loaded automatically by the binary, if present and no other given via cmd argument.

wasu111, 2019-12-03:
Hi.Superb I'll try it tomorrow and let You knowWasu1113 gru

MichaIng, 2019-12-04:
Just to assure that all dependencies are installed:
apt install libxml2 libfreetype6 libflac8 libmpeg2-4 libmpg123-0 malit628-odroid

closed 2020-01-01 by MichaIng