MichaIng/DietPi #3182: When installation of certain software fails, it's impossible to install other software

maartenlangeveld, 2019-10-20:
rpi-zero-w, v6,26.3 (Buster)

Tried to install Java using dietpi-software, additional software, installation failed (see below).

However it is not possible to install other software since when i restart dietpi-software to install another package it still tries to install the failed package. Cannot uninstall failed package etiher it is not ticked in installed software nor in software to de-install.

```
                         ┌───────────────────────────────┤ DietPi Error Handler: ├────────────────────────────────┐
                         │ DietPi-Software: G_AGUG
                         │  - Exit code: 100
                         │  - DietPi version: v6.26.3 (MichaIng/master) | HW_MODEL:1 | HW_ARCH:1 | DISTRO:5
                         │  - Image creator: DietPi Core Team
                         │  - Pre-image: Raspbian Lite
                         │
                         │ Log file content:
                         │ Setting up ca-certificates-java (20190405) ...^M
                         │ head: cannot open '/etc/ssl/certs/java/cacerts' for reading: No such file or directory^M
                         │ Error occurred during initialization of VM^M
                         │ Server VM is only supported on ARMv7+ VFP^M
                         │ dpkg: error processing package ca-certificates-java (--configure):^M
                         │  installed ca-certificates-java package post-installation script subprocess returned
                         │ error exit status 1^M
                         ┌───────────────────────────────┤ DietPi Error Handler: ├────────────────────────────────┐
                         │ Processing triggers for ca-certificates (20190110) ...^M
                         │ Updating certificates in /etc/ssl/certs...^M
                         │ 0 added, 0 removed; done.^M
                         │ Running hooks in /etc/ca-certificates/update.d...^M
                         │ ^M
                         │ Error occurred during initialization of VM^M
                         │ Server VM is only supported on ARMv7+ VFP^M
                         │ E: /etc/ca-certificates/update.d/jks-keystore exited with code 1.^M
                         │ done.^M
                         │ Errors were encountered while processing:^M
                         │  ca-certificates-java^M
                         │ E: Sub-process /usr/bin/dpkg returned an error code (1)
                         │
                         │ If problems persist, please report at https://github.com/MichaIng/DietPi/issues for
                         │ investigation.
                         │
                         │ Unable to continue, DietPi-Software will now terminate.
```

MichaIng, 2019-10-20:
@maartenlangeveld 
Many thanks for your report.

Yeah I already saw this in another bug report by chance. Annoying is that it is the official Raspbian version of `ca-certificates-java` that fails here "by design" on ARMv6. See in error log: `Server VM is only supported on ARMv7+ VFP`

So we need to report this to the Raspbian repo maintainers, but sadly there is no real bug tracker for it... Hopefully we get some attention here: https://www.raspberrypi.org/forums/viewforum.php?f=66
_________
And yeah, this is a thing about APT I don't like as well. If one package fails to configure, APT denies any other installs until this has been fixed. What sounds reasonable in the first place, creates some rats tail in cases, especially when the repo itself is not maintained in a 100% consistent way, like in this  `ca-certificates-java` case.

Some users then tend to remove the related postinst script, e.g. `/var/lib/dpkg/info/ca-certificates-java.postinst`, but I prefer to check the script and only comment the one command that fails.

maartenlangeveld, 2019-10-20:
@MichaIng 

I am surprised since I installed Java before on my rpi-zero-w (stretch) without problems....

Thanks,
Maarten.

MichaIng, 2019-10-20:
@maartenlangeveld 
In both cases (ca-certificates and ca-certificates-java) `/etc/ca-certificates/update.d/jks-keystore` fails which is part of the java package: https://packages.debian.org/buster/all/ca-certificates-java/filelist

Not sure which command inside exactly wants to create a VM, however it must be another external binary:
```
#!/bin/sh

set -e

# use the locale C.UTF-8
unset LC_ALL
LC_CTYPE=C.UTF-8
export LC_CTYPE

storepass='changeit'
if [ -f /etc/default/cacerts ]; then
    . /etc/default/cacerts
fi

arch=`dpkg --print-architecture`
JAR=/usr/share/ca-certificates-java/ca-certificates-java.jar

nsslib_name()
{
    if dpkg --assert-multi-arch 2>/dev/null; then
        echo "libnss3:${arch}"
    else
        echo "libnss3"
    fi
}

echo ""
if [ "$cacerts_updates" != yes ] || [ "$CACERT_UPDATES" = disabled ] || [ ! -e $JAR ]; then
    echo "updates of cacerts keystore disabled."
    exit 0
fi

if ! mountpoint -q /proc; then
    echo >&2 "the keytool command requires a mounted proc fs (/proc)."
    exit 1
fi

for jvm in java-7-openjdk-$arch java-7-openjdk \
           oracle-java7-jre-$arch oracle-java7-server-jre-$arch oracle-java7-jdk-$arch \
           java-8-openjdk-$arch java-8-openjdk \
           oracle-java8-jre-$arch oracle-java8-server-jre-$arch oracle-java8-jdk-$arch \
           java-9-openjdk-$arch java-9-openjdk \
           oracle-java9-jre-$arch oracle-java9-server-jre-$arch oracle-java9-jdk-$arch \
           java-10-openjdk-$arch java-10-openjdk \
           oracle-java10-jre-$arch oracle-java10-server-jre-$arch oracle-java10-jdk-$arch \
           java-11-openjdk-$arch java-11-openjdk \
           oracle-java11-jre-$arch oracle-java11-server-jre-$arch oracle-java11-jdk-$arch; do
    if [ -x /usr/lib/jvm/$jvm/bin/java ]; then
        export JAVA_HOME=/usr/lib/jvm/$jvm
        PATH=$JAVA_HOME/bin:$PATH
    	break
    fi
done

if dpkg-query --version >/dev/null; then
    nsspkg=$(dpkg-query -L "$(nsslib_name)" | sed -n 's,\(.*\)/libnss3\.so$,\1,p'|head -n 1)
    nsscfg=/etc/${jvm%-$arch}/security/nss.cfg
    nssjdk=$(test ! -f $nsscfg || sed -n '/nssLibraryDirectory/s/.*= *\(.*\)/\1/p' $nsscfg)
    if [ -n "$nsspkg" ] && [ -n "$nssjdk" ] && [ "$nsspkg" != "$nssjdk" ]; then
        ln -sf $nsspkg/libnss3.so $nssjdk/libnss3.so
    fi
    softokn3pkg=$(dpkg-query -L "$(nsslib_name)" | sed -n 's,\(.*\)/libsoftokn3\.so$,\1,p'|head -n 1)
    if [ -n "$softokn3pkg" ] && [ -n "$nssjdk" ] && [ "$softokn3pkg" != "$nssjdk" ]; then
        ln -sf $softokn3pkg/libsoftokn3.so $nssjdk/libsoftokn3.so
    fi
fi

do_cleanup()
{
    [ -z "$temp_jvm_cfg" ] || rm -f $temp_jvm_cfg
    if [ -n "$nsspkg" ] && [ -n "$nssjdk" ] && [ "$nsspkg" != "$nssjdk" ]
    then
        rm -f $nssjdk/libnss3.so
    fi
    if [ -n "$softokn3pkg" ] && [ -n "$nssjdk" ] \
       && [ "$softokn3pkg" != "$nssjdk" ]
    then
        rm -f $nssjdk/libsoftokn3.so
    fi
}

if java -Xmx64m -jar $JAR -storepass "$storepass"; then
    do_cleanup
else
    do_cleanup
    exit 1
fi

echo "done."
```
- From what I see best guess is the `java` call itself new the bottom. You could call the script in debug mode to be 100% sure: `sh -x /etc/ca-certificates/update.d/jks-keystore`

maartenlangeveld, 2019-10-20:
@MichaIng 

Thanks, have not much linux knowledge, so do not completely understand you. Have installed java jre 8 instead:

> apt install openjdk-8-jre-zero

I need java since it's required by minimserver.

MichaIng, 2019-10-21:
@maartenlangeveld 
Best workaround I could find so far is:
```
G_CONFIG_INJECT 'cacerts_updates=' 'cacerts_updates=no' /etc/default/cacerts
```
But that should be reverted when the issue has been fixed on Raspbian maintainer side.

However if it works with Java 8, that is of course also an alternative, as long as nothing explicitly requires Java 11.

maartenlangeveld, 2019-10-21:
@MichaIng,

Thanks for your great support. I'll stick with Java 8 for the moment.

maartenlangeveld, 2019-10-31:
@MichaIng ,

Thanks !

Have manually installed Java 8 since MinimServer (also manual installed) requires java. I can still endorse MinimServer to be included in optional software list. It's a fine media server and works hassle-free. It can be remotely controlled using MinimWatch java app.

MichaIng, 2019-11-08:
Resolved with: https://github.com/MichaIng/DietPi/commit/0d54f6ca5334ca30ebfc0b2ac660fdcc4b6e0e5d
Changelog: https://github.com/MichaIng/DietPi/commit/ce7335c61ee590943b3b191a5f8cb1f311848c8b

closed 2020-01-01 by MichaIng