MichaIng/DietPi #3187: [v6.26.3] PSU Noise reduction :  does not toggle

maartenlangeveld, 2019-10-21:
```
┌──────────────────────────────────────────────────┤ DietPi-Config ├───────────────────────────────────────────────────┐
│ Please select an option:                                                                                             │
│                                                                                                                      │
│                             Soundcard           : [justboom-digi]                                                    │
│                             DietPi-JustBoom     : Launches EQ and MPD audio options menu                             │
│                             PSU noise reduction : [Off]                                                              │
│                                                 ●──────────────────────────────────────●                             │
│                             Disable             : Disable all audio capabilities                                     │
│                                                                                                                      │
│                                                                                                                      │
│                                  <Ok>                                      <Back>                                    │
│                                                                                                                      │
                    └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘
```

MichaIng, 2019-10-21:
@maartenlangeveld 
Many thanks for your report.

Most likely we forgot to update headless mode estimation. There was a change in RPi firmware which rendered the previous `config.txt` entry obsolete, so real headless mode must again be applied on runtime via `tvservice -o` and we use the related `dietpi.txt` to check whether to run this on boot or not.
Most likely for noise reduction, the old value is still checked. However headless mode is applied via external script, thus it works regardless. Just check that last output shows headless mode being enabled and `AUTO_SETUP_HEADLESS=1` in `dietpi.txt`.

Will check later and fix.

MichaIng, 2019-10-21:
Fixed with: https://github.com/MichaIng/DietPi/commit/9efedf863f5982b77dde434a56f7f7b0748b8a43
Changelog: https://github.com/MichaIng/DietPi/commit/19af9ecb52e9c58d8a9e3efd886626cfb4dcff61

closed 2020-01-01 by MichaIng