MichaIng/DietPi #3191: HQPlayer NAA daemon (networkaudiod) upgrade from 3.5.6 to 3.6.0 ?

Heroldgray, 2019-10-22:
Hi @MichaIng 
HQPlayer NAA (networkaudiod) version is 3.5.6-41 in last DietPi released version.
NAA has been upgraded from 3.5.6 to 3.6.0
See: https://www.signalyst.eu/bins/naa/linux/stretch/
Thank's for your great work !

MichaIng, 2019-10-22:
@Heroldgray
Many thanks for this info. I will update our download link with v6.27. Meantime you can manually download and update the package, also as a failsafe test that it works well.

MichaIng, 2019-10-22:
PR up: https://github.com/MichaIng/DietPi/pull/3193
- Grabs latest package from website automatically now.
- As well uses Buster packages, if available, currently not yet for armhf: https://www.signalyst.eu/bins/naa/linux/buster/
Last version was available for Buster with amd64 only, current version additionally for arm64, and next version additionally for armhf? 🙂

closed 2020-01-01 by MichaIng