MichaIng/DietPi #3194: jackett crashing every 2 weeks

pengocha, 2019-10-23:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=6
- Distro version | 10.1
- Kernel version | Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- SBC device | RPi 2 Model B (armv7l)

#### Additional Information (if applicable)
- Software title | Jackett

- Bug report ID | 48452fd8-17fb-43e1-bf94-4aa6c5055289^C

#### Actual behaviour
Every 2 weeks my Jackett service stops, in the services the state is set to 'activating'.
When i uninstall and reinstall the service it's ok again for another 2 weeks, after that it stops again.

MichaIng, 2019-10-23:
@pengocha 
Many thanks for your report.

Might it be the internal auto updater? https://github.com/MichaIng/DietPi/issues/2593
I will try to find a way to disable it via install script by default, since fix from developer seems to stall 😞.

pengocha, 2019-10-23:
> 
> 
> @pengocha
> Many thanks for your report.
> 
> Might it be the internal auto updater? #2593
> I will try to find a way to disable it via install script by default, since fix from developer seems to stall 😞.

Thx, i'll give it a try and report back :)

MichaIng, 2019-11-06:
The issue should be solved by the following commit for v6.27: https://github.com/MichaIng/DietPi/commit/9a543c7210dcdeabb5f382f261a6d27efcdc8f6d

closed 2020-01-01 by MichaIng