MichaIng/DietPi #3205: NO-IP buggy?

johnnyt83, 2019-10-25:
Hi Micha,

I just installed No-IP on a new pi zero with DietPi (6.26.3) and noticed it did not go "online" after successfully entering login credentials. I gave "5" as the new refresh interval and noticed this:

https://abload.de/img/thumbnail_img_2019102lpkk9.jpg

https://abload.de/img/thumbnail_img_2019102dwkqz.jpg

CPU usage did not even drop after 10 minutes.
Is the plugin buggy?
BTW: does the plugin prevent me from having to login into the webUI every 30 days to make my domain not expire?
best wishes.

MichaIng, 2019-10-26:
@johnnyt83 
Many thanks for your report. Strange is that there is already an instance of `noip2` running (with close to 0% usage) while the second one then has high usage. Should be only one instance.

I just checked the code:
- https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-config#L4517-L4523
  - Service is stopped, when using the option
  - Config file creation (input prompt) is called
  - Service is restarted
- https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L8340
  - Leads to process being killed gracefully and should wait/delay until it is really stopped.

Did it probably fail to stop the service? However this should produce an error prompt then. But to check:
```
systemctl status noip2
```

MichaIng, 2020-03-18:
@johnnyt83 
I have no No-IP command, hence cannot test 100%, but `noip2 -C` which causes the 100% CPU usage is the config creation, hence the interactive call where you enter your mail and password. After you did so, does it hang or does `Press any key to continue...` appear which lets `dietpi-config` menu reappear?

Once idea I have is that `noip2 -C`, if credentials are correct, as well goes into background to run as regular daemon. If we then start the service, there would be two instances, probably causing the issue. A reboot should solve it. If this is really true, we need to manually kill the spawned noip2 process before starting the service, or add `ExecStartPre=/usr/bin/killall -w noip2` to the service so that is always kills all probably running processes before starting itself. However I doubt that actually, since the `noip2 -C` process in your case is still running as root while it spawns the actual daemon executed with user `nobody`.

MichaIng, 2020-06-01:
Marking as closed due to missing reply. Feel free to reopen if the issue is still present.

closed 2020-06-01 by MichaIng