MichaIng/DietPi #3206: PREP_SYSTEM_FOR_DIETPI.sh failing on G_AGP dbus dhcpcd5 mountall

dtm2001, 2019-10-25:
### Creating a bug report/issue

#### Required Information
#### Details:
- Date           | Fri 25 Oct 20:32:43 BST 2019
- Bug report     | 
- DietPi version | v6.26.3 (MichaIng/master)
- Image creator  | N/A
- Pre-image      | N/A
- SBC device     | NULL (index=21)
- Kernel version | Linux ukdevlinux2 4.19.0-6-amd64 #1 SMP Debian 4.19.67-2+deb10u1 (2019-09-20) x86_64 GNU/Linux
- Distro         | buster (index=5)
- Command        | G_AGP dbus dhcpcd5 mountall
- Exit code      | 100
- Software title | DietPi-PREP

#### Steps to reproduce:
I have just done a clean install of Debian onto a x86_64 Intel PC
Run the script to get PREP_SYSTEM_FOR_DIETPI, per the website instructions:
```
wget https://raw.githubusercontent.com/MichaIng/DietPi/master/PREP_SYSTEM_FOR_DIETPI.sh -O PREP_SYSTEM_FOR_DIETPI.sh
chmod +x PREP_SYSTEM_FOR_DIETPI.sh
./PREP_SYSTEM_FOR_DIETPI.sh
```

#### Expected behaviour:
Nice, clean install of dietpi

#### Actual behaviour:
G_AGP dbus dhcpcd5 mountall fails.
I think this could be because mountall doesn't seem to exist in Buster ("apt-get install mountall" returns "E: Unable to locate package mountall"

#### Extra details:

- ...

#### Additional logs:
```
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

MichaIng, 2019-10-28:
@dtm2001 
Many thanks for your report.

> I think this could be because mountall doesn't seem to exist in Buster ("apt-get install mountall" returns "E: Unable to locate package mountall"

Jep, on Debian it is not even available on Stretch, however on Raspbian Stretch it IS available and depends on "plymouth", which can be quite annoying for first boot. And as a strange matter of fact, even when doing `apt-mark auto mountall && apt autoremove --purge`, it stays both installed, the same for dbus and dhcpcd5 packages, which is why we purge then explicitly.

However `G_AGP` should not be able to fail on non-existent packages. It checks input against the list of installed packages, and hands only those over to APT, e.g.:
```
2019-10-28 13:18:06 root@micha:/etc# G_AGP asfasd
[ INFO ] APT removal for: asfasd, please wait...
[ INFO ] None of the requested packages are currently installed. Aborting...
[  OK  ] G_AGP asfasd
```

To check the issue in your case, please paste the full output of:
```
apt purge dbus
apt purge dhcpcd5
apt purge mountall
```
- One by one since the command fails immediately if one package is not installed and we want to see which is and for which reason the resolver fails.

dtm2001, 2019-10-30:
@MichaIng, many thanks for your reply.

Before I forget, when I first boot debian and try to run the PREP script I get the error:
line 129: dpkg-reconfigure : command not found

I have to first update update $PATH to include /usr/local/sbin:/usr/sbin:/sbin
THEN run `dpkg --configure -a`

Apologies that I forgot to mention this before!

So, running the three commands you suggested:

> apt purge dbus 

FAILS:
```
root@ukdevlinux2:/home/foo# apt purge dbus
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgtk-3-0 : Depends: libgtk-3-common (>= 3.24.5-1) but it is not going to be installed
 libsoup2.4-1 : Depends: glib-networking (>= 2.32.0) but it is not going to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
root@ukdevlinux2:/home/foo# 

```

> apt purge dhcpcd5

 SUCCESS:
```
The following packages will be REMOVED:
  dhcpcd5*
0 upgraded, 0 newly installed, 1 to remove and 3 not upgraded.
After this operation, 404 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 140112 files and directories currently installed.)
Removing dhcpcd5 (7.1.0-2) ...
Processing triggers for man-db (2.8.5-2) ...
(Reading database ... 140087 files and directories currently installed.)
Purging configuration files for dhcpcd5 (7.1.0-2) ...
Processing triggers for systemd (241-7~deb10u1) ...
```

> apt purge mountall

NOT FOUND:
```
root@ukdevlinux2:/home/foo# apt purge mountall
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package mountall
root@ukdevlinux2:/home/foo# 

```

MichaIng, 2019-10-30:
@dtm2001 
> I have to first update update $PATH to include /usr/local/sbin:/usr/sbin:/sbin

Which Debian image did you install first? Very strange thing that these default binary paths are not included in the $PATH variable. For non-root users this would be thinkable, but for root + sudo, actually the `/etc/bash.bashrc` should set the $PATH variable correctly for all bash shells.

But nevertheless, since we do not know the system where users run DietPi-PREP on, it is a good idea to set the PATH variable right from the beginning.

> Some packages could not be installed.

Okay obviously there is a package that depends on either dbus or another package that depends on these libraries or on them directly. I am a bid confused that the automated purge does not simply purges all dependants as well, especially since we allow "recommends" auto-removal before doing this G_AGP call: https://github.com/MichaIng/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh#L696-L699
Probably one of these packages has status "important", and our apt-get flags does not include automated removal of those. I removed "--allow-remove-essential" some time ago, but actually there is a difference between "important" and "essential" packages. The latter are always expected and other packages can depend on them without listing them as dependencies. Important packages are, depending on requirements, not too important after all :wink:, especially they can (or should) not be expected by other packages without listing them as dependencies explicitely.

However, probably "--allow-remove-essential" indeed includes "important" packages as well. To test:
```
G_AGP --allow-remove-essential dbus
```

The issue should be solved after we found time to rework DietPi-PREP so that required packages are installed first and all others are autoremove/purged afterwards. Currently its the other way round which in cases lead to broken network or other missing core functionalities, if those were enabled via one package (set) while DietPi uses another one.

To workaround the issue, you can simply remove from the line: https://github.com/MichaIng/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh#L953
After the script went through, it is either autoremoved already or you should be able to manually purge it then.

MichaIng, 2019-10-30:
PATH issue resolved with: https://github.com/MichaIng/DietPi/commit/c93e618e7b40ed1ecb52f23fd9db5fd349a671f2
Many thanks for reporting :+1:, a totally reasonable step to be compatible with unexpected system setups.

dtm2001, 2019-11-06:
@MichaIng,
Just wanted to confirm that removing that line from the script allowed the install to complete successfully.
Many thanks for your help!

MichaIng, 2019-11-06:
@dtm2001 
You mean removing the `G_AGP` line? Was dbus and dhcpcd5 then autoremoved later or are they still present on the final image/system?

dtm2001, 2019-11-08:
@MichaIng 
Yes, sorry, removing the `G_AGP` line allows the install to continue.
I have since installed other packages from dietpi-software that may have re-installed packages, but my system now still has `dbus` installed, but `dhcpd5` has been removed.
I am happy with the build as it currently is, but do you want me to re-test the install and check the package status immediately following install?

MichaIng, 2019-11-08:
@dtm2001 
Not necessary to run further tests. I guess the planned DietPi-PREP rework will fix such issues all together: https://github.com/MichaIng/DietPi/projects/3

MichaIng, 2020-02-04:
I faced the same issue on ARMbian desktop images. I fixed it by actively purging all xserver and desktop related packages as well: https://github.com/MichaIng/DietPi/commit/98fbba104973f75e95f0523499f4ede080a4b822
Cleaner fix is on ToDo, but I mark this as closed for now.

closed 2020-02-04 by MichaIng