MichaIng/DietPi #3207: [v6.23.3] ALSA Output Info (DietPi-Justboom) returns error

maartenlangeveld, 2019-10-26:
```

┌─────────────────────────────────────────────────┤ DietPi-JustBoom ├──────────────────────────────────────────────────┐
│ Tweak your High-End Audiophile JustBoom (and others) sound card:                                                     │
│                                                                                                                      │
│                                           ●─ Global Options ─────────────────────────────●                           │
│                          ALSA Mixer       : Launches ALSA Mixer, tweak sound card settings                           │
│                          ALSA Output Info : View info on current ALSA playback stream                                │
│                          EQ Enable        : [Off] | Select to toggle EQ                                              │
│                          Reset            : Reset all options to defaults                                            │
│                                           ●─ MPD Specific ───────────────────────────────●                           │
│                          Output Frequency :  Hz | Select to change                                                   │
│                          Output BitRate   :  Bit | Select to change                                                  │
│                          Audio Buffer     : 2048 Bytes | Select to change                                            │
│                          SOXR Enable      : [Off] | Select to toggle SOXR resampling                                 │
│                                                                                                                      │
│                                                                                                                      │
                                   <Ok>                                      <Exit>                                    │
│                                                                                                                      │
          └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘

```

```
 DietPi-JustBoom
─────────────────────────────────────────────────────
 Mode: ALSA Output Info

/DietPi/dietpi/misc/dietpi-justboom: line 305: /proc/asound/card*/pcm0p/sub0/hw_params: ambiguous redirect

[  OK  ] DietPi-JustBoom | Completed, press any key to exit...

Press any key to exit...
```

MichaIng, 2019-10-28:
@maartenlangeveld 
Many thanks for your report.

Indeed there is a syntax. The line:
```
local stream_result=$(</proc/asound/card*/pcm0p/sub0/hw_params)
```
needs to be replaced by:
```
local stream_result=$(cat /proc/asound/card*/pcm0p/sub0/hw_params)
```
in `/DietPi/dietpi/misc/dietpi-justboom`.
Redirects do not support globbing.

Hmm IMO this is not THAT nice anyway. "pcm0p/sub0" does not necessarily match the device + sub device the audio is played on, does not? And if there were multiple audio devices, it would be nice to not only concatenate their content but separate it to show which content is from which device, right?

So my idea is then:
```
local stream_result
for i in /proc/asound/card[0-9]/pcm[0-9]p/sub[0-9]/hw_params
do
	stream_result+="Audio device: $i\n$(<$i)\n\n"
done
```

maartenlangeveld, 2019-10-28:
@MichaIng 

Thanks! Just wondering, how was it coded in previous version (6.25.3) ?

MichaIng, 2019-10-28:
Exactly as it is with this commit. Was my mistake since usually cat is an unnecessary overhead (external commands), so I replace it with redirect where possible. Here the error slipped through since I didn't run an audio source when testing the script I guess...

closed 2020-01-01 by MichaIng