MichaIng/DietPi #3210: Updating Kernel in NanoPC-T4 with dietpi released image

ramgopalkota, 2019-10-28:
I need to understand the kernel update procedure for NanoPC-T4(RK3399)
I change the kernel config options and DTS file. Recompile the kernel and DTS files.
Created resource.img and kernel.img .. replaced the files in the dietpi released zip file.
[image]

Kernel is still old .. Is there anything I have to do ? Please help ...

Ramgopal Kota

ramgopalkota, 2019-10-28:
I am new to the Android world and I am familar with older Kernel / DTB upgrade procedure where the zImage and DTB are copied to /Boot folder and the Board has new Kernel / DTB.

I need to upgrade since I want to add external RTC (DS3231) , but the filesystem and DTB does not have entries hence I decided to build the kernel and DTB.

I downloaded the latest kernel 4.4.179 and also updated the DTB to indicate the i2c rtc device @ 0x68.

Following is done:

Downloaded DietPi_NanoPCT4-ARMv8-Stretch.7z and using Android Tool I have updated the Board and Boots OK.
Compiled the Linux source from Friendlyarm GIT as explained in thier wiki
git clone https://github.com/friendlyarm/kernel-rockchip --depth 1 -b nanopi4-linux-v4.4.y kernel-rockchip
cd kernel-rockchip
export PATH=/opt/FriendlyARM/toolchain/6.4-aarch64/bin/:$PATH
make ARCH=arm64 nanopi4_linux_defconfig
make ARCH=arm64 nanopi4-images nanopi4-bootimg
This created kernel.img , resource.img & boot.img ..
I replaced with the ones in DietPi_NanoPCT4-ARMv8-Stretch .. The Board does not Boot.
If I dont replace boot.img file , board boots but the kernel is 4.4.154 .. So Kernel is in boot.img file.

From some googling there is a link I found http://opensource.rock-chips.com/wiki_Rockchip_Kernel .. This also did not work.
As per the link , the boot.img file should be programmed @ 0x8000 but in the parameter.txt file it is @ 0x1E000.

Please help me prepare the boot.img file so that I can upgrade the kernel and proceed further.

MichaIng, 2019-10-28:
@ramgopalkota 
Hmm, to be true my knowledge about these Android tools is minimal as well. Actually my idea was to provide a new image soon, based on Debian Buster with updated kernel as Clonezilla-based installer. Although not yet sure if/how this will work.

AFAIK boot.img indeed is the image for the whole boot partition, hence kernel + bootloader is included already and the kernel.img is probably for /lib/modules only and/or for personal resource. If this is true and it overwrites the boot partition including config files etc, then the UUID/PARTUUID of this partition most likely changes as well. So at least `/etc/fstab` (within root partition image) needs to be adjusted then, so that /boot mount UUID (or PARTUUID) matches the new boot partition.

ramgopalkota, 2019-10-28:
Thanks for the reply .. I found the issue. The issue is with the
partitioning. Once I used latest tool it worked.
One more question , I am using LXDE and it is taking 2-3 minutes after
kernel logo appears. Till that time (2-3 minutes) the screen is black.
Is there anyway to continue to use the logo till the X11 starts ? I tried
splash_disable=0 ..

MichaIng, 2019-10-29:
@ramgopalkota 
2-3 minutes until desktop is loaded? That is indeed very long, should not take more than 10 seconds usually. Do you use autologin as root or other user or LightDM?

Not sure about this particular splash screen, plymouth AFAIK hides boot logs until `systemd-user-sessions.service` has been reached, so logins become possibly and gettys are started. But that does not mean that all boot scripts have finished. `journalctl` should allow you to check the boot process, or `systemd-analyze blame` for a more compact overview.

ramgopalkota, 2019-10-29:
I use custom auto-start with xinit chromium in kiosk mode.
I am not getting any boot logs on the HDMI just blank screen until the
chromium starts.

I will check the systemd-analyze blame and update.

Ramgopal Kota

MichaIng, 2019-10-29:
@ramgopalkota 
Damn thing, the contained .img files are a special kind that one cannot simply mount as loop devices or extract via wrapper tools. So no idea how it finally looks like, where the kernel cmdline is located to adjust boot output, console handling and such 🤔.

But at least you can check the cmdline, which probably gives a hint: `cat /proc/cmdline`

MichaIng, 2020-02-11:
New Buster-based image is available: https://github.com/MichaIng/DietPi/issues/2979
It's an SDcard image, but it should be possible to flash it to eMMC directly via adapter.

closed 2020-05-03 by MichaIng