MichaIng/DietPi #3213: Error installing influx db and grafana

TBail, 2019-10-31:
### Creating a bug report/issue

Here are the blue lines

#### Details:
- Date           | Thu 31 Oct 17:56:05 CET 2019
- Bug report     | 7ddfeeae-12fb-450b-aec2-04330add1a55
- DietPi version | v6.26.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi B (armv6l) (index=1)
- Kernel version | Linux wireguard 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux
- Distro         | buster (index=5)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. Selected software influx and grafana by using the search 
2. Startet the installation process using the menu

#### Expected behaviour:

- .Software should be installed

#### Actual behaviour:

- Currently i dont now havent not checked

#### Additional logs:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Ign:2 https://dl.bintray.com/fg2it/deb-rpi-1b buster InRelease
Err:3 https://dl.bintray.com/fg2it/deb-rpi-1b buster Release
  404  Not Found [IP: 18.184.126.218 443]
Hit:4 https://repos.influxdata.com/debian buster InRelease
Hit:5 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
E: The repository 'https://dl.bintray.com/fg2it/deb-rpi-1b buster Release' does not have a Release file.
```

MichaIng, 2019-11-01:
@TBail 
Many thanks for your report.

Indeed the 3rd party bintray repo that we use to pull Grafana on ARMv6 models, does not contain a Buster branch. I just rechecked and actually it ships a totally outdated version.

The official Grafana APT repo seems to not have an ARMv6-compatible branch, but I am not 100% sure since they don't allow to browser their repo dir structure 😉. Will recheck/ask them. Because they DO ship updated ARMv6 packages.

Please do the following to install the newest package (the bintray repo ships v5.1.4 to compare):
```
rm /etc/apt/sources.list.d/grafana.list
apt-key del '8756 C4F7 65C9 AC3C B6B8  5D62 379C E192 D401 AB61'
cd /tmp
wget https://dl.grafana.com/oss/release/grafana-rpi_6.4.3_armhf.deb
dpkg -i grafana-rpi_6.4.3_armhf.deb
apt -f install
rm grafana-rpi_6.4.3_armhf.deb
```

To apply DietPi setup to have userdata in /mnt/dietpi_userdata/grafana:
```
mv /var/lib/grafana /mnt/dietpi_userdata/
ln -s /mnt/dietpi_userdata/grafana /var/lib/grafana
chown -R grafana:grafana /mnt/dietpi_userdata/grafana
```

To adjust web UI port and password: `/etc/grafana/grafana.ini`
____________
I applied this as well for v6.27: https://github.com/MichaIng/DietPi/pull/3214

What do you think about an automated reinstall on v6.27 patch to have the above updated package applied to all ARMv6 systems? Existing data, configs, permissions etc are preserved, hence we would only remove the obsolete/outdated bintray repo and install the package as above on top.

SteluHH, 2019-11-10:
Hi! Just started using dietpi! Awesome work guys and thanks.

Apparently the same issue applies for installing the GitClient using `dietpi-software`: 

```
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Ign:2 https://dl.bintray.com/fg2it/deb-rpi-1b buster InRelease
Err:3 https://dl.bintray.com/fg2it/deb-rpi-1b buster Release
  404  Not Found [IP: 35.156.31.209 443]
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
E: The repository 'https://dl.bintray.com/fg2it/deb-rpi-1b buster Release' does not have a Release file.
```
Does this issue relate?

Thanks in advance,

Stefano

MichaIng, 2019-11-10:
@SteluHH
Many thanks for your feedback.

Jep the issue is just the same, although strange that it did not show up earlier. When removing the grafana.list as I mentioned above, the Git install should work fine as well. So the error indeed has nothing to do with Git but the symptom of invalid Grafana repo just shows up on any APT update call.

SteluHH, 2019-11-10:
Well I installed GIT using apt-get then. 
However Grafana just worked fine with your script :-)

Once again thank you so much for the great work.

[image]

MichaIng, 2019-11-11:
@SteluHH 
> Well I installed GIT using apt-get then.

Jep, dietpi-software does nothing else, no custom changes on top if the apt-get install. It's listed only in "Software Additional" for completeness and since it is required for some other major titles like Pi-hole.

> However Grafana just worked fine with your script :-)

Ha, I just found that Grafana is available via Raspbian repo as well (opposed to Debian):
```
2019-11-11 12:04:53 root@micha:/var/log# apt search grafana
Sorting... Done
Full Text Search... Done
grafana/testing 2.6.0+dfsg-3+b1 armhf
  feature rich metrics dashboard and graph editor

grafana-data/testing 2.6.0+dfsg-3 all
  feature rich metrics dashboard and graph editor - data files

grafana-zabbix/testing 2.5.1-1 all
  Zabbix datasource for Grafana
```
This must be the reason why the install then still succeeds. However the version 2.6.0 is VERY outdated: http://raspbian.raspberrypi.org/raspbian/pool/main/g/grafana/

As long as it works, okay, however you can easily update to the current version 6.4.3 with the first set of commands I provided above 😉: https://github.com/MichaIng/DietPi/issues/3213#issuecomment-548788294

SteluHH, 2019-11-11:
Interestingly, I AM at version 6 with your script :)

MichaIng, 2019-11-11:
@SteluHH 
Ah sorry, you meant the commands above then, I misunderstood. Okay makes sense then and great to have another successful test of the method that we use implemented for DietPi v6.27.

closed 2020-01-01 by MichaIng