MichaIng/DietPi #3216: Shairport Update

kmplngj, 2019-11-02:
Hi,

would it be possible to get a Shairport Update? 

There are some nice bugfixes and support for MQTT in the new release.

Thanks
Jan

MichaIng, 2019-11-02:
@kmplngj 
Many thanks for your request. I'll see if I find time to dig into this compilation. Which SBC do you own, hence which architecture are you able to test the binary on?

kmplngj, 2019-11-03:
Thanks. I could test on the RaspberryPi 3.

MichaIng, 2020-03-21:
@kmplngj 
I was in mood to compile them as package, ready for testing: #3428
For RPi Raspbian Buster:
```
cd /tmp
wget https://dietpi.com/downloads/binaries/buster/shairport-sync_armv6l.deb
apt-get -y install ./shairport-sync_armv6l.deb
rm shairport-sync_armv6l.deb
```

closed 2020-05-03 by MichaIng

MichaIng, 2020-10-09:
MQTT support has been added with the latest package updates. Install procedure same as above.