MichaIng/DietPi #3217: Rpi4 - Updating firmware

trueaspects, 2019-11-04:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- ...
I could not use rpi-eeprom-update to update firmware on rpi 4. 
#### Describe the solution you'd like:

- ...
Be able to use rpi-eeprom-update to upgrade firmware on rip 4. 
#### Describe alternatives you've considered:

- ...

#### Additional context

- ...

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2019-11-05:
@trueaspects 
Many thanks for providing this info. I was not aware of the EEPROM on RPi4 yet, of course we should have the `rpi-eeprom` package installed on RPi4 then by default.

I am not yet sure how to handle this in regards to lower RPi models:
- `rpi-eeprom` installs a service that runs on every boot and checks for EEPROM updates. This of course, as well as the whole package, is an unnecessary overhead on all other RPi models.
- We could keep the package installed but mask the service and run the updates from within our own boot scripts.
- We could check the RPi model and purge `rpi-eeprom` on all non-RPi4 models, but this breaks the possibility to switch the SDcard without loosing full features. But probably it is enough to check and install the package again when choosing the "Update firmware" option from dietpi-config on RPi4?
- A strong reason to not have it installed by default, but targeted when RPi4 is detected, is the `python` dependency... This installs outdated EOL Python 2, rarely required for anything else, taking unnecessary disk space and package upgrade effort...

> I was able to install eeprom, but when I run rpi-eeprom-update I get the following:

How exactly did you install it?

> Rpi4 now updates firmware differently from prior Rpi as it is now directly on the Rpi rather than on the SD card. It would be great to be able to update firmware from dietpi.

Note that this is only about the bootloader, all other firmware, kernel, device tree, overlays, VC drivers/libs/binaries are still installed the same way.

MichaIng, 2019-11-05:
@trueaspects 
> However, I have a feeling that many, if not all, future RPi models may be using rpi-eeprom.

Jep, I am pretty sure about that.

> Perhaps the compromise should be that so long as we can install rpi-eeprom and run it on our own, it doesn't need to be pre-installed since all other RPi models do not need it, and it uses outdated dependencies.

I will open an issue on RPi repo to add `binutils` as package dependency, so installing `rpi-eeprom` and running the update will then work without the need to manually install `binutils` as well.

I think its okay to install it on firstrun setup, if done on an RPi4, as well as when running firmware update from dietpi-config manu. For RPi4 it is important enough IMO.

MichaIng, 2019-11-10:
PR up: #3222

MichaIng, 2019-11-11:
@trueaspects 
Do you know if the `rpi-eeprom` package is pre-installed and automatic update service on boot is enabled by default on fresh Raspbian?

I am trying to evaluate how save this flash/update process can be considered. If this is done on boot by default on RPi4, then it can be considered as very safe and we can apply it as well non-interactively. However if it is not, then I am thinking if we should never do this non-interactively but instead give RPi4 users a hint how to run EEPROM updates but to read into the topic first to know how to do emergency recovery etc.

trueaspects, 2019-11-11:
@MichaIng 

If I recall correctly, I installed/used raspbian lite and rpi-eeprom was already installed.  However, I do not recall whether the update service on boot was enabled by default, but I would guess yes, because all the guides tell you how to disable it and then re-enable it, rather than how to enable it.

MichaIng, 2019-11-12:
@trueaspects 
Just rechecked the Raspbian image and indeed this exists:
`/etc/systemd/system/multi-user.target.wants/rpi-eeprom-update.service` => `/lib/systemd/system/rpi-eeprom-update.service`

So the service is enabled by default and we can assume this being safe to run non-interactively. Hence I will add it to firstrun installs and next dietpi-update patch to be installed automatically on RPi4. Especially the VL805 USB drivers are subject ob certain energy/speed balance/rework which hence affects heat emissions as well, so nothing that I am keen to have users run outdated versions.

PR #3222 (2019-11-12): RPi4 | Implement EEPROM firmware update
**Status**: Ready
- [x] DietPi-Set_hardware: Add "rpi-eeprom" option
- [x] DietPi-Software | Install rpi-eeprom automatically if RPi4 is detected or give user a choice by opening dietpi-config or give a hint only?
- [x] DietPi-Patch | Install rpi-eeprom automatically if RPi4 is detected or give user a choice by opening dietpi-config or give a hint only?
- [x] Changelog

Fixes: #3217

**Commit list/description**:
+ DietPi-Config | RPi4: Add a menu entry to install and update the RPi4 EEPROM firmware
+ DietPi-Set_hardware | Add RPi4 EEPROM update command " [... cut]
closed 2020-01-01 by MichaIng