MichaIng/DietPi #3219: Pyhton 3.7 for Home Assistant

sdomotica, 2019-11-04:
The new Home Assistant release 100.2 require Python 3.7, python 3.6 will be deprecated on the next 
December, 15
Bye
Sandro

MichaIng, 2019-11-05:
@sdomotica 
Many thanks for providing this info. We'll move to current Python for Home Assistant with next DietPi release.

MichaIng, 2019-11-06:
@sdomotica 
Any specific recommendation for the Python version? I would use 3.7.5 which seems to be the most current. HA works well with this?

Would be actually nice to install the latest version automatically, as long as HA supports this, but there seem to be no native pyenv command/option to pull the latest, besides scraping the versions --list.

sdomotica, 2019-11-06:
@MichaIng No specific request, I'm using 3.7.3 manually upgraded

MichaIng, 2019-11-06:
@sdomotica 
Okay, since we pull a specific version for the pyenv environment, I used most current 3.7.5 now. Will run some tests if this works fine, otherwise revert to 3.7.3 as this is current Debian Buster default as well.

Switching to a non-pyenv install method or automatically pull the lastest available Python version is another point of discussion, but out of scope for DietPi v6.27.

MichaIng, 2019-11-06:
Hmm, facing this issue when testing on Stretch VM:
```
2019-11-06 23:14:59 INFO (MainThread) [homeassistant.setup] Setting up lovelace
2019-11-06 23:14:59 ERROR (MainThread) [homeassistant.setup] Error during setup of component lovelace
Traceback (most recent call last):
  File "/home/homeassistant/.pyenv/versions/3.7.5/envs/homeassistant-3.7.5/lib/python3.7/site-packages/homeassistant/setup.py", line 172, in _async_setup_compon                                                                             ent
    hass, processed_config
  File "/home/homeassistant/.pyenv/versions/3.7.5/envs/homeassistant-3.7.5/lib/python3.7/site-packages/homeassistant/components/lovelace/__init__.py", line 66,                                                                              in async_setup
    hass.components.frontend.async_register_built_in_panel(
  File "/home/homeassistant/.pyenv/versions/3.7.5/envs/homeassistant-3.7.5/lib/python3.7/site-packages/homeassistant/loader.py", line 421, in __getattr__
    component: Optional[ModuleType] = integration.get_component()
  File "/home/homeassistant/.pyenv/versions/3.7.5/envs/homeassistant-3.7.5/lib/python3.7/site-packages/homeassistant/loader.py", line 223, in get_component
    cache[self.domain] = importlib.import_module(self.pkg_path)
  File "/home/homeassistant/.pyenv/versions/3.7.5/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/homeassistant/.pyenv/versions/3.7.5/envs/homeassistant-3.7.5/lib/python3.7/site-packages/homeassistant/components/frontend/__init__.py", line 10,                                                                              in <module>
    import hass_frontend
ModuleNotFoundError: No module named 'hass_frontend'
2019-11-06 23:14:59 INFO (SyncWorker_8) [homeassistant.loader] Loaded binary_sensor from homeassistant.components.binary_sensor
2019-11-06 23:14:59 INFO (MainThread) [homeassistant.setup] Setting up history
2019-11-06 23:14:59 ERROR (MainThread) [homeassistant.setup] Error during setup of component history
Traceback (most recent call last):
  File "/home/homeassistant/.pyenv/versions/3.7.5/envs/homeassistant-3.7.5/lib/python3.7/site-packages/homeassistant/setup.py", line 172, in _async_setup_compon                                                                             ent
    hass, processed_config
  File "/home/homeassistant/.pyenv/versions/3.7.5/envs/homeassistant-3.7.5/lib/python3.7/site-packages/homeassistant/components/history/__init__.py", line 283,                                                                              in async_setup
    hass.components.frontend.async_register_built_in_panel(
  File "/home/homeassistant/.pyenv/versions/3.7.5/envs/homeassistant-3.7.5/lib/python3.7/site-packages/homeassistant/loader.py", line 421, in __getattr__
    component: Optional[ModuleType] = integration.get_component()
  File "/home/homeassistant/.pyenv/versions/3.7.5/envs/homeassistant-3.7.5/lib/python3.7/site-packages/homeassistant/loader.py", line 223, in get_component
    cache[self.domain] = importlib.import_module(self.pkg_path)
  File "/home/homeassistant/.pyenv/versions/3.7.5/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/homeassistant/.pyenv/versions/3.7.5/envs/homeassistant-3.7.5/lib/python3.7/site-packages/homeassistant/components/frontend/__init__.py", line 10,                                                                              in <module>
    import hass_frontend
ModuleNotFoundError: No module named 'hass_frontend'
2019-11-06 23:14:59 INFO (MainThread) [homeassistant.setup] Setting up websocket_api
2019-11-06 23:14:59 INFO (MainThread) [homeassistant.setup] Setup of domain websocket_api took 0.0 seconds.
2019-11-06 23:14:59 INFO (MainThread) [homeassistant.setup] Setting up config
2019-11-06 23:14:59 ERROR (MainThread) [homeassistant.setup] Error during setup of component config
Traceback (most recent call last):
  File "/home/homeassistant/.pyenv/versions/3.7.5/envs/homeassistant-3.7.5/lib/python3.7/site-packages/homeassistant/setup.py", line 172, in _async_setup_compon                                                                             ent
    hass, processed_config
  File "/home/homeassistant/.pyenv/versions/3.7.5/envs/homeassistant-3.7.5/lib/python3.7/site-packages/homeassistant/components/config/__init__.py", line 34, in                                                                              async_setup
[... cut ...]
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/homeassistant/.pyenv/versions/3.7.5/envs/homeassistant-3.7.5/lib/python3.7/site-packages/homeassistant/components/frontend/__init__.py", line 10,                                                                              in <module>
    import hass_frontend
ModuleNotFoundError: No module named 'hass_frontend'
```
Result is that web UI does not work: `404: Not Found`
Same was reported here: https://community.home-assistant.io/t/error-setting-up-frontend/42448/5

sdomotica, 2019-11-06:
I solved installing before a previous version of HA 0.100.0 the the last.
I think is a 0.101 issue

MichaIng, 2019-11-06:
Will try on Buster and Bullseye and with Python 3.7.3, lets see...

huettenwirt, 2019-12-31:
I'm having problems reinstalling home assistant using the current beta:

DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing Home Assistant: open-source home automation platform

[ INFO ] DietPi-Software | ha_user: homeassistant
[ INFO ] DietPi-Software | ha_home: /home/homeassistant
[ INFO ] DietPi-Software | ha_pyenv_activation: source /home/homeassistant/pyenv-activate.sh
[ INFO ] DietPi-Software | ha_python_version: 3.8.0
[  OK  ] DietPi-Software | Connection test: https://github.com/pyenv/pyenv/archive/master.tar.gz
[ INFO ] DietPi-Software | G_THREAD_START_0 | wget https://github.com/pyenv/pyenv/archive/master.tar.gz -O 157.tar
[ INFO ] DietPi-Software | APT installation for: gcc libc6-dev make zlib1g-dev libbz2-dev libreadline-dev libssl-dev libsqlite3-dev libffi-dev, please wait...
[  OK  ] DietPi-Software | G_AGI gcc libc6-dev make zlib1g-dev libbz2-dev libreadline-dev libssl-dev libsqlite3-dev libffi-dev
[  OK  ] DietPi-Software | G_THREAD: All threads finished
[  OK  ] DietPi-Software | tar xf 157.tar
[  OK  ] DietPi-Software | rm 157.tar
dash: 2: source: not found
dash: 3: pyenv: not found
dash: 4: pyenv: not found
Collecting pip
  Downloading https://files.pythonhosted.org/packages/00/b6/9cfa56b4081ad13874b0c6f96af8ce16cfbc1cb06bedf8e9164ce5551ec1/pip-19.3.1-py2.py3-none-any.whl (1.4MB)
    100% |████████████████████████████████| 1.4MB 197kB/s
Installing collected packages: pip
Successfully installed pip-19.3.1
Traceback (most recent call last):
  File "/usr/bin/pip3", line 9, in <module>
    from pip import main
ImportError: cannot import name 'main'

Any ideas why the virtual environment is not created?

Best regards and a happy new year!
Huettenwirt

MichaIng, 2020-01-01:
@huettenwirt 
Many thanks for reporting. Indeed a dumb bug, fixed with: https://github.com/MichaIng/DietPi/commit/fa0a9ffcd38135dc08cd7b859675af31ff5a26bd

```
sed -i '/local ha_pyenv_activation/s/source/\./' /DietPi/dietpi/dietpi-software
dietpi-software install 157
```

closed 2020-01-01 by MichaIng

huettenwirt, 2020-01-03:
@MichaIng 
Thanks for replying!

I got HA installed, but I can't start HA. The status tells me the following:

`[2020-01-03 14:52:22] --- Home Assistant is starting, please wait...
● home-assistant.service - Home Assistant (DietPi)
   Loaded: loaded (/etc/systemd/system/home-assistant.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-01-03 14:51:50 CET; 32s ago
  Process: 19688 ExecStart=/home/homeassistant/homeassistant-start.sh (code=exited, status=1/FAILURE)
 Main PID: 19688 (code=exited, status=1/FAILURE)

Jan 03 14:51:49 RaspberryPi homeassistant-start.sh[19688]:   File "/home/homeassistant/.pyenv/versions/3.8.0/lib/python3.8/site-packages/homeassistant/bootstrap.py", line 135, in async_from_config_fil
e
Jan 03 14:51:49 RaspberryPi homeassistant-start.sh[19688]:     await hass.async_add_executor_job(conf_util.process_ha_config_upgrade, hass)
Jan 03 14:51:49 RaspberryPi homeassistant-start.sh[19688]:   File "/home/homeassistant/.pyenv/versions/3.8.0/lib/python3.8/concurrent/futures/thread.py", line 57, in run
Jan 03 14:51:49 RaspberryPi homeassistant-start.sh[19688]:     result = self.fn(*self.args, **self.kwargs)
Jan 03 14:51:49 RaspberryPi homeassistant-start.sh[19688]:   File "/home/homeassistant/.pyenv/versions/3.8.0/lib/python3.8/site-packages/homeassistant/config.py", line 409, in process_ha_config_upgrad
e
Jan 03 14:51:49 RaspberryPi homeassistant-start.sh[19688]:     with open(version_path, "wt") as outp:
Jan 03 14:51:49 RaspberryPi homeassistant-start.sh[19688]: PermissionError: [Errno 13] Permission denied: '/mnt/dietpi_userdata/homeassistant/.HA_VERSION'
Jan 03 14:51:50 RaspberryPi systemd[1]: home-assistant.service: Main process exited, code=exited, status=1/FAILURE
Jan 03 14:51:50 RaspberryPi systemd[1]: home-assistant.service: Unit entered failed state.
Jan 03 14:51:50 RaspberryPi systemd[1]: home-assistant.service: Failed with result 'exit-code'.

`

Did any permissions change since last release? It seems HA can't use the HA user directory...?

Thx for any hint on this and best regards!
Huettenwirt

MichaIng, 2020-01-03:
@huettenwirt 
Actually permissions did not change, I guess something was skipped due to first failed install attempt:
Please try to resolve with:
```
chown -R homeassistant:homeassistant /mnt/dietpi_userdata/homeassistant
```
I'll go through the install code to assure this is done in every case before first service start.