MichaIng/DietPi #3224: NanoPi NEO2, Unable to exit from DietPi-Software

jryx0, 2019-11-12:
After first login, the sytem automatic run DietPi-Software, but couldn't exit when you select exit.

MichaIng, 2019-11-12:
@jryx0 
Many thanks for your report.

This is intended, please select "Install" at the bottom to finish firstrun setup.

There is a known issue with an older DietPi version that the DietPi-Software menu is too small so that one does not see the "Install" entry at the bottom. But you can "scroll" down and select it via arrow buttons.

jryx0, 2019-11-13:
Thanks for you reply.
Yes, I saw the "install" entry.   I selected some softwares in "software Optimizied" entry, and  selected "intall", the screen flash and nothing happened. The DietPi-Software menu was till in there. I also tried  to select other entry and nothing happened. At last I selected "exit", the DietPi-Software menu could not exit.
Is there any options to disable  the DietPi-Software menu ?

MichaIng, 2019-11-13:
@jryx0 
On firstrun you don't need to select any software title btw, but you can hit "Install" as well without anything selected.

Hmm, that is strange. You have SSH up, right? You can login on a second SSH session while the first one stays with dietpi-software up. From the second session you can then kill it: `killall -w dietpi-software` or at best `htop` and manually nagivate (arrow buttons) to the related processes and kill them with F9.
- When checking the other session then, probably below the whiptail prompt you'll find some console output that might give a hint why selecting "Install" did not work as expected.

Rerun dietpi-update, just to be failsafe: `dietpi-update -1`
Assure that everything works fine (no error message inside), then reboot. DietPi-Software will automatically spawn again when you login, but I hope this time it works. There are no flaws in the code that I am aware of, nor any other reports, hence if the update went well, it must work.

jryx0, 2019-11-15:
I found why I had this problem. I used serial to connect the Dietpi,  On firstrun the system stop at "please login to contiune", the keyborad had no respose and the system seamed halted.  I did not login on a second SSH session but power off (no input poweroff) and then power on, the DietPi_Software came into endless mode.

MichaIng, 2019-11-15:
@jryx0 
Hmm okay, actually the serial TTYs should be ready for login on first boot. You mean on serial console the keyboard/input was not possible? Most likely the serial getty was not initialised as expected. I just checked the image file and actually the used ttyS0 is enabled for login. But I have no experience with serial consoles to be true, e.g. if there can be input issues externally as well, even that server side everything is setup correctly.

If you poweroff/on then, it "should" boot just the same, having a login prompt on any console you connect (local, serial or SSH). Without manual login, `dietpi-software` cannot start, hence you were then able to login?
Then you should see a license agreement, are asked to change user and software passwords and then a DietPi update should be applied, which should end with a reboot and final `dietpi-software` prompt after login. And there hitting "Install" did not do anything? From which console type (serial, local or SSH)?

jryx0, 2019-11-16:
@MichaIng 
On the first run with serial TTYs console, I just asked to input username/password, and then the keyborad did not work and soon  it came to prompt "please login to contiune" .

When I poweroff/on, it again asked to input  username/password, the license agreement shown.  After I choose ok the license agreement disappear and the Dietpi-software started automaticly .

Maybe I use friendfly modified image? [http://wiki.friendlyarm.com/wiki/index.php/DietPi](url)
Or some programs did not initialize correctly because of internet connection?(I saw some error with RED message in console.)

MichaIng, 2019-11-17:
@jryx0
Your first login prompt sounds like our postboot finished has finished after the getty has. This is expected and you need to hit return one time to have the username login prompt up for input. An info about that ("Please hit <return> to login") has been added with the last DietPi version, but images need to be redone to have this shown on first login. I aim for this anyway with new kernel and Debian buster after DietPi v6.27 release.

However after login the dietpi-software prompt is wanted, but you should be able to just select Install and it should then finish the setup.

As of your last post, did I understand correctly that you solved it now? Otherwise the new images should be ready til end of next week.

jryx0, 2019-11-17:
@MichaIng 
Yes, I has solved the problem, thank you.

MichaIng, 2019-12-09:
Okay, I'll mark this issue as closed. Feel free to reopen if required.

closed 2019-12-09 by MichaIng