MichaIng/DietPi #3227: dietpi-software isn´t working - failure

Elluminatus, 2019-11-13:
### Creating a bug report/issue

#### Details:
- Date           | Wed 13 Nov 20:02:54 CET 2019
- Bug report     | d3ad8196-fd6a-4d53-a6a6-3e9f37a32ef6
- DietPi version | v6.26.3 (Fourdee/master)
- Image creator  | N/A
- Pre-image      | N/A
- SBC device     | Native PC (x86_64) (index=21)
- Kernel version | Linux athene 4.9.0-6-amd64 #1 SMP Debian 4.9.88-1+deb9u1 (2018-05-07) x86_64 GNU/Linux
- Distro         | stretch (index=4)
- Command        | Connection test: https://deb.debian.org/debian/
- Exit code      | 4
- Software title | DietPi-Software

#### Steps to reproduce:

1. dietpi-software
2. abort with failure

#### Expected behaviour:

- show dietpi-software menu

#### Actual behaviour:

- abort

#### Extra details:

- ...

#### Additional logs:
```
Spider mode enabled. Check if remote file exists.
--2019-11-13 20:01:56--  https://deb.debian.org/debian/
Resolving deb.debian.org (deb.debian.org)... 130.89.148.77, 128.31.0.62, 149.20.4.15, ...
Connecting to deb.debian.org (deb.debian.org)|130.89.148.77|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://cdn-aws.deb.debian.org/debian/ [following]
Spider mode enabled. Check if remote file exists.
--2019-11-13 20:01:57--  https://cdn-aws.deb.debian.org/debian/
Resolving cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)... 54.240.168.16, 54.240.168.89, 54.240.168.6, ...
Connecting to cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)|54.240.168.16|:443... failed: Connection timed out.
Connecting to cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)|54.240.168.89|:443... failed: Connection timed out.
Connecting to cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)|54.240.168.6|:443... failed: Connection timed out.
Connecting to cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)|54.240.168.14|:443... failed: Connection timed out.
Connecting to cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)|2600:9000:21c7:4600:c:a756:bac0:93a1|:443... failed: Network is unreachable.
Connecting to cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)|2600:9000:21c7:3600:c:a756:bac0:93a1|:443... failed: Network is unreachable.
Connecting to cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)|2600:9000:21c7:b200:c:a756:bac0:93a1|:443... failed: Network is unreachable.
Connecting to cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)|2600:9000:21c7:fc00:c:a756:bac0:93a1|:443... failed: Network is unreachable.
Connecting to cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)|2600:9000:21c7:3a00:c:a756:bac0:93a1|:443... failed: Network is unreachable.
Connecting to cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)|2600:9000:21c7:ee00:c:a756:bac0:93a1|:443... failed: Network is unreachable.
Connecting to cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)|2600:9000:21c7:a800:c:a756:bac0:93a1|:443... failed: Network is unreachable.
Connecting to cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)|2600:9000:21c7:a00:c:a756:bac0:93a1|:443... failed: Network is unreachable.
```

---------------------------------------------------------------------

[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

MichaIng, 2019-11-13:
@Elluminatus 
Many thanks for your report.

Please try to raise the connection test timeout via: `dietpi-config` > `Network Options: Misc` > `G_CHECK_URL_TIMEOUT`

Elluminatus, 2019-11-14:
Hi, I reased the time over 10 and 15 to 20 and 20 works! So thank you very much for this hint. I will wait 3-4 days if there are new peoblems, otherwise this issue ma be closed.

Thank you Elluminatus

MichaIng, 2019-11-14:
@Elluminatus 
Great to here.

Note that 20 seconds just for a ping is massive delay, e.g. mine:
```
root@VM-Buster:~# time wget --spider https://deb.debian.org/debian/
Spider mode enabled. Check if remote file exists.
--2019-11-14 14:54:45--  https://deb.debian.org/debian/
Resolving deb.debian.org (deb.debian.org)... 130.89.148.77, 149.20.4.15, 128.31.0.62, ...
Connecting to deb.debian.org (deb.debian.org)|130.89.148.77|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://cdn-aws.deb.debian.org/debian/ [following]
Spider mode enabled. Check if remote file exists.
--2019-11-14 14:54:45--  https://cdn-aws.deb.debian.org/debian/
Resolving cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)... 143.204.247.18, 143.204.247.39, 143.204.247.100, ...
Connecting to cdn-aws.deb.debian.org (cdn-aws.deb.debian.org)|143.204.247.18|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 0 [text/html]
Remote file exists and could contain further links,
but recursion is disabled -- not retrieving.

real    0m0.277s
user    0m0.059s
sys     0m0.017s
```

We observed longer than usual DNS resolving, but in your case its indeed the connection time to the CDN 🤔. If you have actually good cable connection, you might want to test another APT source, probably the CDN somehow is not good reachable from your location.

MichaIng, 2019-12-08:
I'll mark this as closed, feel free to reopen if required.

closed 2019-12-08 by MichaIng