MichaIng/DietPi #3228: ALLO SPARKY Chromium kiosk mode crash

roudaille79, 2019-11-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=26
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='Fourdee'
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
stretch
9.11
- Kernel version | `uname -a`
Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
Sparky SBC (armv7l)
- Power supply used | (EG: 5V 1A RAVpower)
5V 3A
- SDcard used | (EG: SanDisk ultra)
SanDisk Ultra 16G A1 Class 10

#### Additional Information (if applicable)
- Software title | Chromium: web browser for desktop or autostart
- Was the software title installed freshly or updated/migrated? installed freshly
- Can this issue be replicated on a fresh installation of DietPi? yes
- Bug report ID | 234d4cbe-37e1-4bd1-87a8-adfbf981e814

#### Steps to reproduce

1. ...from dietpi-software, install Chromium
2. ...from dietpi-config launch AutoStart Options and select 11: Chromium
3. ...use root UID:0 as user to login with
4. ...reboot

#### Expected behaviour

- ...chromium launching in kiosk mode with welcome page displayed

#### Actual behaviour

- ...error message (see below Extra details)

#### Extra details

- ...
X.Org X Server 1.19.2
Release Date: 2017-03-02
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.9.0-8-armmp-lpae armv7l Debian
Current Operating System: Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l
Kernel command line: console=ttyS3 earlyprintk clk_ignore_unused selinux=0 scandelay console=tty0 loglevel=1 real_rootflag=rw root=/dev/mmcblk0p2 rootwait init=/lib/systemd/systemd aotg.urb_fix=1 aotg.aotg1_speed=0 boot_dev=sd0 hdmi_vid=2
Build Date: 03 November 2018  05:53:43AM
xorg-server 2:1.19.2-1+deb9u5 (https://www.debian.org/support) 
Current version of pixman: 0.34.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Thu Nov 14 15:52:51 2019
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
shared memfd open() failed: Function not implemented
[3259:3465:1114/155253.772882:ERROR:bus.cc(396)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
libEGL warning: DRI2: failed to authenticate
Received signal 11 SEGV_MAPERR 000000000000
#0 0x0000b3489bf0 <unknown>
#1 0x0000b34184e0 <unknown>
#2 0x0000b3489f3e <unknown>
#3 0x0000b348a16c <unknown>
#4 0x0000ad399fe0 <unknown>
#5 0x0000b1cb314c <unknown>
#6 0x0000b376f53c <unknown>
#7 0x0000b3c4050e <unknown>
#8 0x0000b4860b78 <unknown>
#9 0x0000b48611aa <unknown>
#10 0x0000b4861234 <unknown>
#11 0x0000b486126a <unknown>
#12 0x0000b32ea636 <unknown>
#13 0x0000b32ea672 <unknown>
#14 0x0000b316ac72 <unknown>
#15 0x0000b317680e <unknown>
#16 0x0000b3176d10 <unknown>
#17 0x0000b2249f70 <unknown>
#18 0x0000b2507e9c <unknown>
#19 0x0000b224b97e <unknown>
#20 0x0000b224c5aa <unknown>
#21 0x0000b2240918 <unknown>
#22 0x0000b314209c <unknown>
#23 0x0000b3142206 <unknown>
#24 0x0000b3142488 <unknown>
#25 0x0000b31493ee <unknown>
#26 0x0000b3140e08 <unknown>
#27 0x0000b1b2f510 ChromeMain
#28 0x0000ad38b4aa __libc_start_main
[end of stack trace]
Calling _exit(1). Core file will not be generated.
xinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.

MichaIng, 2019-11-15:
@roudaille79 
Many thanks for your report. I marked for investigation.

Does starting Chromium from desktop work instead?
_________
> shared memfd open() failed: Function not implemented

This is expected since the function was implemented with Linux 3.17 while Sparky SBC comes with Linux 3.10.

> [3259:3465:1114/155253.772882:ERROR:bus.cc(396)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")

Similar: https://unix.stackexchange.com/questions/423198/google-chrome-starts-but-fails-to-open-https-websites-this-site-can-t-be-reach
Although the address type error seems to be not what leads to the crash, but still should be checked what causes this, related to dbus, possibly polkit.

> libEGL warning: DRI2: failed to authenticate

That should be it, gl vs egl use error. Of course there is no GPU acceleration on Sparky SBC, only generic Mesa OpenGL drivers. We need to go through installed libraries (with Xserver) and Chromium options/flags.

MichaIng, 2020-07-05:
I mark this as closed due to missing reply. Feel free to reopen if required.

closed 2020-07-05 by MichaIng