MichaIng/DietPi #3229: [  OK  ] DietPi-PREP | Completed, disk can now be saved to .img for later use,

hifitime, 2019-11-15:
[  OK  ] DietPi-PREP | Completed, disk can now be saved to .img for later use, or, reboot system to start first run of DietPi.

How can I save img firmware in this step? Can someone give me a document step? Thank you

MichaIng, 2019-11-15:
@hifitime 
Generally you can clone the drive which holds the DietPi OS with any method from any external system. However to create an img file with minimal size (instead of size that matches the whole drive), it makes sense to shrink file system + partition(s) to a minimum first.

We created a script for that. If you have another DietPi system, plug the drive/SDcard there and run:
```
bash -c "$(curl -sSL https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-imager)"
```
- Select the drive which holds the fresh DietPi
- Choose whether to create a Clonezilla-based installer image (experimental) or a ready-to-run img file, i.e. to flash to SDcard and go (recommended currently, as proven to work well)
- Select the root partition (usually the last one)
- Give it a reasonable name, note that it is currently mainly for internal use, hence for creating our official images.

The script will shrink the root file system and partition to the minimum size, then uses `dd` to clone all to an img file and wraps it as 7zip archive with hashes and our readme.

hifitime, 2019-11-17:
OK, I'll try. Thank you for your help

hifitime, 2019-11-17:
[image]
[image]

**How can I choose this step? What does this mean? I don't quite understand. Please clarify it for me. Thank you very much**

MichaIng, 2019-11-18:
@hifitime 
You have to select "Drive" as source and "dd" as creation method. And note that you need to run this from an external DietPi system. You cannot create an image of a running system.

hifitime, 2019-11-21:
[image]

	
I plugged in the SD card that installed the dietpi system in another system running dietpi, but how do I choose when I get to this step? thank you.

hifitime, 2019-11-21:
Oh, I got it. This script is really good.

closed 2019-11-21 by hifitime

MichaIng, 2019-11-21:
@hifitime 
Ok great, In the first place "Drive" needs to be selected to select the SDcard then.

The GUI for sure requires some fine tuning. It is still under heavy development 😅. When we find some time we'll add some main overview page from where source and target type etc can be selected and changed and all info filled, then when everything is as desired, hitting "Run" will do the rest. So if anything was selected wrong accidentally of because not 100% clear explained, one will not need to rerun from start 😉.