MichaIng/DietPi #3230: fail to upgrade from version : v6.17.12 to version : v6.25.3

i3dcor, 2019-11-15:
#### Details:
- Date           | Fri 15 Nov 11:28:24 CET 2019
- Bug report     | N/A
- DietPi version | v6.17.12 (Fourdee/master)
- Image creator  | N/A
- Pre-image      | N/A
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (index=4)
- Command        | Connection test: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
- Exit code      | 8
- Software title | DietPi-Set_software

#### Steps to reproduce:
trying to upgrade from version : v6.17.12 to version  : v6.25.3

#### Actual behaviour:
fail to download  https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
#### Extra details:
if i put  https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
onto web navigator then it shows the file

#### Additional logs:
```
Spider mode enabled. Check if remote file exists.
--2019-11-15 11:27:39--  https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 82.223.97.47
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|82.223.97.47|:443... connected.
WARNING: The certificate of ‘raw.githubusercontent.com’ is not trusted.
WARNING: The certificate of ‘raw.githubusercontent.com’ hasn't got a known issuer.
The certificate's owner does not match hostname ‘raw.githubusercontent.com’
HTTP request sent, awaiting response... 404 Not Found
Remote file does not exist -- broken link!!!
```

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=17
G_DIETPI_VERSION_RC=12
G_GITBRANCH='master'
G_GITOWNER='Fourdee'

- Distro version | stretch
- Kernel version | Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- SBC device | RPi 3 Model B+ (armv7l)
- Power supply used | 5V 1A 
- SDcard used | SanDisk ultra

MichaIng, 2019-11-15:
@i3dcor 
Many thanks for your report.

It looks like your ca-certificates are outdated. Please try: `apt install --reinstall ca-certificates`
Then repeat the update.

i3dcor, 2019-11-18:
i've only checked to install ca-certificates not reinstall 
now updates ok
i  think this issue must be closed
thanks

closed 2019-11-18 by i3dcor

MichaIng, 2019-11-18:
Great, jep this package is required to practically access any HTTPS resource.