MichaIng/DietPi #3232: Waiting for DietPi-Postboot to finish...

hifitime, 2019-11-17:
### Creating a bug report/issue
 DietPi-Services
─────────────────────────────────────────────────────
 Mode: start

[  OK  ] DietPi-Services | start : avahi-daemon
[ .... ] DietPi-Services | nmbd
Debian Buster with Armbian Linux ttyMV0

DietPi login: root
Password:
[ INFO ] Waiting for DietPi-Postboot to finish... (Press CTRL+C to abort) (71)

### **Every time I start to this step, I will stay for a long time. I don't know why or how to solve it. Please help me, thank you**

#### Required Information
- DietPi version | `DietPi v6.26.3`
- Distro version | `10.2
- Kernel version | `Linux DietPi 4.19.59-mvebu64 #5.91 SMP PREEMPT Mon Jul 15 22:11:29 CEST 2019 aarch64 GNU/Linux
`
- SBC device | `Generic Device (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

hifitime, 2019-11-17:
DietPi login: root
Password:
[  OK  ] DietPi-Services | start : nmbd
[ .... ] DietPi-Services | smbd[ INFO ] Waiting for DietPi-Postboot to finish...[  OK  ] DietPi-Services | start : smbd
[  OK  ] DietPi-Services | start : nfs-kernel-server
[  OK  ] DietPi-Services | start : shairport-sync
[  OK  ] DietPi-Services | start : squeezelite
[  OK  ] DietPi-Services | start : logitechmediaserver
[  OK  ] DietPi-Services | start : cron

hifitime, 2019-11-18:
Can someone help me, please? Thank you

hifitime, 2019-11-18:
After my investigation, I found that the cause of this problem is the nmbd service startup problem

MichaIng, 2019-11-18:
@hifitime 
Many thanks for your report.

Jep the login script waits for the postboot script to finish, to have all services and boot scripts/tasks started. From your initial post it looks indeed like nmbd takes long to start up. The attempt should timeout after 90 seconds, however from your later post it looks like it finally succeeds.

Actually we could remove this wait. The login console does not depend on the postboot script. This only starts background services, does a DietPi update check, starts custom user boot scripts and such: https://github.com/MichaIng/DietPi/blob/dev/dietpi/postboot
But the script will still print the service start messages to your console, breaking e.g. input. So best would be indeed to have username+password login prompt (local console getty) strictly after postboot has finished. Sadly this is not trivial due to the kind of those systemd units (Type=idle) which do not really support ordering. Both simply start after all other boot tasks have finished.

We'll have a closer look after v6.27 release to clean this up.

hifitime, 2019-11-21:
OK, thank you very much. We are patient

MichaIng, 2019-11-21:
I'll reopen to not forget about implementing this.

hifitime, 2020-01-12:
A new discovery is that this problem only appears after Samba is installed....

MichaIng, 2020-01-12:
Jep that makes sense. Only if the postboot script hangs, the login prompt will be too early. And the hanging nmbd service is part of Samba.

MichaIng, 2026-11-16:
Finally done in #8507.

closed 2026-11-16 by MichaIng