MichaIng/DietPi #3233: Forced Password reset via `chage` not working on RPi Zero W

CupOfTea696, 2019-11-17:
### Creating a bug report/issue

#### Required Information
- DietPi version | v6.26.3
- Distro version | buster / 10.1
- Kernel version | Linux 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux
- SBC device | RPi Zero W (armv6l)
- SDcard used | SanDisk ultra (16 GB)

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce
1. `ssh dietpi@192.168.x.x`
2. `sudo -i`
3. `chage -d 0 dietpi`
4. `exit`
5. `exit`
6. `ssh dietpi@192.168.x.x`

#### Expected behaviour
After entering the password when ssh-ing back into DietPi (step 6), a prompt should appear informing the user that the password needs to be changed immediately. 

#### Actual behaviour
It just logs in normally, and the user has access to the system without having to change their password.

#### Extra details
I'm trying to force a password change after install via the `Automation_Custom_Script.sh`, so that the user immediately needs to change the password after logging in to the DietPi for the first time. While the password does seem to expired when checking with `chage -l dietpi`, the user never gets prompted to change their password. I used the same script when I set up my RPi 4, and I did get the password change prompt as expected there. 

```
$ chage -l dietpi
Last password change					: password must be changed
Password expires					: password must be changed
Password inactive					: password must be changed
Account expires						: never
Minimum number of days between password change		: 0
Maximum number of days between password change		: 99999
Number of days of warning before password expires	: 7
```

MichaIng, 2019-11-18:
@CupOfTea696 
Many thanks for your report.

I'll try to replicate.

CupOfTea696, 2019-11-18:
It's weird, though, because I think I was using Dropbear on my RPi 4 as well? I'll check later.

MichaIng, 2019-11-18:
@CupOfTea696 
Maybe you logged in via local console instead of SSH on RPi4? There it works. Since both are Buster-based and I tested on x86_64 Buster VM, it is pretty sure the same issue on all Buster-based systems when logging in via Dropbear SSH.

CupOfTea696, 2019-11-19:
> Maybe you logged in via local console instead of SSH on RPi4? There it works. Since both are Buster-based and I tested on x86_64 Buster VM, it is pretty sure the same issue on all Buster-based systems when logging in via Dropbear SSH.

I'm gonna double check just to make sure, but that might've been the case.

MichaIng, 2019-11-26:
@CupOfTea696 
Okay, so indeed the chage password timeout thing is a PAM feature, and if/how much PAM is used/supported or not, is an optional thing, hence it can be completely avoided, e.g. see the OpenSSH UsePAM option, which if set to "no" should as well not send password change requests anymore.

Dropbear has limited PAM support, password timeout/reset is not included, and more importantly Debian APT repo builds have Dropbear PAM support not enabled at all: https://buildd.debian.org/status/fetch.php?pkg=dropbear&arch=amd64&ver=2019.78-2&stamp=1564266264&raw=0
> configure: Disabling PAM

But just the same day of my conversation with Matt, a patch was added via mailing list to add exactly this feature:
- Mail: https://lists.ucc.gu.uwa.edu.au/pipermail/dropbear/2019q4/002175.html
- Patch: https://lists.ucc.gu.uwa.edu.au/pipermail/dropbear/2019q4/002174.html

When this has been added, we could open a request to Debian to enable PAM support for their builds. I can imagine that the missing password change dialog was a reason to disable PAM support, since, if I understood correctly, otherwise, when the password has expired, login via Dropbear is not possible anymore at all, which of course means that users can lock themselves out on headless systems.

CupOfTea696, 2019-11-28:
@MichaIng thanks for all the effort you're putting into this getting in touch with external dependencies!