MichaIng/DietPi #3236: Fresh installation fails to boot on Asus TinkerBoard

rpecb, 2019-11-20:
Can't start fresh installation of dietpi on TB. I tried full format of sdcard (`dd if=/dev/zero of=/dev/<sd card>`) it didn't help. Other systems (TinkerOS, Armbian) start up successfully.
Image checksums match.

#### Required Information
- DietPi version | 6.25
- SBC device | Asus TinkerBoard (non-S)
- SDcard used | SanDisk Ultra microSDHC 32GB 10 Class UHS-I [SDSQUNS-032G-GN3MN]

#### Steps to reproduce
1. Flash image to sd card with balena etcher (on Mac OS Catalina)
2. Power on device

#### Expected behaviour
- Success boot

#### Actual behaviour
- No any output on HDMI, just blinking leds, after a while they stop blinking

MichaIng, 2019-11-21:
@rpecb 
Many thanks for your report.

Are you able to SSH into the machine?

rpecb, 2019-11-21:
> @rpecb
> Many thanks for your report.
> 
> Are you able to SSH into the machine?

no, device does not appear on the network.
behaviour looks like TB tries to start a couple of times, then it turns off

MichaIng, 2019-11-22:
Strange, since in other cases it boots fine :thinking:.

There was another similar case, although on older image (Stretch-based): #2533

The new Buster-based image is some months old and reported ATB installs raised meanwhile, which I take as a general prove of function: https://github.com/MichaIng/DietPi/issues/2977#issuecomment-518005651

Did you try another flash method then Ether? I remember another case where this had some issue lately...
E.g. `dd if=/path/to/DietPi_v6.25_ASUSTB-ARMv7-Buster.img of=/dev/<sd card>`

MichaIng, 2020-01-01:
I mark this issue as closed, since it has been a while. A new image will be created the next days.

closed 2020-01-01 by MichaIng