MichaIng/DietPi #3239: DietPi-Software | Wallabag

alexandrebarsacq, 2019-11-22:
Wallabag is a self-hosted alternative to Pocket, as their [website](https://wallabag.org/en) states : 
"Save and classify articles. Read them later. Freely."

I found an [old](https://github.com/MichaIng/DietPi/issues/560) request that was closed during a cleaning, since I'm afraid it's long forgotten I went ahead and created this new one. Also on [Feathub](https://feathub.com/MichaIng/DietPi/+130).

Please consider adding Wallabag to dietpi, and many thanks for all the great work on DietPi !

MichaIng, 2019-12-09:
@alexandrebarsacq 
Thanks for your request. Lets see how much attention it receives. For now I'm pretty occupied by taking care about supported software titles, and for new ones, there are some very high on FeatHub list that I'd implement first.

However if anyone wants to do a start, I'll quickly assist to get it done: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

MichaIng, 2026-10-27:
Laravel-style PHP app with packed releases on GitHUb: https://github.com/wallabag/wallabag