MichaIng/DietPi #3241: DietPi has encounted an error, and, is unable to continue

hifitime, 2019-11-26:
DietPi has encounted an error, and, is unable to continue         -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket              -
---------------------------------------------------------------------

#### Details:
- Date           | Tue 26 Nov 11:43:33 GMT 2019
- Bug report     | aab1ba87-8d35-4542-a3e0-ddefe0644dc5
- DietPi version | v6.26.3 (MichaIng/master)
- Image creator  | wumin
- Pre-image      | hifitime.ine
- SBC device     | Generic Device (aarch64) (index=22)
- Kernel version | Linux DietPi 4.19.59-mvebu64 #5.91 SMP PREEMPT Mon Jul 15 22:11:29 CEST 2019 aarch64 GNU/Linux
- Distro         | buster (index=5)
- Command        | unzip -o 148.zip
- Exit code      | 9
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Archive:  148.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of 148.zip or
        148.zip.zip, and cannot find 148.zip.ZIP, period.
```

---------------------------------------------------------------------

[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

hifitime, 2019-11-26:
This is another error, when I install o! MPD

hifitime, 2019-11-26:
---------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue         -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket              -
---------------------------------------------------------------------

#### Details:
- Date           | Tue 26 Nov 12:01:49 GMT 2019
- Bug report     | aab1ba87-8d35-4542-a3e0-ddefe0644dc5
- DietPi version | v6.26.3 (MichaIng/master)
- Image creator  | wumin
- Pre-image      | hifitime.ine
- SBC device     | Generic Device (aarch64) (index=22)
- Kernel version | Linux DietPi 4.19.59-mvebu64 #5.91 SMP PREEMPT Mon Jul 15 22:11:29 CEST 2019 aarch64 GNU/Linux
- Distro         | buster (index=5)
- Command        | unzip -o 129.zip -d /var/www
- Exit code      | 9
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Archive:  129.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of 129.zip or
        129.zip.zip, and cannot find 129.zip.ZIP, period.
```

---------------------------------------------------------------------

[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

MichaIng, 2019-11-26:
@hifitime 
Many thanks for your report.

The related zip files are definitely working zip archives, e.g. check: https://github.com/jcorporation/myMPD/archive/master.zip
So either your "unzip" is broken or the downloads somehow corrupted without wget exiting with error. Disk corruption cannot be the case since the files are downloaded to tmpfs (RAMdisk).

Please try:
```
cd /tmp
apt install --reinstall unzip
wget https://github.com/jcorporation/myMPD/archive/master.zip
unzip master.zip
rm master.zip
rm -R myMPD-master
```
If this works fine, please try to rerun the dietpi-software install.

hifitime, 2019-11-26:
Okay may be a network problem. I tried several times to solve the problem

MichaIng, 2019-11-26:
@hifitime 
Okay, great it finally worked. Strange that `wget` did not report an error. I aim to add checksum verification to dietpi-software downloads, e.g. for GitHub downloads this is possible and would throw a more meaningful error message in the first place.

hifitime, 2019-11-26:
Please wait, there seems to be a problem....

hifitime, 2019-11-26:
root@DietPi:/tmp# wget https://github.com/jcorporation/myMPD/archive/master.zip
--2019-11-26 13:35:19--  https://github.com/jcorporation/myMPD/archive/master.zip
Resolving github.com (github.com)... 13.229.188.59
Connecting to github.com (github.com)|13.229.188.59|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/jcorporation/myMPD/zip/master [following]
--2019-11-26 13:35:21--  https://codeload.github.com/jcorporation/myMPD/zip/master
Resolving codeload.github.com (codeload.github.com)... 13.250.162.133
Connecting to codeload.github.com (codeload.github.com)|13.250.162.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘master.zip’

master.zip              [    <=>             ]   3.50M  14.6KB/s    in 8m 51s

2019-11-26 13:44:17 (6.74 KB/s) - ‘master.zip’ saved [3669142]

root@DietPi:/tmp# unzip master.zip
Archive:  master.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of master.zip or
        master.zip.zip, and cannot find master.zip.ZIP, period.
root@DietPi:/tmp#

MichaIng, 2019-11-26:
@hifitime 
Interesting, works fine here:
```
2019-11-26 14:46:40 root@micha:/tmp# wget https://github.com/jcorporation/myMPD/archive/master.zip
--2019-11-26 14:46:56--  https://github.com/jcorporation/myMPD/archive/master.zip
Resolving github.com (github.com)... 140.82.118.4
Connecting to github.com (github.com)|140.82.118.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/jcorporation/myMPD/zip/master [following]
--2019-11-26 14:46:57--  https://codeload.github.com/jcorporation/myMPD/zip/master
Resolving codeload.github.com (codeload.github.com)... 140.82.114.10
Connecting to codeload.github.com (codeload.github.com)|140.82.114.10|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘master.zip’

master.zip                                  [                                                     <=>                        ]  15.79M  1.65MB/s    in 11s

2019-11-26 14:47:08 (1.44 MB/s) - ‘master.zip’ saved [16554717]

2019-11-26 14:47:08 root@micha:/tmp# unzip master.zip
Archive:  master.zip
2c4a773b15da8bbdd84632969af0b80164d63677
   creating: myMPD-master/
  inflating: myMPD-master/.eslintrc.js
...
  inflating: myMPD-master/src/web_server_embedded_files.c
2019-11-26 14:47:09 root@micha:/tmp# rm master.zip
2019-11-26 14:47:09 root@micha:/tmp# rm -R myMPD-master
```
The local mirror is a different one, probably yours has some temporary issues 🤔. Bad that an IP-based download is somehow not possible. Even with `--no-check-certificate` a 400 error breaks download.

hifitime, 2019-11-26:
😭😭

hifitime, 2019-11-26:
I tried many times. There are always the same problems. There are errors in the downloaded files. Is there any other way  please

Joulinar, 2019-11-26:
@MichaIng 
Would ist be an option to tweak the `/etc/hosts` file to have `codeload.github.com` resolved with the IP that is working for you?

MichaIng, 2019-11-26:
@hifitime
Did you try to reinstall unzip and wget?
```
apt install --reinstall unzip wget
```
And when you download the archive via browser on desktop PC, does it open fine there? 

Yeah /etc/hosts would be a workaround if indeed the mirror ships somehow a broken file.
```
echo '140.82.114.10 codeload.github.com' >> /etc/hosts
```
But should not be kept forever. Actually I believe the issue is not the mirror, but the desktop download should verify that.

hifitime, 2019-11-30:
Okay may be a network problem.

closed 2019-11-30 by hifitime