MichaIng/DietPi #3242: dietpi-config resets chosen user for autostart login

zedrdave, 2019-11-27:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.26.3 branch master
- Distro version | buster
- Kernel version | Linux framboise 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
- SBC device | RPi 4 Model B (armv7l)
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)

When choosing an autostart option with auto-login (eg Desktop with auto-login), there is an option to choose a non-root user (thanks to the feature brought by this PR: https://github.com/MichaIng/DietPi/pull/2926).

Things work as expected when rebooting, but the setting seems to be reset (to the same autostart option, but with *root* instead of the preferred user) when `dietpi-config` is run again.

#### Steps to reproduce

1. open `dietpi-config`
2. browse menu to `9 - autostart options`
3. select an auto-login option (eg `2`: desktop with autologin)
4. on next screen pick a user *other* than `root` (eg `1000: dietpi`)
5. apply, exit and reboot
6. [dietpi autologs the correct user: `dietpi`]
7. open `dietpi-config`/`dietpi-software` and change other unrelated settings/install other software
8. reboot

#### Expected behaviour

dietpi should keep autologging the chosen user (`dietpi`)

#### Actual behaviour

Choice of autolog user gets reset and dietpi starts with chosen autostart (eg `desktop`) but under user `root` instead of `dietpi`

MichaIng, 2019-11-30:
@zedrdave 
Many thanks for your report. I'll have a look into the code, but actually the related setting should be touched by the `dietpi-autostart` script only, nothing else 🤔.

zedrdave, 2019-11-30:
Yes, I did a quick scan of the code and couldn't figure out how that happened.
The bug definitely happens (had to switch back user at least 4-5 times in the course of tweaking my install), but it is possible that it requires some specific installs/config to take place. I will try and get a perfectly reduced case.

MichaIng, 2019-11-30:
@zedrdave 
Found it. Indeed we did not store the chosen user to dietpi.txt, only apply it to the getty drop-in config. When any other script calls `dietpi-autostart <int>`, the drop-in config is removed and recreated based on dietpi.txt entry, non-interactively. 
DietPi-Software calls this at the end of all installs: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L14562-L14565
Actually there is not much reason to do that, especially since it is called interactively already when choosing any software title that has an autostart option: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L15294

DietPi-Config btw never calls the script non-interactively, hence it should not reset the choice, but regardless updating the dietpi.txt entry simply makes sense 😉.

Fix is: https://github.com/MichaIng/DietPi/commit/eb723566e3a16db4934683bd779160e67d152ca0
This is the line that you can manually add to your `/DietPi/dietpi/dietpi-autostart` script, everything else is minor coding:
```
G_CONFIG_INJECT 'AUTO_SETUP_AUTOSTART_LOGIN_USER=' "AUTO_SETUP_AUTOSTART_LOGIN_USER=$user" /DietPi/dietpi.txt
```

closed 2020-01-01 by MichaIng