MichaIng/DietPi #3244: rtorrent/rutorrent not accessible

notDavid, 2019-11-28:
### Creating a bug report/issue

#### Required Information
- DietPi version |
```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=26
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `10.2`
- Kernel version | `Linux pi4 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux`
- SBC device | `RPi 4 Model B (armv7l)`
- Power supply used | `original pi`

#### Additional Information (if applicable)
- Software title | `rtorrent` (rutorrent)
- Was the software title installed freshly or updated/migrated?
`installed freshly, *only* rtorrent package selected at install`
- Can this issue be replicated on a fresh installation of DietPi?
`Yes, also tested in clean Vmware image of Dietpi`
Also see forum: https://dietpi.com/phpbb/viewtopic.php?f=11&t=6534&p=21066

Please advise, thank you :)

#### Steps to reproduce

Opening the web interface of rutorrent produces the following errors in the browser/rutorrent window:
```
[28.11.2019 13:12:00] WebUI started.
[28.11.2019 13:12:00] Webserver user doesn't have read/write/execute access to the torrents directory. You cannot add torrents via ruTorrent. (/var/www/rutorrent/share/users/root/torrents)
[28.11.2019 13:12:00] Webserver user doesn't have read/write/execute access to the settings directory. ruTorrent settings cannot be saved. (/var/www/rutorrent/share/users/root/settings)
[28.11.2019 13:12:00] rTorrent user must have read/execute access to the torrents directory. You cannot add torrents via ruTorrent. (/var/www/rutorrent/share/users/root/torrents)
[28.11.2019 13:12:00] rTorrent user must have read/write/execute access to the settings directory. (/var/www/rutorrent/share/users/root/settings)
[28.11.2019 13:12:00] _cloudflare: Plugin will not work. Webserver user can't access external program (python).
[28.11.2019 13:12:00] trafic: Plugin will not work. rTorrent user can't access external program (php).
[28.11.2019 13:12:00] ratio: Some functionality will be unavailable. rTorrent user can't access external program (php).
[28.11.2019 13:12:00] _task: Plugin will not work. rTorrent user can't access external program (php).
[28.11.2019 13:12:00] unpack: Plugin will not work. rTorrent user can't access external program (php).
[28.11.2019 13:12:00] retrackers: Plugin will not work. rTorrent user can't access external program (php).
[28.11.2019 13:12:00] scheduler: Plugin will not work. rTorrent user can't access external program (php).
[28.11.2019 13:12:00] rss: Plugin will not work. rTorrent user can't access external program (php).
[28.11.2019 13:12:00] history: Plugin will not work. rTorrent user can't access external program (php).
[28.11.2019 13:12:00] xmpp: Plugin will not work. rTorrent user can't access external program (php).
[28.11.2019 13:12:00] autotools: Plugin will not work. rTorrent user can't access external program (php).
[28.11.2019 13:12:00] datadir: Plugin will not work. rTorrent user can't access external program (php).
[28.11.2019 13:12:00] rutracker_check: Plugin will not work. rTorrent user can't access external program (php).
[28.11.2019 13:12:00] loginmgr: Some functionality will be unavailable. rTorrent user can't access external program (php).
[28.11.2019 13:12:00] create: Plugin will not work. rTorrent user can't access external program (php).
[28.11.2019 13:12:00] screenshots: Plugin will not work. rTorrent user can't access external program (ffmpeg).
[28.11.2019 13:12:00] _task: Plugin will not work. rTorrent user can't access external program (pgrep).
[28.11.2019 13:12:00] spectrogram: Plugin will not work. rTorrent user can't access external program (sox).
[28.11.2019 13:12:00] mediainfo: Plugin will not work. rTorrent user can't access external program (mediainfo).
[28.11.2019 13:12:00] rss: Some functionality will be unavailable. rTorrent user can't access external program (curl).
[28.11.2019 13:12:01] Bad response from server: (500 [error,list]) Warning: XMLRPC call is failed.
```

Also see some elses screenshot in the forum: https://imgur.com/a/r5VUkGC

MichaIng, 2019-11-28:
@notDavid 
Many thanks for your report. Please run the following to fix the access issues:
```
chown -R www-data:rtorrent /var/www/rutorrent/share
chown root:root /var/www/rutorrent/share/.htaccess
```
Has been added to our installer for v6.27 as well.

notDavid, 2019-11-28:
Fix works indeed, thanks for the quick reply!

closed 2020-01-01 by MichaIng