MichaIng/DietPi #3245: no HDMI output on KODI start

ezekini, 2019-11-30:
### Creating a bug report/issue

#### Required Information
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=26
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
Stretch
RPi 3 Model B+ (armv7l)
Power supply used | (EG: 5V 1A RAVpower)
SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | Kodi
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce
1. Start Kodi / Set kodi to load on startup
2.

#### Expected behaviour

- See Kodi on external monitor/TV

#### Actual behaviour

- There is no output/signal

#### Extra details

- I can see in the tv Diepi loading, so the HDMI cable works, the tv recieves signal, but when Kodi loads, there is no more output

MichaIng, 2019-12-04:
@ezekini 
Many thanks for your report.

Does `/var/log/xorg.log.0` contain any error?

ezekini, 2019-12-04:
I can't find a file with that name in that folder.

MichaIng, 2019-12-09:
@ezekini 
Is there any relevant output if you run Kodi manually from console: `kodi --standalone`
If black screen shows up, try to exit with CTRL+C or kill it on another console/SSH session and see if there is any relevant output done to the underlying console.

ezekini, 2019-12-11:
No, tried killing and starting the service again running from console, but there is no output to the TV or the terminal....

MichaIng, 2019-12-12:
@ezekini 
Please assure the following:
- dietpi-config > Display Options > GPU/RAM Memory Split (or /DietPi/config.txt): gpu_mem_1024 split is at least 128M, at best 320M.
- dietpi-config > Display Options > Display Resolution is not "Headless". Not sure if some OpenGL resolution (which enables the related OpenGL driver) is required, but on RPi3 the legacy driver (non-GL) should work as well and usually best: `/DietPi/dietpi/func/dietpi-set_hardware rpi-opengl disable`
- Not sure in which version we fixed it, but Kodi 18 on RPi at least prints a warning when fbset is not installed and fails to recover desktop: `G_AGI fbset`
- Assure that only one monitor/output device is attached and that it shows standard boot + console. Then run `kodi --standalone` from that console.

ezekini, 2019-12-16:
After trying a few software solutions, I ended up changing other things, and the the HDMI cable turned to be the culprit. The cable is working (from PC to monitor for example), it just doesn't work with the rPI. Although this may have nothing to do with Dietpi, any idea on why it doesn't output signal after loading Kodi?

MichaIng, 2019-12-16:
@ekinig 
Did you try to increase HDMI signal strength: dietpi-config > Display Option > HDMI boost, e.g. raise to 7.

ekinig, 2019-12-16:
yes, it is currently on 7 but no luck.

MichaIng, 2019-12-16:
@ekinig 
It can be increased up to 11, just to test, however then indeed it sounds like the HDMI cable quality is not too good. Is it a very long cable or usual up to 1.5 meters?

I could imagine as well that power/voltage is not sufficient when loading Kodi additionally, which raises CPU + GPU power usage of course. `dmesg` could be at least checked for related entries.

ekinig, 2019-12-16:
It is a bit long, I guess between 2-3m. Don't worry, I swapped it with another one. I'll close this. Than you

closed 2019-12-16 by ekinig

MichaIng, 2019-12-16:
@ekinig 
Okay great. Yeah better cable beats high HDMI boost of course.