MichaIng/DietPi #3246: Abnormal start of gmrender does not work normally。。。。Please help

hifitime, 2019-11-30:
### Creating a bug report/issue
 **systemd status log**

gmrender.service - GMediaRender (DietPi)
   │    Loaded: loaded (/etc/systemd/system/gmrender.service; disabled;
   │ vendor preset: enabled)
   │    Active: active (running) since Sat 2019-11-30 05:01:33 GMT; 6s ago
   │   Process: 2871 ExecStartPre=/usr/bin/touch /var/log/gmrender.log
   │ (code=exited, status=0/SUCCESS)
   │   Process: 2872 ExecStartPre=/bin/chown gmrender:dietpi
   │ /var/log/gmrender.log (code=exited, status=0/SUCCESS)
   │  Main PID: 2873 (dash)
   │     Tasks: 8 (limit: 1160)
   │    Memory: 24.0M
CGroup: /system.slice/gmrender.service
   │            ├─2873 /bin/dash -c gmediarender -u
   │ "7639dc98-902c-47b1-aaa3-ea8769484e6d" -f "DietPi"
   │ --gstout-audiodevice=sysdefault --gstout-initial-volume-db=-1
   │ --logfile=/var/log/gmrender.log -I "$(sed -n 4p
   │ /DietPi/dietpi/.network)"
   │            └─2878 gmediarender -u 7639dc98-902c-47b1-aaa3-ea8769484e6d
   │ -f DietPi --gstout-audiodevice=sysdefault
   │ --gstout-initial-volume-db=-1 --logfile=/var/log/gmrender.log -I
   │ fe80::f2ad:4eff:fe03:647f
 Nov 30 05:01:35 DietPi gmediarender[2878]: gst_bin_add: assertion
   │ 'GST_IS_ELEMENT (element)' failed
   │ Nov 30 05:01:35 DietPi gmediarender[2878]: gst_bin_add: assertion
   │ 'GST_IS_ELEMENT (element)' failed
   │ Nov 30 05:01:35 DietPi gmediarender[2878]: gst_element_link_pads_full:
   │ assertion 'GST_IS_ELEMENT (src)' failed
   │ Nov 30 05:01:35 DietPi gmediarender[2878]: gst_element_get_static_pad:
   │ assertion 'GST_IS_ELEMENT (element)' failed
   │ Nov 30 05:01:35 DietPi gmediarender[2878]: gst_bin_add: assertion
   │ 'GST_IS_ELEMENT (element)' failed
   │ Nov 30 05:01:35 DietPi gmediarender[2878]: gst_bin_add: assertion
   │ 'GST_IS_ELEMENT (element)' failed
 Nov 30 05:01:35 DietPi gmediarender[2878]: gst_element_link_pads_full:
   │ assertion 'GST_IS_ELEMENT (src)' failed
   │ Nov 30 05:01:35 DietPi gmediarender[2878]: gst_element_get_static_pad:
   │ assertion 'GST_IS_ELEMENT (element)' failed
   │ Nov 30 05:01:35 DietPi gmediarender[2878]: gst_element_link_pads_full:
   │ assertion 'GST_IS_ELEMENT (dest)' failed
   │ Nov 30 05:01:35 DietPi gmediarender[2878]:
   │                                            Trying to dispose object
   │ "mixer", but it still has a parent "(null)".
   │                                            You need to let the parent
   │ manage the object instead of unreffing the object directly.
=============================================================
**/var/log/gmrender.log**

INFO  [2019-11-30 05:07:06.110299 | connmgr] Registering support for 'image/x-pcx'
INFO  [2019-11-30 05:07:06.138589 | gstreamer] Setting buffer duration to 1000ms
INFO  [2019-11-30 05:07:06.138954 | gstreamer] Set mute to off
INFO  [2019-11-30 05:07:06.139011 | gstreamer] Set volume fraction to 0.891251
INFO  [2019-11-30 05:07:06.139333 | webserver] Provide /upnp/grender-64x64.png (image/png) from /usr/share/gmediarender/grender-64x64.png
INFO  [2019-11-30 05:07:06.139538 | webserver] Provide /upnp/grender-128x128.png (image/png) from /usr/share/gmediarender/grender-128x128.png
INFO  [2019-11-30 05:07:06.143488 | webserver] Provide /upnp/rendertransportSCPD.xml (text/xml) from buffer
INFO  [2019-11-30 05:07:06.144544 | webserver] Provide /upnp/renderconnmgrSCPD.xml (text/xml) from buffer
INFO  [2019-11-30 05:07:06.147157 | webserver] Provide /upnp/rendercontrolSCPD.xml (text/xml) from buffer
ERROR [2019-11-30 05:07:07.151133 | upnp] UpnpInit(ip=fe80::f2ad:4eff:fe03:647f, port=49494) Error: UPNP_E_SOCKET_ERROR (-208). Retrying... (59s)
ERROR [2019-11-30 05:07:08.160800 | upnp] UpnpInit(ip=fe80::f2ad:4eff:fe03:647f, port=49494) Error: UPNP_E_SOCKET_ERROR (-208). Retrying... (58s)
ERROR [2019-11-30 05:07:09.171847 | upnp] UpnpInit(ip=fe80::f2ad:4eff:fe03:647f, port=49494) Error: UPNP_E_SOCKET_ERROR (-208). Retrying... (57s)
ERROR [2019-11-30 05:07:10.179355 | upnp] UpnpInit(ip=fe80::f2ad:4eff:fe03:647f, port=49494) Error: UPNP_E_SOCKET_ERROR (-208). Retrying... (56s)
ERROR [2019-11-30 05:07:11.183353 | upnp] UpnpInit(ip=fe80::f2ad:4eff:fe03:647f, port=49494) Error: UPNP_E_SOCKET_ERROR (-208). Retrying... (55s)
.........................

ERROR [2019-11-30 05:08:00.646413 | upnp] UpnpInit(ip=fe80::f2ad:4eff:fe03:647f, port=49494) Error: UPNP_E_SOCKET_ERROR (-208). Retrying... (6s)
ERROR [2019-11-30 05:08:01.659193 | upnp] UpnpInit(ip=fe80::f2ad:4eff:fe03:647f, port=49494) Error: UPNP_E_SOCKET_ERROR (-208). Retrying... (5s)
ERROR [2019-11-30 05:08:02.669644 | upnp] UpnpInit(ip=fe80::f2ad:4eff:fe03:647f, port=49494) Error: UPNP_E_SOCKET_ERROR (-208). Retrying... (4s)
ERROR [2019-11-30 05:08:03.679329 | upnp] UpnpInit(ip=fe80::f2ad:4eff:fe03:647f, port=49494) Error: UPNP_E_SOCKET_ERROR (-208). Retrying... (3s)
ERROR [2019-11-30 05:08:04.686541 | upnp] UpnpInit(ip=fe80::f2ad:4eff:fe03:647f, port=49494) Error: UPNP_E_SOCKET_ERROR (-208). Retrying... (2s)
ERROR [2019-11-30 05:08:05.699760 | upnp] UpnpInit(ip=fe80::f2ad:4eff:fe03:647f, port=49494) Error: UPNP_E_SOCKET_ERROR (-208). Retrying... (1s)
ERROR [2019-11-30 05:08:06.710078 | upnp] UpnpInit(ip=fe80::f2ad:4eff:fe03:647f, port=49494) Error: UPNP_E_SOCKET_ERROR (-208). Retrying... (0s)
ERROR [2019-11-30 05:08:06.720533 | upnp] UpnpInit(ip=fe80::f2ad:4eff:fe03:647f, port=49494) Error: UPNP_E_SOCKET_ERROR (-208). Giving up.
ERROR [2019-11-30 05:08:06.720894 | main] ERROR: Failed to initialize UPnP device

#### Required Information
DietPi version | DietPi v6.26.3
Distro version | `10.2
Kernel version | Linux DietPi 4.19.59-mvebu64 #5.91 SMP PREEMPT Mon Jul 15 22:11:29 CEST 2019 aarch64 GNU/Linux
SBC device | `Generic Device (aarch64)
Power supply used | (EG: 5V 1A RAVpower)
SDcard used | (EG: SanDisk ultra)

hifitime, 2019-12-01:
Please help me, thanks

hifitime, 2019-12-01:
If it is installed according to the official compilation method, it can run normally
it will run normally

git clone git://git.code.sf.net/p/minidlna/git minidlna-git

hifitime, 2019-12-02:
😭

MichaIng, 2019-12-02:
@hifitime 
Many thanks for your report.

Could you paste:
```
systemctl cat gmrender
sed -n 4p /DietPi/dietpi/.network
```
I'll also do some test later.

hifitime, 2019-12-26:
root@DietPi-test:~# cat /var/log/gmrender.log
INFO  [2019-12-27 00:06:29.353453 | main] gmediarender 0.0.7-git log started [ gmediarender 0.0.7-git (libupnp-1.8.4; glib-2.58.1; gstreamer-1.14.4) ]
INFO  [2019-12-27 00:06:29.353750 | output] Using output module: gst (GStreamer multimedia framework)
INFO  [2019-12-27 00:06:29.450773 | connmgr] Registering support for 'audio/x-ac3'
INFO  [2019-12-27 00:06:29.450876 | connmgr] Registering support for 'audio/ac3'
INFO  [2019-12-27 00:06:29.450900 | connmgr] Registering support for 'audio/x-private1-ac3'
INFO  [2019-12-27 00:06:29.450921 | connmgr] Registering support for 'audio/*'
INFO  [2019-12-27 00:06:29.554860 | connmgr] Registering support for 'video/x-raw'
INFO  [2019-12-27 00:06:29.586940 | connmgr] Registering support for 'audio/x-wavpack'
INFO  [2019-12-27 00:06:29.587656 | connmgr] Registering support for 'audio/x-raw'
INFO  [2019-12-27 00:06:29.605443 | connmgr] Registering support for 'audio/mpeg'
INFO  [2019-12-27 00:06:29.605544 | connmgr] Registering support for 'audio/x-mpeg'
INFO  [2019-12-27 00:06:29.605566 | connmgr] Registering support for 'audio/x-scpls'
INFO  [2019-12-27 00:06:29.605586 | connmgr] Registering support for 'audio/L16;rate=44100;channels=2'
INFO  [2019-12-27 00:06:29.606075 | connmgr] Registering support for 'audio/x-amr-nb-sh'
INFO  [2019-12-27 00:06:29.606137 | connmgr] Registering support for 'audio/x-amr-wb-sh'
INFO  [2019-12-27 00:06:29.606687 | connmgr] Registering support for 'audio/x-eac3'
INFO  [2019-12-27 00:06:29.610200 | connmgr] Registering support for 'audio/x-dts'
INFO  [2019-12-27 00:06:29.610303 | connmgr] Registering support for 'audio/x-private1-dts'
INFO  [2019-12-27 00:06:29.619312 | connmgr] Registering support for 'audio/x-flac'
INFO  [2019-12-27 00:06:29.627271 | connmgr] Registering support for 'audio/x-sbc'
INFO  [2019-12-27 00:06:29.720471 | connmgr] Registering support for 'video/mpeg'
INFO  [2019-12-27 00:06:29.720569 | connmgr] Registering support for 'video/x-cdxa'
INFO  [2019-12-27 00:06:29.741188 | connmgr] Registering support for 'application/x-yuv4mpeg'
INFO  [2019-12-27 00:06:29.757022 | connmgr] Registering support for 'application/x-subtitle'
INFO  [2019-12-27 00:06:29.757120 | connmgr] Registering support for 'application/x-subtitle-sami'
INFO  [2019-12-27 00:06:29.757145 | connmgr] Registering support for 'application/x-subtitle-tmplayer'
INFO  [2019-12-27 00:06:29.757167 | connmgr] Registering support for 'application/x-subtitle-mpl2'
INFO  [2019-12-27 00:06:29.757187 | connmgr] Registering support for 'application/x-subtitle-dks'
INFO  [2019-12-27 00:06:29.757208 | connmgr] Registering support for 'application/x-subtitle-qttext'
INFO  [2019-12-27 00:06:29.757229 | connmgr] Registering support for 'application/x-subtitle-lrc'
INFO  [2019-12-27 00:06:29.757250 | connmgr] Registering support for 'application/x-subtitle-vtt'
INFO  [2019-12-27 00:06:29.757662 | connmgr] Registering support for 'application/x-ssa'
INFO  [2019-12-27 00:06:29.757702 | connmgr] Registering support for 'application/x-ass'
INFO  [2019-12-27 00:06:29.784971 | connmgr] Registering support for 'video/x-msvideo'
INFO  [2019-12-27 00:06:29.786256 | connmgr] Registering support for 'audio/x-alaw'
INFO  [2019-12-27 00:06:29.786338 | connmgr] Registering support for 'audio/x-mulaw'
INFO  [2019-12-27 00:06:29.786362 | connmgr] Registering support for 'audio/x-wma'
INFO  [2019-12-27 00:06:29.786388 | connmgr] Registering support for 'image/jpeg'
INFO  [2019-12-27 00:06:29.786409 | connmgr] Registering support for 'video/x-divx'
INFO  [2019-12-27 00:06:29.786431 | connmgr] Registering support for 'video/x-msmpeg'
INFO  [2019-12-27 00:06:29.786455 | connmgr] Registering support for 'video/x-h263'
INFO  [2019-12-27 00:06:29.786475 | connmgr] Registering support for 'video/x-h264'
INFO  [2019-12-27 00:06:29.786496 | connmgr] Registering support for 'video/x-dv'
INFO  [2019-12-27 00:06:29.795534 | connmgr] Registering support for 'video/x-huffyuv'
INFO  [2019-12-27 00:06:29.795589 | connmgr] Registering support for 'video/x-wmv'
INFO  [2019-12-27 00:06:29.795613 | connmgr] Registering support for 'image/x-jpc'
INFO  [2019-12-27 00:06:29.795684 | connmgr] Registering support for 'video/x-vp8'
INFO  [2019-12-27 00:06:29.795709 | connmgr] Registering support for 'image/png'
INFO  [2019-12-27 00:06:29.796300 | connmgr] Registering support for 'application/x-subtitle-avi'
INFO  [2019-12-27 00:06:29.808396 | connmgr] Registering support for 'text/x-raw'
INFO  [2019-12-27 00:06:29.900365 | connmgr] Registering support for 'audio/x-sid'
INFO  [2019-12-27 00:06:29.936293 | connmgr] Registering support for 'subpicture/x-dvd'
INFO  [2019-12-27 00:06:29.936397 | connmgr] Registering support for 'subpicture/x-pgs'
INFO  [2019-12-27 00:06:29.958154 | connmgr] Registering support for 'audio/midi'
INFO  [2019-12-27 00:06:29.958289 | connmgr] Registering support for 'audio/riff-midi'
INFO  [2019-12-27 00:06:29.984649 | connmgr] Registering support for 'application/x-rtp'
INFO  [2019-12-27 00:06:29.984766 | connmgr] Registering support for 'application/x-rtcp'
INFO  [2019-12-27 00:06:29.985600 | connmgr] Registering support for 'application/x-srtp'
INFO  [2019-12-27 00:06:29.985710 | connmgr] Registering support for 'application/x-srtcp'
INFO  [2019-12-27 00:06:30.101130 | connmgr] Registering support for 'video/x-dirac'
INFO  [2019-12-27 00:06:30.101256 | connmgr] Registering support for 'audio/x-lpcm'
[... cut ...]
ERROR [2019-12-27 00:10:51.427530 | upnp] UpnpInit(ip=fe80::f2ad:4eff:fe03:647f, port=49494) Error: UPNP_E_SOCKET_ERROR (-208). Retrying... (1s)
ERROR [2019-12-27 00:10:52.431002 | upnp] UpnpInit(ip=fe80::f2ad:4eff:fe03:647f, port=49494) Error: UPNP_E_SOCKET_ERROR (-208). Retrying... (0s)
ERROR [2019-12-27 00:10:52.433512 | upnp] UpnpInit(ip=fe80::f2ad:4eff:fe03:647f, port=49494) Error: UPNP_E_SOCKET_ERROR (-208). Giving up.
ERROR [2019-12-27 00:10:52.433636 | main] ERROR: Failed to initialize UPnP device

hifitime, 2019-12-26:
root@DietPi-test:/etc/systemd/system# cat gmrender.service 
[Unit]
Description=GMediaRender (DietPi)
After=network.target dietpi-boot.service

[Service]
User=gmrender
Group=dietpi
ExecStartPre=+/usr/bin/touch /var/log/gmrender.log
ExecStartPre=+/bin/chown gmrender:dietpi /var/log/gmrender.log
ExecStart=/bin/dash -c 'gmediarender -u "54de1800-c864-4ce1-bb01-4aa09d7d9e1c" -f "Walle-test" --gstout-audiodevice=sysdefault --gstout-initial-volume-db=-1 --logfile=/var/log/gmrender.log -I "$(sed -n 4p /DietPi/dietpi/.network)"'

[Install]
WantedBy=multi-user.target

hifitime, 2019-12-26:
There's still a problem. Gmrender can't start normally
Please help, thank you...

hifitime, 2019-12-26:
[image]

hifitime, 2019-12-28:
[image]

hifitime, 2019-12-28:
[image]

MichaIng, 2019-12-28:
See: #3263
I am now compiling new binaries, as those from Raspbian Buster seem to be the issue for some reason.

MichaIng, 2019-12-28:
@hifitime 
Could you please test the following:
```
wget https://dietpi.com/downloads/binaries/buster/gmediarender_armv6l -O /usr/bin/gmediarender
systemctl restart gmrender
```

hifitime, 2019-12-28:
After completing the above operations, the program cannot be started

[image]

This configuration cannot be started：

[image]

This configuration cannot be started either
[image]

 I guess it may be a configuration problem. I tried to modify it to start normally

[image]

hifitime, 2019-12-28:
After completing the above operations, the program cannot be started

`root@DietPi:/etc/systemd/system# systemctl restart gmrender

^C`

This configuration cannot be started：

`[Unit]
Description=GMediaRender (DietPi)
Wants=network-online.target
After=network-online.target dietpi-boot.service

[Service]
User=gmrender
ExecStartPre=+/bin/dash -c 'systemctl set-environment ACTIVE_IP=\$(mawk 'NR==4' /DietPi/dietpi/.network)'
ExecStart=/usr/local/bin/gmediarender -u '$(mawk 'NR==5' /DietPi/dietpi/.hw_model)' -f '$(</etc/hostname)' --gstout-audiosink=alsasink --gstout-audiodevice=sysdefault --gstout-initial-volume-db=-1 --logfile=stdout -I "\$ACTIVE_IP"

[Install]
WantedBy=multi-user.target`

I guess it may be a configuration problem. I tried to modify it to start normally

`[Unit]
Description=GMediaRender (DietPi)
Wants=network-online.target
After=network-online.target dietpi-boot.service

[Service]
User=gmrender
#ExecStartPre=+/bin/dash -c 'systemctl set-environment ACTIVE_IP=\$(mawk 'NR==4' /DietPi/dietpi/.network)'
#ExecStart=/usr/local/bin/gmediarender -u '$(mawk 'NR==5' /DietPi/dietpi/.hw_model)' -f '$(</etc/hostname)' --gstout-audiosink=alsasink --gstout-audiodevice=sysdefault --gstout-initial-volume-db=-1 --logfile=stdout -I "\$ACTIVE_IP"

ExecStartPre=+/bin/chown gmrender:dietpi /var/log/gmrender.log
ExecStart=/bin/dash -c 'gmediarender -u "54de1800-c864-4ce1-bb01-4aa09d7d9e1c" -f "DietPi" --gstout-audiodevice=sysdefault --gstout-initial-volume-db=-1 --logfile=/var/log/gmrender.log'

[Install]
WantedBy=multi-user.target`

Check the startup log and it is running normally

`INFO  [2019-12-28 15:25:42.998628 | gstreamer] Query volume fraction: 0.891251
INFO  [2019-12-28 15:25:42.998756 | control] Output initial volume is 0.891251; setting control variables accordingly.
INFO  [2019-12-28 15:25:42.998808 | control] Setting volume-db to -1.20db == #97
INFO  [2019-12-28 15:25:42.999174 | main] Ready for rendering.`

hifitime, 2019-12-28:
I think new binaries has solved this problem, and the same configuration cannot be started normally

Another message is that I compiled gmrender recurrect on this machine and it works normally,
In order to match your judgment, I have carried out relevant operations according to your steps, and I don't know whether it will have additional impact on the judgment

MichaIng, 2019-12-28:
@hifitime 
Ah, somehow the $(...) command substitutions did not expand. Seems to behave differently when copying to console compared to script :sweat_smile:.

Forgot to asking: Which SBC you run this on? The binary is for RPi, I thought yours is the same as of the other issue.

hifitime, 2019-12-28:
Anyway, the problem is solved. Thank you very much for your support
I am a customized SBC, not the mainstream device on the market,

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=26
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
10.2
- Kernel version | `uname -a`
Linux DietPi 4.19.84-mvebu64 #5.99 SMP PREEMPT Mon Nov 18 22:13:49 CST 2019 aarch64 GNU/Linux

- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
Generic Device (aarch64)

hifitime, 2019-12-28:
English is not good, all depends on Google translation, hahahaha

closed 2019-12-28 by hifitime

MichaIng, 2019-12-28:
@hifitime 
Ah okay, just verified that `cat` command when copying the above to console resolves command substitutions as expected. I guess you copied the content to file directly via command line editor?

Okay yeah when you run an aarch64 SBC, then of course the ARMv6 binary does not work. Okay this proves that it is no an RPi issue, but a Debian Buster issue in general. I also recognised that the Debian binary is MUCH smaller and they do not install the gstreamer addons for compilation, probably this is the reason for certain features missing, leading to the issue.

I will play around a bid and compile binaries for all architectures.

> English is not good, all depends on Google translation, hahahaha

Jep, learning SBCs and Linux usually means learning English as well, no worries, your translator results are pretty fine. Btw, to get whole sentences nicely translated: https://www.deepl.com/translator
Sadly a very limited amount of supported languages so far.

MichaIng, 2019-12-29:
Issue resolved with: f70a008faa48687977a62ec4f8644610872d6500
Changelog: https://github.com/MichaIng/DietPi/commit/3f95e2e294bcd29d53ebc677aa286fd31aff27a1

MichaIng, 2019-12-30:
@hifitime 
Last question: There are two icons the binary expects/the build installs:
- `/usr/local/share/gmediarender/grender-128x128.png`
- `/usr/local/share/gmediarender/grender-64x64.png`

On startup it complains if those are missing, although just as a warning. Since I have not much experience with GMediaRender, do you know where those are used and if its bad when missing?