MichaIng/DietPi #3247: State degraded - 3 failed units

copain60, 2019-11-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2019-12-04:
@copain60 
Not sure what enabled them, but `dhcpcd.service` and `wpa_supplicant.service` must both be disabled. Those conflict with the `networking.service` (`ifupdown` package, default on Debian) and `isc-dhcp-client` (default on DietPi). The default implementation does DHCP lease and WPA authentication on interface init by itself, without the need of any external service running/being active.

`dhcpcd5` can be actually purged completely. I know that it is currently installed by Pi-hole to set the static IP. But this is done via /etc/network/interfaces in our case, hence the bid strange method to use a DHCP client to configure a network to **not** use DHCP, is not required on DietPi.

`dietpi-ramlog_disable.service` is enabled when you uninstall DietPi-RAMlog/choose full logging option. It should disable itself, hence only run once. If it fails, then this would mean lost log files, which could cause other services failing to start if they require a log file to exist. Lets see:
```
journalctl -u dietpi-ramlog_disable
```
However since nothing else failed, it is obsolete anyway.

So to solve all together:
```
systemctl disable --now dhcpcd wpa_supplicant dietpi-ramlog_disable
apt purge dhcpcd5
```

copain60, 2019-12-07:
@MichaIng 
Thanks for your details. I disabled all 3 services.

Journalctl of dietpi-ramlog_disable does not list any entries.
I had indeed activated full logging for debugging purposes some time ago. However after all issues were resolved, I went back to default ramlog #1.

Service dhcpcd5 can not be found on my RPi.

May I ask an additional question in that context?

After reboot 'systemctl status certbot' shows the warning, that the service changed on disk. I should run 'systemctl daemon-reload' to reload units.

To resolve this issue, is there a way to find out these changes in order to update and save back all necessary files?

Many thanks.

MichaIng, 2019-12-07:
@copain60 
Hmm, aside from APT, nothing should change the service actually, aside you do a `dietpi-backup`, although even timestamps are preserved 🤔.
Could you paste the services content?
```
systemctl cat certbot
cat /lib/systemd/system/certbot.service
```
Both above should show the same content, although not sure if systemctl shows the "loaded" content (which you could reload from actual file via `systemctl daemon-reload`).

copain60, 2019-12-07:
@MichaIng 
Content is as follows:

```
root@DietPi:~# systemctl cat certbot
# /lib/systemd/system/certbot.service
[Unit]
Description=Certbot
Documentation=file:///usr/share/doc/python-certbot-doc/html/index.html
Documentation=https://letsencrypt.readthedocs.io/en/latest/
[Service]
Type=oneshot
ExecStart=/usr/bin/certbot -q renew
PrivateTmp=true

# /etc/systemd/system/certbot.service.d/dietpi-lighttpd.conf
[Service]
ExecStartPost=/bin/bash -c '/bin/cat /etc/letsencrypt/live/<url name>.com/p
rivkey.pem /etc/letsencrypt/live/<url name>.com/cert.pem > /etc/letsencrypt
/live/<url name>.com/combined.pem'
```

```
root@DietPi:~# cat /lib/systemd/system/certbot.service
[Unit]
Description=Certbot
Documentation=file:///usr/share/doc/python-certbot-doc/html/index.html
Documentation=https://letsencrypt.readthedocs.io/en/latest/
[Service]
Type=oneshot
ExecStart=/usr/bin/certbot -q renew
PrivateTmp=true
```

This is 'loaded' content, since 'systemctl daemon-reload' has been run before.

MichaIng, 2019-12-07:
@copain60 
Okay that is all fine. The `combined.pem` that is required for Lighttpd can actually be created more elegant with current Certbot version, by adding its creation script to `/etc/letsencrypt/renewal-hooks/post/`. However nothing urgent.

MichaIng, 2019-12-09:
I mark this issue as closed now. Feel free to reopen if required.

closed 2019-12-09 by MichaIng

copain60, 2019-12-10:
@MichaIng 
Sorry for my delayed reply.
Actually I added the following script to /etc/letsencrypt/renewal-hooks/deploy/ without updating any other config files:

```
#!/bin/sh

for domain in $RENEWED_DOMAINS
do
   # Combine the certificate and private key file
    cat /etc/letsencrypt/live/$domain/privkey.pem /etc/letsencrypt/live/$domain/cert.pem > \
      /etc/letsencrypt/live/$domain/combined.pem
done

systemctl restart lighttpd.service
```

Please refer as follows:
[https://dietpi.com/phpbb/viewtopic.php?t=5904](url)
[https://tech.borpin.co.uk/2019/03/22/letsencrypt-ssl-certificates-by-dns-challenge-with-lighttpd/](url)

Could that be the reason for the changed service warning?
How would you recommend me to update my configuration?

MichaIng, 2019-12-10:
@copain60 
> Could that be the reason for the changed service warning?

This should not be the case.

> How would you recommend me to update my configuration?

Your method is what I would recommend as well. Note that `dietpi-letsencrypt` adds the same as certbot drop-in config, which was required on Jessie since the legacy certbot didn't support those hook scripts. With the deploy hook, remove `/etc/systemd/system/certbot.service.d/dietpi-lighttpd.conf` to not have things doubled. THIS then indeed changes the systemd unit, hence `systemctl daemon-reload` is required to reload the files.