MichaIng/DietPi #3248: gitea fails to start due to Segmentation fault

Cabur, 2019-12-01:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
`G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=26
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `10.2`
- Kernel version | `Linux DietPi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux`
- SBC device | `RPi3`
- Power supply used | `5V 1A`
- SDcard used | `SanDisk ultra`

#### Additional Information (if applicable)
- Software title | gitea
- Was the software title installed freshly or updated/migrated?
 Fresh Install
- Can this issue be replicated on a fresh installation of DietPi?
Fresh Install

- Bug report ID | `0db9287d-176f-4c67-b39f-ca13a08ff016`

#### Steps to reproduce

1. Use dietpi-software to install gitea

#### Expected behaviour
- gitea should be running and accessible on port 3000

#### Actual behaviour
service file fails
`code=killed, signal=SEGV`

#### Extra details
running the same command outside of service files produces the same results 
root@DietPi:/home/dietpi# /mnt/dietpi_userdata/gitea/gitea web
Segmentation fault

MichaIng, 2019-12-01:
Many thanks for your report.

Hmm a SEG sounds like a wrong or corrupted binary, especially since Gitea comes without any additional files or needs. Could you try the most recent one:
```
wget https://dl.gitea.io/gitea/1.9.6/gitea-1.9.6-linux-arm-6 -O /mnt/dietpi_userdata/gitea/gitea
```

Cabur, 2019-12-01:
Tied and got the same results:

```
root@DietPi:/mnt/dietpi_userdata/gitea# wget https://dl.gitea.io/gitea/1.9.6/gitea-1.9.6-linux-arm-6 -O /mnt/dietpi_userdata/gitea/gitea
    --2019-12-01 18:34:01--  https://dl.gitea.io/gitea/1.9.6/gitea-1.9.6-linux-arm-6
Resolving dl.gitea.io (dl.gitea.io)... 104.27.143.155, 104.27.142.155, 2606:4700:30::681b:8e9b, ...
Connecting to dl.gitea.io (dl.gitea.io)|104.27.143.155|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 86613588 (83M) [application/octet-stream]
Saving to: ‘/mnt/dietpi_userdata/gitea/gitea’

/mnt/dietpi_userdata/gitea/gitea        100%[============================================================================>]  82.60M  3.22MB/s    in 43s

2019-12-01 18:34:45 (1.92 MB/s) - ‘/mnt/dietpi_userdata/gitea/gitea’ saved [86613588/86613588]

root@DietPi:/mnt/dietpi_userdata/gitea# systemctl start gitea
root@DietPi:/mnt/dietpi_userdata/gitea# ^start^status
systemctl status gitea
● gitea.service - Gitea (DietPi)
   Loaded: loaded (/etc/systemd/system/gitea.service; disabled; vendor preset: enabled)
   Active: failed (Result: signal) since Sun 2019-12-01 18:34:55 GMT; 6s ago
  Process: 2046 ExecStart=/mnt/dietpi_userdata/gitea/gitea web (code=killed, signal=SEGV)
 Main PID: 2046 (code=killed, signal=SEGV)

Dec 01 18:34:55 DietPi systemd[1]: Started Gitea (DietPi).
Dec 01 18:34:55 DietPi systemd[1]: gitea.service: Main process exited, code=killed, status=11/SEGV
Dec 01 18:34:55 DietPi systemd[1]: gitea.service: Failed with result 'signal'.
root@DietPi:/mnt/dietpi_userdata/gitea# ./gitea web
Segmentation fault
```

MichaIng, 2019-12-01:
@Cabur 
Ah I forget about a permissions issue that was just solved with v6.27. Please try:
```
chown -R dietpi:dietpi /mnt/dietpi_userdata/gitea
systemctl restart gitea
```
Although this should not lead to a segmentation fault 🤔.

Cabur, 2019-12-02:
Still no luck. Also looked at the side note, Pulled the latest from their git V1.10 and same results. going to go back and try some more tomorrow.

MichaIng, 2019-12-02:
@Cabur 
I'll try it on my RPi as well, on the VMs it works just fine on all distros (just tested).

Cabur, 2019-12-02:
@MichaIng 
So interesting, when running inside gdb, it comes right up, outside still seg faults (still running V1.10 from their github)
`dietpi@DietPi:~$ gdb /mnt/dietpi_userdata/gitea/gitea web`

MichaIng, 2019-12-04:
@Cabur 
Very strange indeed, I make some test on my RPi as well, since on VM it works very well.

MichaIng, 2019-12-09:
🈹 Bug confirmed: This binary fails with segmentation fault on my RPi2 as well: https://dl.gitea.io/gitea/1.9.6/gitea-1.9.6-linux-arm-6

MichaIng, 2019-12-09:
Okay indeed seems to be an RPi kernel bug, which has been resolved upstream but has not yet been released: https://github.com/raspberrypi/linux/issues/3271#issuecomment-541227379

Note that `rpi-update` is not installed by default on DietPi, `G_AGI rpi-update` will do, however as a general recommendation I would wait for the official package release, as long as one can wait.

MichaIng, 2020-01-22:
New firmware packages have been released, tested it and works fine now. I mark this as closed.

#### Solution:
```
apt update
apt upgrade
systemctl restart gitea
```

closed 2020-01-22 by MichaIng

jcw, 2020-10-03:
Running DietPi on an x86_64, after installing gitea, `/mnt/dietpi_userdata/gitea/` was still owned by root. After a bit of puzzling (well before I saw this issue), a `chown -R dietpi:dietpi /mnt/dietpi_userdata/gitea/` got it all working. This is with prep-script on Debian buster, kernel 4.19.0, and latest DietPi v6.32.2 release.

Joulinar, 2020-10-03:
Hi,

was this a fresh installations of Gitea? Because permission should have been changed on installation
https://github.com/MichaIng/DietPi/blob/208ab64c88ce168d2b3af4205544ea66edeacaf4/dietpi/dietpi-software#L12052

I did a quick test installation on my VM and permissions got adjusted as expected
```
root@DietPiVM1:~# ls -la /mnt/dietpi_userdata | grep gitea
drwxr-xr-x 6 dietpi dietpi 4096 Oct  4 01:08 gitea
root@DietPiVM1:~#
```

MichaIng, 2020-10-03:
Strange, that step is included in the installer, I was just verifying and the directory is owned as expected by dietpi:dietpi 🤔: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L12052

.... lol while I was writing, thanks @Joulinar 😄.

Joulinar, 2020-10-03:
2nd time today 🤣

jcw, 2020-10-03:
Odd. This on a fresh DietPi install from yesterday, and a fresh gitea install today. Only other install I did before that was Nextcloud, which went fine (incl lighttpd & mariadb). Running on a 2009 Mac Mini, which I'm setting up as server, but that should not be relevant, as the Debian 10.6 installer proceeded without any issues.

Anyway, it works. Water under the bridge, I don't think there's a way to get to the bottom of this without doing it all over again.

jcw, 2020-10-11:
Just did a fresh gitea install on a fresh rpi3 (along with nextcloud and nginx) - same issue. Solved with `chown -R dietpi:dietpi /mnt/dietpi_userdata/gitea/` + reboot, then gitea shows up on port 3000. It's not tied into nginx out of the box, but that's no big deal, and well-documented [here](https://docs.gitea.io/en-us/reverse-proxies/).

jcw, 2020-10-11:
Here's an example setup which brings gitea to `http://[my-ip-addr]/git/`:

```
# cat /etc/nginx/sites-dietpi/dietpi-gitea.conf
    location /git/ { # Note: Trailing slash
        proxy_pass http://localhost:3000/; # Note: Trailing slash
    }
```

This location also needs to be specified on first connect to gitea, where it asks for the URL to use:

```
# grep ROOT_URL /mnt/dietpi_userdata/gitea/custom/conf/app.ini
ROOT_URL         = http://192.168.188.92/git/
```

Joulinar, 2020-10-11:
Well but setting up a reverse proxy has nothing to do with the initial issue. I will test setting up Gitea and missing permission later on once I'm back home.

jcw, 2020-10-11:
My apologies, I should have created a separate issue for this.

Joulinar, 2020-10-11:
I verified it again, using same setup as you. RPi3B+, installing Nginx, Nextcloud as well as Gitea

That's how it looks like during installation
```
root@DietPi3:/mnt/dietpi_userdata# ls -la
total 44
drwxrwxr-x 11 dietpi   dietpi   4096 Oct 11 19:20 .
drwxr-xr-x  6 root     root     4096 Oct  8 12:58 ..
drwxrwxr-x  2 dietpi   dietpi   4096 Sep 22 13:54 Music
drwxrwxr-x  2 dietpi   dietpi   4096 Sep 22 13:54 Pictures
drwxrwxr-x  2 dietpi   dietpi   4096 Sep 22 13:54 Video
drwxrwxr-x  2 dietpi   dietpi   4096 Sep 22 13:54 downloads
drwxr-xr-x  3 root     root     4096 Oct 11 19:20 gitea
drwxr-xr-x  5 mysql    mysql    4096 Oct 11 19:20 mysql
drwxr-xr-x  2 www-data www-data 4096 Oct 11 19:20 nextcloud_data
drwx------  3 dietpi   dietpi   4096 Oct  8 21:58 syncthing
drwxr-xr-x  3 dietpi   dietpi   4096 Oct  8 11:45 syncthing_data
root@DietPi3:/mnt/dietpi_userdata#
```
But as soon as configurations steps completed
```
[ SUB1 ] DietPi-Software > Configuring Gitea: Git with a cup of tea
[ INFO ] DietPi-Create_MySQL_DB | Creating MariaDB database gitea for user gitea
[  OK  ] Creating MariaDB database gitea for user gitea | Completed
```
permission changed as expected
```
root@DietPi3:/mnt/dietpi_userdata# ls -la
total 44
drwxrwxr-x 11 dietpi   dietpi   4096 Oct 11 19:20 .
drwxr-xr-x  6 root     root     4096 Oct  8 12:58 ..
drwxrwxr-x  2 dietpi   dietpi   4096 Sep 22 13:54 Music
drwxrwxr-x  2 dietpi   dietpi   4096 Sep 22 13:54 Pictures
drwxrwxr-x  2 dietpi   dietpi   4096 Sep 22 13:54 Video
drwxrwxr-x  2 dietpi   dietpi   4096 Sep 22 13:54 downloads
drwxr-xr-x  3 dietpi   dietpi   4096 Oct 11 19:20 gitea
drwxr-xr-x  6 mysql    mysql    4096 Oct 11 19:24 mysql
drwxrwx---  4 www-data www-data 4096 Oct 11 19:24 nextcloud_data
drwx------  3 dietpi   dietpi   4096 Oct  8 21:58 syncthing
drwxr-xr-x  3 dietpi   dietpi   4096 Oct  8 11:45 syncthing_data
root@DietPi3:/mnt/dietpi_userdata#
```
I'm still not able to replicate your behaviour.

jcw, 2020-10-11:
Curiouser and curiouser. Thanks for checking. I'll redo my test later. I checked well after the gitea install was over. Puzzling!