MichaIng/DietPi #3251: Documentation for dietpi.txt auto install conflicts with github software list

aberusugi, 2019-12-02:
Version Info:
    G_DIETPI_VERSION_CORE=6
    G_DIETPI_VERSION_SUB=25
    G_DIETPI_VERSION_RC=3
    G_GITBRANCH='master'
    G_GITOWNER='MichaIng'

Other info asked for by the template should not be necessary, since this is before booting it, this is the current raspberry pi version downloaded from the dietpi website as of today just a few moments ago. The dietpi.txt links to https://github.com/MichaIng/DietPi/wiki/DietPi-Software-list - and one of the commented out examples says - AUTO_SETUP_INSTALL_SOFTWARE_ID=74    # will install LAMP webserver stack - but the link on github says the software ID of the LAMP webserver stack is 76 and that 74 is NULL. Documentation should be updated in both places to reflect this.

MichaIng, 2019-12-03:
@aberusugi 
Many thanks for your report.

Indeed these info is outdated due to MySQL removal and MariaDB replacing it. With Jessie, there was still a dedicated MySQL server installable. MariaDB was a separate choice and the webserver stack was called LAAP (Linux Apache Aria PHP). Since Stretch, the `mysql-server` is just a meta package which pulls MariaDB as well, hence we had two identical choices. So we removed MySQL and renamed prior LAAP to LAMP (Linux Apache MariaDB PHP) after some users were missing the well known LAMP name from the list.

However by chance I removed those multiple default entries from `dietpi.txt` with version 6.26, see how it looks now: https://github.com/MichaIng/DietPi/blob/master/dietpi.txt#L102-L107
Of course this only has effect on images which are created after v6.26 release.

MichaIng, 2019-12-09:
I mark this issue as closed. New images will be created after v6.27 release which then contain the adjusted `dietpi.txt`. Feel free to reopen if required.

closed 2019-12-09 by MichaIng