MichaIng/DietPi #3255: samba doesn't start

ernero93, 2019-12-06:
#### Details:
- Date           | Fri  6 Dec 21:03:46 GMT 2019
- Bug report     | N/A
- DietPi version | v6.26.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid N2 (aarch64) (index=15)
- Kernel version | Linux Dietpi-N2 4.9.196+ #1 SMP PREEMPT Mon Oct 21 08:51:53 CEST 2019 aarch64 GNU/Linux
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Setting up samba (2:4.5.16+dfsg-1+deb9u2) ...
Failed to preset unit: Unit file /etc/systemd/system/samba-ad-dc.service is masked.
/usr/bin/deb-systemd-helper: error: systemctl preset failed on samba-ad-dc.service: No such file or directory
Job for smbd.service failed because the control process exited with error code.
See "systemctl status smbd.service" and "journalctl -xe" for details.
invoke-rc.d: initscript smbd, action "start" failed.
● smbd.service - Samba SMB Daemon
   Loaded: loaded (/lib/systemd/system/smbd.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2019-12-06 20:56:11 GMT; 44ms ago
     Docs: man:smbd(8)
           man:samba(7)
           man:smb.conf(5)
  Process: 7475 ExecStart=/usr/sbin/smbd $SMBDOPTIONS (code=exited, status=1/FAILURE)
 Main PID: 7475 (code=exited, status=1/FAILURE)

MichaIng, 2019-12-07:
@ernero93 
Many thanks for your report.

> Unit file /etc/systemd/system/samba-ad-dc.service is masked.

Very strange, we do not mask any service, hence this must come from the package itself or another package. I'll try to replicate.

ernero93, 2019-12-07:
root@DietPi:~# journalctl -u smbd
-- Logs begin at Fri 2019-12-06 20:24:12 GMT, end at Sat 2019-12-07 17:31:48 GMT. --
Dec 06 20:24:15 DietPi systemd[1]: Starting Samba SMB Daemon...
Dec 06 20:24:16 DietPi systemd[1]: smbd.service: Main process exited, code=exited, status=1/FAILURE
Dec 06 20:24:16 DietPi systemd[1]: Failed to start Samba SMB Daemon.
Dec 06 20:24:16 DietPi systemd[1]: smbd.service: Unit entered failed state.
Dec 06 20:24:16 DietPi systemd[1]: smbd.service: Failed with result 'exit-code'.
Dec 06 20:24:23 DietPi systemd[1]: smbd.service: Unit cannot be reloaded because it is inactive.
Dec 07 17:21:33 DietPi systemd[1]: Starting Samba SMB Daemon...
Dec 07 17:21:33 DietPi systemd[1]: smbd.service: Main process exited, code=exited, status=1/FAILURE
Dec 07 17:21:33 DietPi systemd[1]: Failed to start Samba SMB Daemon.
Dec 07 17:21:33 DietPi systemd[1]: smbd.service: Unit entered failed state.
Dec 07 17:21:33 DietPi systemd[1]: smbd.service: Failed with result 'exit-code'.
Dec 07 17:21:44 DietPi systemd[1]: smbd.service: Unit cannot be reloaded because it is inactive.
root@DietPi:~#

ernero93, 2019-12-07:
[samba.txt](https://github.com/MichaIng/DietPi/files/3935368/samba.txt)

MichaIng, 2019-12-07:
@ernero93 
Binary logs seem to be done to a file. Please check /var/log/samba (or smbd or similar) for error logs, which should give more details.

ernero93, 2019-12-07:
Unfortunately these two files are empty (size 0)

"/var/log/samba/log.nmbd"
"/var/log/samba/log.smbd"

BTY I noticed another issue. Kodi doesn't start if libGLESv2.so is not installed.

After running these commands

sudo apt-get update
sudo apt-get upgrade
sudo apt-get install libgles2-mesa libsdl2-dev

kodi starts

MichaIng, 2019-12-08:
@ernero93 
> "/var/log/samba/log.nmbd"
"/var/log/samba/log.smbd"

When restarting the service the files should fill. DietPi-RAMlog clears it every hour.
> BTY I noticed another issue. Kodi doesn't start if libGLESv2.so is not installed.
>
> After running these commands
>
> sudo apt-get update
> sudo apt-get upgrade
> sudo apt-get install libgles2-mesa libsdl2-dev
>
> kodi starts

Hmm strange, `libgles2-mesa` should be installed automatically with Xserver (and Kodi), `libsdl2-dev` should definitely not be required, `libsdl2-2.0-0` in case only, but also not from what I know and on my tests. However I'll have a deeper look into this for Odroid N2.

MichaIng, 2019-12-09:
Okay, when installing Kodi via `dietpi-software`, [mesa-utils-extra](https://packages.debian.org/stretch/mesa-utils-extra) is installed through Xserver which pulls `libgles2-mesa` as dependency. The installed `kodi-odroid` package should pull libSDL2, if required, it is even shipped by Meverics repo and e.g. pulled when installing `tvheadend`.

But what I found now is an Odroid N2 special version for Kodi, provided by Meveric: `kodi-n2-fbdev-odroid`
This makes sense since the other Kodi uses X11 while Odroid N2 does not support X11, hence uses fbdev directly. I'll check [setup-odroid](http://fuzon.co.uk/meveric/pool/main/s/setup-odroid/setup-odroid_0.0.1-50_all.deb) utility about how this is applied there.

MichaIng, 2019-12-09:
@ernero93 
Okay on Odroid N2 the following should work and provide full GPU support:
```
G_AGP kodi kodi-odroid # Purge previous Kodi versions
apt install aml-libs-odroid malig52-fbdev-opencl-odroid libnfs8 kodi-n2-fbdev-odroid
echo 'media_clock
firmware
decoder_common
stream_input
amvdec_avs
amvdec_h264
amvdec_h264_4k2k
amvdec_mh264
amvdec_h264mvc
amvdec_h265
amvdec_mjpeg
amvdec_mmjpeg
amvdec_mpeg12
amvdec_mpeg4
amvdec_mmpeg4
amvdec_real
amvdec_vc1
amvdec_vp9' > /etc/modules-load.d/dietpi-n2-kodi.conf
```

ernero93, 2019-12-10:
Hello,
The case about samba can be closed. After several tests from a fresh install, I never have any more issue with samba, the service starts as expected once installed.
Regards 
Ernero93

MichaIng, 2019-12-10:
@ernero93 
Okay great, if you find time to test the above for fully accelerated Kodi, I could add it to our code.

ernero93, 2019-12-11:
I tested the command :
 **apt install aml-libs-odroid malig52-fbdev-opencl-odroid libnfs8 kodi-n2-fbdev-odroid** 
on a fresh beta version. It ran without any problem, and then, I put kodi in autostart mode, but it didn't start automatically; I got the following message 
/DietPi/dietpi/misc/start_kodi: line 49: xinit: command not found.
Let me know !
Regards

MichaIng, 2019-12-12:
@ernero93 
Ah yeah, that needs to be adjusted as well since this Kodi does not run on X11. Could you please try the following from console and if your have from desktop terminal emulator as well:
```
kodi --standalone
```

If this works fine, open `/DietPi/dietpi/misc/start_kodi` and replace
```
if (( $G_HW_MODEL < 10 )); then
```
with
if (( $G_HW_MODEL < 10 || $G_HW_MODEL == 15 )); then
```
which leads to Kodi autostart option doing just the same.

ernero93, 2019-12-12:
After modifiying start_kodi as indicated , kodi starts automatically as expected.
Regards.
Ernero93

MichaIng, 2019-12-13:
@ernero93
Great, many thanks for testing. I'll implement it then as above for v6.27.

MichaIng, 2019-12-18:
Done: https://github.com/MichaIng/DietPi/pull/3269

closed 2020-01-01 by MichaIng