MichaIng/DietPi #3256: Microphone not detected rpi 4 4gb

jpie726, 2019-12-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.26.3
- Distro version | Buster, Debian 10.2
- Kernel version | Linux DietPi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 
  armv7l GNU/Linux
- SBC device | RPi 4 4GB
- Power supply used | Factory Samsung base and generic cord
- SDcard used | 32GB Sandisk

#### Additional Information (if applicable)
- Software title | Chromium
- Installed fresh other than Chromium, LXDE, and Nukkit

#### Steps to reproduce

1. install dietpi
2. install Nukkit
3. install LXDE
4. install Chromium
5. find a website that uses microphone
6. try to use that microphone and click on the new button in the top right

#### Expected behaviour

- Microphone should be detected

#### Actual behaviour

- No microphone is being detected

#### Extra details

- and also discordapp.com isn't showing a popup to enable microphone, although that could be a settings issue or an issue with discord itself

MichaIng, 2019-12-07:
@jpie726 
Many thanks for your report.

Did you try to use the microphone right from the console? E.g.
```
arecord -d 5 test.wav # Speak for 5 seconds
aplay test.wav
```
How did you enable audio capability, via `dietpi-config` > `Audio Options`?
Did you enable equaliser as well?

jpie726, 2019-12-07:
no, i did not enable audio or test mic from console, will try

jpie726, 2019-12-07:
arecord gave me this error `arecord: main:828: audio open error: No such file or directory`

jpie726, 2019-12-07:
i am currently in root if that changes anything

jpie726, 2019-12-07:
also i just changed overclocking profile to gpu high and now i cant access the performance menu to change it back

MichaIng, 2019-12-09:
@jpie726 
> also i just changed overclocking profile to gpu high and now i cant access the performance menu to change it back

You mean GPU high profile made your RPi4 unbootable? Plug your SDcard to external system and edit `config.txt` to comment the `arm_freq` and `core_freq` lines. How much current is your power supply producing and any rainbow low voltage warning when attaching a monitor? We already increased over_voltage to one level higher then what was reported as stable, hence I would wonder if some boards out there still require more. However probably there are some unlucky boards that require `over_voltage=6` for the frequencies we apply with "high" profiles 🤔.

jpie726, 2020-01-07:
Sorry, on vaca right now but i should be back soon

jpie726, 2020-01-07:
Also, i dont remember the exact problem with gpu overclock but i could still boot back in

MichaIng, 2020-06-01:
Marking as closed, feel free to reopen if issue is still present.

closed 2020-06-01 by MichaIng