MichaIng/DietPi #3257: Open Beta v6.27 | Please help testing and hardening the upcoming release

MichaIng, 2019-12-08:
RC version | v6.27.0
---------- | -------
Changelog | https://github.com/Fourdee/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/Fourdee/DietPi/compare/master...beta
How to apply | https://github.com/Fourdee/DietPi/blob/master/BRANCH_SYSTEM.md
Release planned | Mid December

#### Important testing cases:
- New **Amiberry** release, especially on Stretch systems: #3156
- New **Home Assistant** with Python3.8: #3219
- **RPi4**: EEPROM update implementation: #3217
- New software title **TasmoAdmin**: #3103

______________
#### Closes issues:
#2593 #2897 #3103 #3156 #3175 #3177 #3180 #3181 #3182 #3187 #3191 #3194 #3203 #3207 #3208 #3213 #3217 #3219 #3242 #3244
_______________
### Known issues
#### DietPi functionality
- DietPi-Config | Enabling WiFi + Ethernet adapters, both on different subnets, breaks WiFi connection in some cases: #2103
  - Current workaround is to disable the adapter that is not in use (not for internet connection).
  - Otherwise a custom routing table is required.

#### SBC/device related
- RPi | On TigerVNC virtual desktop, LXAppearance hangs on dbus-launch: #1791
- Pine A64 | Bluetooth add-on module currently not supported: #528
- Odroid XU4 | Kodi freezes shortly on video playback: #2584
- Rock64 | 3.5mm A/V jack is currently not functional: #2522

#### Software title related
- DietPi-Software | Node-RED: Pre-installed modules cannot be updated via web UI: #2073
- DietPi-Software | Raspimjpeg: With Lighttpd, streaming mjpeg does not work: #1747
- DietPi-Software | MATE desktop: When logging in as root, desktop items and right-clock context menu is missing: #3160
- DietPi-Software | Sonarr/Mono: With current Mono version 6, import to a file system without UNIX permissions support (exFAT, FAT32/vfat and NTFS without "permissions" option) fails, regardless of user/umask mount options: #3179
- DietPi-Software | Transmission: On Raspbian/Debian Stretch, RAM usage raises unlimited over time: #2413

Joulinar, 2019-12-09:
Hi Guys,

for testing purposes, I was going to setup a DEV system. But I was facing some issues during initial software installation. Could it be, that `dietpi.com` is offline??

```
───────────────────────────────
DietPi Error Handler: 
───────────────────────────────
 DietPi-Software: Connection test: https://dietpi.com/downloads/conf/desktop
- Exit code: 4
- DietPi version: v6.26.3 (MichaIng/master) | HW_MODEL:-1 | HW_ARCH:2 | DISTRO:5
- Image creator: DietPi Core Team
- Pre-image: Raspbian Lite

Log file content:
Spider mode enabled. Check if remote file exists.
--2019-12-09 23:31:13--  https://dietpi.com/downloads/conf/desktop
Resolving dietpi.com (dietpi.com)... 104.27.178.199, 104.27.179.199, 2606:4700:30::681b:b3c7, ...
Connecting to dietpi.com (dietpi.com)|104.27.178.199|:443... connected.
HTTP request sent, awaiting response... Read error (Success.) in headers.
Giving up.
```

Trying the website as well, is giving some `Cloudflare's Always Online™` information, instead of the usual website :(

MichaIng, 2019-12-09:
@Joulinar 
I noticed this as well some hours ago, but when I was home it worked again. No hint on server logs, probably temporary VPS provider network issue.

Please try again now.

Joulinar, 2019-12-09:
@MichaIng 
hmmm I tried to install a VNC Server now, but it still keeps failing

```
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing LXDE: ultra lightweight desktop

[FAILED] DietPi-Software | Connection test: https://dietpi.com/downloads/conf/desktop
```

After a Chromium test installation, it still fails to connect `ssh.dietpi.com` to sent survey data.

```
[FAILED] DietPi-Survey | Connection test: ssh.dietpi.com
```

MichaIng, 2019-12-10:
@Joulinar 
Okay even stranger, since `ssh.dietpi.com` works from my module phone... I though it is related to Cloudflare, but seems not.

Works fine from my server as well:
```
2019-12-10 00:44:53 root@micha:/tmp# dietpi-survey 1
[  OK  ] DietPi-Survey | Connection test: ssh.dietpi.com
[  OK  ] DietPi-Survey | Successfully sent survey data
```

Server shows no overload, no error logs, nor anything else suspicious. Still VPS provider network issue, I guess. I have to sleep now, will check back tomorrow.

Joulinar, 2019-12-10:
@MichaIng 
Maybe it's depending on Geo Location. Will check it tomorrow as well. Good Night.

Fourdee, 2019-12-10:
@MichaIng 

Great work as always, thank you :+1: 

Twitter REF: https://twitter.com/DietPi_/status/1204297321820934144

Joulinar, 2019-12-10:
hmm still strange. I'm still getting the following once trying to open `dietpi.com` website

[image]

MichaIng, 2019-12-10:
@Joulinar 
Could you try to disable any adblock and data savings feature? In my case access is only refused when using Operas data savings feature on mobile phone, which relays requests through Operas server then.

I tested with all firewall unblocked and disabled, but it is not our server itself which refuses the requests, they do not even reach it. It must be the VPS network.

DutchFlash, 2019-12-10:
I get the same error page.

MichaIng, 2019-12-10:
Lets move to #3260 with this topic...

Joulinar, 2019-12-13:
@MichaIng 
I was running a basic update on an empty system from v6.26.3 > v6.27.0 to test the RPi4: EEPROM feature. But I'm not sure if it was installed or activated. because I don't see anything within the logs. Or do I misunderstand how EEPROM is working? Within `journalctl` I don't see that much as well.

[update_log.txt](https://github.com/MichaIng/DietPi/files/3960058/update_log.txt)

MichaIng, 2019-12-13:
@Joulinar
Many thanks for testing. Hmm is your RPi4 correctly recognised as such: `echo $G_HW_MODEL`

Joulinar, 2019-12-13:
@MichaIng 
hmm I guess that not the correct output :)

```
root@DietPi4:~# echo $G_HW_MODEL
-1
root@DietPi4:~#
```

MichaIng, 2019-12-13:
@Joulinar
Okay, something wrong with model estimation, probably the syntax of the file which holds the release code has changed. Now I remember a similar reported issue. Could you paste: `cat /proc/cpuinfo`

Joulinar, 2019-12-13:
@MichaIng 
see attached

[cpu_info.txt](https://github.com/MichaIng/DietPi/files/3960374/cpu_info.txt)

MichaIng, 2019-12-13:
@Joulinar 
Ah your board is a new revision PCB1.1. I'll have to add this to detection, at best switch to a generic code decoding method, since every bit has a meaning stating all the info up to RAM size.

Joulinar, 2019-12-13:
@MichaIng 
lucky me having a newer model :)

Btw I noticed something more after I switched to v6.27.0. During boot I'm getting the following lines. 
[image]

these are the corresponding lines from `journalctl`
```
Dez 13 12:33:36 DietPi4 systemd[1]: Starting DietPi-Boot...
Dez 13 12:33:36 DietPi4 systemd[1]: Starting /etc/rc.local Compatibility...
Dez 13 12:33:36 DietPi4 systemd[1]: Starting Permit User Sessions...
Dez 13 12:33:36 DietPi4 systemd[1]: Started /etc/rc.local Compatibility.
Dez 13 12:33:36 DietPi4 systemd[1]: Started Permit User Sessions.
Dez 13 12:33:36 DietPi4 sh[386]: eth0=eth0
Dez 13 12:33:37 DietPi4 systemd[1]: Starting Network Time Synchronization...
Dez 13 12:33:37 DietPi4 systemd[1]: Started Network Time Synchronization.
Dez 13 12:33:37 DietPi4 systemd[1]: Reached target System Time Synchronized.
Dez 13 12:33:37 DietPi4 kernel: bcmgenet fd580000.genet eth0: Link is Down
Dez 13 12:33:39 DietPi4 systemd[1]: systemd-rfkill.service: Succeeded.
Dez 13 12:33:40 DietPi4 kernel: bcmgenet fd580000.genet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
Dez 13 12:33:40 DietPi4 kernel: IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Dez 13 12:33:53 DietPi4 systemd-timesyncd[548]: Synchronized to time server for the first time 212.18.3.18:123 (0.debian.pool.ntp.org).
Dez 13 12:33:53 DietPi4 systemd[1]: Stopping Network Time Synchronization...
Dez 13 12:33:53 DietPi4 systemd[1]: systemd-timesyncd.service: Succeeded.
Dez 13 12:33:53 DietPi4 systemd[1]: Stopped Network Time Synchronization.
Dez 13 12:33:53 DietPi4 systemd[1]: Started DietPi-Boot.
```

MichaIng, 2019-12-13:
@Joulinar 
Okay I added support for this revision and possible further ones, at least when produces by Sony as well: https://github.com/MichaIng/DietPi/commit/b24939e16459f9489d8d0db5a9c9197fdb6b2f32
You can test:
```
wget https://raw.githubusercontent.com/MichaIng/DietPi/dev/dietpi/func/dietpi-obtain_hw_model -O /DietPi/dietpi/func/dietpi-obtain_hw_model
/DietPi/dietpi/func/dietpi-obtain_hw_model
. /DietPi/dietpi/func/dietpi-globals
echo $G_HW_MODEL # should now print "4"
/DietPi/dietpi/func/dietpi-set_hardware rpi-eeprom # Install EEPROM updater + update related firmware
```

Joulinar, 2019-12-13:
@MichaIng 
jup, it's being detected correctly now. 
```
root@DietPi4:~# echo $G_HW_MODEL
4
```

as well during login showing RPi4, which was not the case before.
```
 ─────────────────────────────────────────────────────
 DietPi v6.27.0 (beta) : 20:26 - Fr 13.12.2019
 ─────────────────────────────────────────────────────
 - Device model : RPi 4 Model B (armv7l)
 - Uptime : up 7 minutes
 - CPU temp : 40'C : 104'F (Optimal temperature)
 - LAN IP : 192.168.0.12 (eth0)
 - Info Text : !!! DEV Pi4 !!!
 ─────────────────────────────────────────────────────
```
and EEPROM is working finally.
```
*** INSTALLING EEPROM UPDATES ***
BOOTLOADER: update required
CURRENT: Mon 15 Jul 12:59:55 UTC 2019 (1563195595)
 LATEST: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
VL805: update required
CURRENT: 00013701
 LATEST: 000137ab
EEPROM updates pending. Please reboot to apply the update.
[  OK  ] rpi-eeprom  | Completed
root@DietPi4:~#
```

well done.

MichaIng, 2019-12-13:
@Joulinar 
Perfect, many thanks for testing 🚀.

MichaIng, 2019-12-20:
### Beta v6.27.1 has been merged: #3270

dezren39, 2019-12-21:
Hi! Are there any plans to use 'releases' in GitHub sometime in the future? I'm interested in keeping up with this project, but there is a _lot_ of activity day-to-day. I'd love if I could follow 'releases only' instead.  😅👍

MichaIng, 2019-12-21:
@dezren39 
No plans to do so. We use the master branch for releases and do not want to keep older versions provided for download 😉. Also the raw code download does not make the system run, one needs to install it correctly and run the incremental patches, which is what `dietpi-update` does. So downloadable releases IMO give a wrong impression about these two things.

But yeah, I can imagine that following this project leads to a mass of notifications about commits and issue conversations. Actually I aimed to use PRs for everything, as well to have code changes, testing and changelog all done outside the dev branch and then merge it at once, which should reduce the amount of notifications/events on the commit-side. But especially for small changes with few code lines I am simply too lazy to do the PR overhead 😅.
The opened issues are again something out of our control of course.

Would be actually nice if it was possible to follow commits to a single branch only (Using master and beta branch then would keep you informed about all releases.), or excluding issue conversations and generally having some more control. Same in my case for [Nextcloud](https://github.com/nextcloud/server) and some other projects which are under HEAVY development, impossible to watch.

Joulinar, 2019-12-21:
@MichaIng 
I was running a quick update check. For me it seems working fine without issues.

[update.txt](https://github.com/MichaIng/DietPi/files/3991541/update.txt)

MichaIng, 2019-12-21:
@Joulinar 
Great, many thanks for testing, also good to see EEPROM and asix ax88179 driver install going through nicely.

Joulinar, 2019-12-24:
@MichaIng 
Not sure where to place it but I notices some strange behavior. I was running Nextcloud for testing purposes and I removed it afterwards. However if I run dietpi-software > uninstall again, I still see MariaDB, Redis and Nextcloud even if it was already removed. 

[image]

Trying to uninstall them again will lead to an error message as the software is not available anymore :confused:

```
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Uninstalling Nextcloud: File sync, sharing and collaboration platform

Ignoring unknown module: dietpi-nextcloud
Run "service lighttpd force-reload" to enable changes
grep: /var/www/nextcloud/config/config.php: No such file or directory
grep: /var/www/nextcloud/config/config.php: No such file or directory
grep: /var/www/nextcloud/config/config.php: No such file or directory
Failed to start mariadb.service: Unit mariadb.service not found.
/DietPi/dietpi/dietpi-software: line 12445: mysql: command not found
/DietPi/dietpi/dietpi-software: line 12449: mysqladmin: command not found
cp: cannot stat '/var/www/nextcloud/.': No such file or directory
rm: cannot remove '/var/www/nextcloud': No such file or directory

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Uninstalling Redis: optional non-sql database store

[ INFO ] DietPi-Software | APT removal for: redis-server redis-tools php*-redis, please wait...
[ INFO ] DietPi-Software | None of the requested packages are currently installed. Aborting...
[  OK  ] DietPi-Software | G_AGP redis-server redis-tools php*-redis

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Uninstalling MariaDB: database

[FAILED] DietPi-Software | systemctl start mariadb
```

MichaIng, 2019-12-24:
@Joulinar 
Strange, checked the code and couldn't find how this could happen. Running some test currently.
Do you have the end of the first uninstall attempt?

Please mark MariaDB manually as uninstalled:
```
G_CONFIG_INJECT 'aSOFTWARE_INSTALL_STATE\[88\]=' 'aSOFTWARE_INSTALL_STATE[88]=0' /DietPi/dietpi/.installed
```
We should probably remove the error handler for MariaDB start when its about to be uninstalled. This is to create a full database backup first, so users cannot accidentally remove important data.

Joulinar, 2019-12-24:
First of all I wish you Mary Christmas @MichaIng and many thanks for your incredible support over the last year. Much appreciated.

Coming back the the challenge 😄  

I now removed LLMP, Redis and Nextcloud one by one again, using dietpi-software uninstall dialog. Even if nothing was done as no package exist, it's gone from the list. MariaDB is still failing as expected. However I created a Bug report, even if it is not needed.

reference code: 26a63a3e-1699-4250-8d78-67b8ba8155c9

Honestly I'm not sure what happen during the real first de-installation. It should have been removed sucessfull as I checked the software catalog and it was gone. I'm sure. Anyway I now set it back to 0 in `/DietPi/dietpi/.installed` as suggested.

MichaIng, 2019-12-29:
@Joulinar 
Since `/DietPi/dietpi/.installed` is on RAMdisk, it might be possible that some unclean shutdown or crash made the install state changes lost. This is one of the reason I want to move away from RAMdisk (#3288) for such state info, as well for the very important `/DietPi/dietpi/.install_state` e.g., which, when lost, would lead to firstrun setup and install steps run+prompt again on boot+login.

Joulinar, 2019-12-29:
@MichaIng 
Could be it was caused by an unclean shutdown as it happen on my test system. And usually I simply unplug power if not needed. :grimacing:

Joulinar, 2019-12-30:
@MichaIng 
I know it's spooky but MariaDB is back to my software catalog. Going to remove it again. 

[image]

MichaIng, 2019-12-31:
@Joulinar 
Indeed strange. Note that you need to do a regular `shutdown now` or `reboot` currently to have any changes to RAMdisk (including install flags) synced to disk. #3288 will allow to skip this by replacing the RAMdisk with a locked disk cache for DietPi files instead of copying and syncing back those to and from RAMdisk.

Joulinar, 2019-12-31:
Honestly I never was taking care on these things. But starting now I will do a clean shutdown before going to deplug power. Btw, does `poweroff` is doing the same in regards to write back the RAMdisk?

MichaIng, 2020-01-01:
@Joulinar 
Jep, `poweroff` and `shutdown now` should do exactly the same.

MichaIng, 2020-01-01:
### v6.27 has been released, many thanks to all contributors and all testers ❤️: #3290

closed 2020-01-01 by MichaIng