MichaIng/DietPi #3258: Docker not working at all - exit code 139

SaturnusDJ, 2019-12-08:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=26
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
9.11

- Kernel version | `uname -a`
Linux RPi_1B 4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019 armv6l GNU/Linux

- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi B (armv6l)

- Power supply used | (EG: 5V 1A RAVpower)
5V thing that always worked.

- SDcard used | (EG: SanDisk ultra)
2GB thing that always worked.

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
Docker

- Was the software title installed freshly or updated/migrated?
Both. Not sure what the version of the old install was.

- Can this issue be replicated on a fresh installation of DietPi?
Yes. Never had this problem. Then after not using this specific RPi for some months, used it again today: problem. Reinstalled DietPi, still problem.

*Problem*
Most Docker containers exit with code 139.  Very difficult to find any errors and logs.
I tried:
- images that work on other systems.
- build those images locally.
- try other very basic images from the internet, such as hello-world.

Journalctl -xe during hello-world:
```
Dec 08 22:30:39 RPi_1B kernel: docker0: port 1(veth74c498a) entered blocking state
Dec 08 22:30:39 RPi_1B kernel: docker0: port 1(veth74c498a) entered disabled state
Dec 08 22:30:39 RPi_1B kernel: device veth74c498a entered promiscuous mode
Dec 08 22:30:41 RPi_1B containerd[441]: time="2019-12-08T22:30:41.350661838Z" level=info msg="shim containerd-shim started" address="/containerd-shim/moby/bdb178ca834a0d2d35856f21a17c972c7b7
0e2e68ef87e1853c2683609967e65/shim.sock" debug=false pid=3782
Dec 08 22:30:43 RPi_1B kernel: eth0: renamed from veth5c866db
Dec 08 22:30:43 RPi_1B kernel: docker0: port 1(veth74c498a) entered blocking state
Dec 08 22:30:43 RPi_1B kernel: docker0: port 1(veth74c498a) entered forwarding state
Dec 08 22:30:45 RPi_1B containerd[441]: time="2019-12-08T22:30:45.450144838Z" level=info msg="shim reaped" id=bdb178ca834a0d2d35856f21a17c972c7b70e2e68ef87e1853c2683609967e65
Dec 08 22:30:45 RPi_1B dockerd[668]: time="2019-12-08T22:30:45.461778838Z" level=info msg="ignoring event" module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
Dec 08 22:30:45 RPi_1B kernel: docker0: port 1(veth74c498a) entered disabled state
Dec 08 22:30:45 RPi_1B kernel: veth5c866db: renamed from eth0
Dec 08 22:30:45 RPi_1B kernel: docker0: port 1(veth74c498a) entered disabled state
Dec 08 22:30:45 RPi_1B kernel: device veth74c498a left promiscuous mode
Dec 08 22:30:45 RPi_1B kernel: docker0: port 1(veth74c498a) entered disabled state
Dec 08 22:30:45 RPi_1B dockerd[668]: time="2019-12-08T22:30:45.987541838Z" level=warning msg="bdb178ca834a0d2d35856f21a17c972c7b70e2e68ef87e1853c2683609967e65 cleanup: failed to unmount IPC:
 umount /mnt/dietpi_userdata/docker-data/containers/bdb178ca834a0d2d35856f21a17c972c7b70e2e68ef87e1853c2683609967e65/mounts/shm, flags: 0x2: no such file or directory"
```
This might be related. Not sure. Timings does match at least.

No fix:
- https://success.docker.com/article/error-oci-runtime-create-failed-docker-1809
- https://code-issue.site/issues/476118705 (?)

MichaIng, 2019-12-09:
@SaturnusDJ 
Many thanks for your report. Can you please paste the following:
```
journalctl -u docker
journalctl -u containerd
```
And there should be log files in `/mnt/dietpi_userdata/docker-data/containers/` per-container as well.
Those should show some more logs compared to `journalctl -xe`.

There was an external issue with Docker on ARMv6: https://github.com/MichaIng/DietPi/issues/3128
It was fixed meanwhile, but I thought that before and then some update broke it again. However this was an issue on binary/arch level, hence the binary call should break immediately and not produce any output like in your case 🤔. Lets see if the logs give some more hints.

SaturnusDJ, 2019-12-09:
Reboot: 14:47.
Then date command.
Then ran `docker run hello-world`.
Then date command.

Output:
```
root@RPi_1B:~# date
Mon  9 Dec 14:48:13 GMT 2019
root@RPi_1B:~# docker run hello-world
dateroot@RPi_1B:~# date
Mon  9 Dec 14:48:30 GMT 2019
```

`journalctl -u docker`
```
-- Logs begin at Mon 2019-12-09 14:47:19 GMT, end at Mon 2019-12-09 14:48:28 GMT. --
Dec 09 14:47:29 RPi_1B systemd[1]: Starting Docker Application Container Engine...
Dec 09 14:47:33 RPi_1B dockerd[426]: time="2019-12-09T14:47:33.147083000Z" level=info msg="libcontainerd: started new docker-containerd process" pid=480
Dec 09 14:47:33 RPi_1B dockerd[426]: time="2019-12-09T14:47:33.152566000Z" level=info msg="parsed scheme: \"unix\"" module=grpc
Dec 09 14:47:33 RPi_1B dockerd[426]: time="2019-12-09T14:47:33.156186000Z" level=info msg="scheme \"unix\" not registered, fallback to default scheme" module=grpc
Dec 09 14:47:33 RPi_1B dockerd[426]: time="2019-12-09T14:47:33.233516000Z" level=info msg="ccResolverWrapper: sending new addresses to cc: [{unix:///var/run/docker/containerd/docker-containerd.sock 0  <nil>}]" module=grpc
Dec 09 14:47:33 RPi_1B dockerd[426]: time="2019-12-09T14:47:33.236665000Z" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
Dec 09 14:47:33 RPi_1B dockerd[426]: time="2019-12-09T14:47:33.237365000Z" level=info msg="pickfirstBalancer: HandleSubConnStateChange: 0x14933ea0, CONNECTING" module=grpc
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=info msg="starting containerd" revision=468a545b9edcd5932818eb9de8e72413e616e86e version=v1.1.2
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=info msg="loading plugin "io.containerd.content.v1.content"..." type=io.containerd.content.v1
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=info msg="loading plugin "io.containerd.snapshotter.v1.btrfs"..." type=io.containerd.snapshotter.v1
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.btrfs" error="path /mnt/dietpi_userdata/docker-data/containerd/daemon/io.containerd.snapshotter.v1.bt
rfs must be a btrfs filesystem to be used with the btrfs snapshotter"
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=info msg="loading plugin "io.containerd.snapshotter.v1.aufs"..." type=io.containerd.snapshotter.v1
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.aufs" error="modprobe aufs failed: "modprobe: FATAL: Module aufs not found in directory /lib/modules/
4.19.66+\n": exit status 1"
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=info msg="loading plugin "io.containerd.snapshotter.v1.native"..." type=io.containerd.snapshotter.v1
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=info msg="loading plugin "io.containerd.snapshotter.v1.overlayfs"..." type=io.containerd.snapshotter.v1
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=info msg="loading plugin "io.containerd.snapshotter.v1.zfs"..." type=io.containerd.snapshotter.v1
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.zfs" error="path /mnt/dietpi_userdata/docker-data/containerd/daemon/io.containerd.snapshotter.v1.zfs 
must be a zfs filesystem to be used with the zfs snapshotter"
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=info msg="loading plugin "io.containerd.metadata.v1.bolt"..." type=io.containerd.metadata.v1
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=warning msg="could not use snapshotter zfs in metadata plugin" error="path /mnt/dietpi_userdata/docker-data/containerd/daemon/io.containerd.snapshotter.v1.zfs must b
e a zfs filesystem to be used with the zfs snapshotter"
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=warning msg="could not use snapshotter btrfs in metadata plugin" error="path /mnt/dietpi_userdata/docker-data/containerd/daemon/io.containerd.snapshotter.v1.btrfs mu
st be a btrfs filesystem to be used with the btrfs snapshotter"
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=warning msg="could not use snapshotter aufs in metadata plugin" error="modprobe aufs failed: "modprobe: FATAL: Module aufs not found in directory /lib/modules/4.19.6
6+\n": exit status 1"
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=info msg="loading plugin "io.containerd.differ.v1.walking"..." type=io.containerd.differ.v1
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=info msg="loading plugin "io.containerd.gc.v1.scheduler"..." type=io.containerd.gc.v1
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=info msg="loading plugin "io.containerd.service.v1.containers-service"..." type=io.containerd.service.v1
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=info msg="loading plugin "io.containerd.service.v1.content-service"..." type=io.containerd.service.v1
Dec 09 14:47:35 RPi_1B dockerd[426]: time="2019-12-09T14:47:35Z" level=info msg="loading plugin "io.containerd.service.v1.diff-service"..." type=io.containerd.service.v1
[... cut ...]
                    "NetworkID": "4a0cef660cf8cac5ea56569c055605a714f34b6dfacaafe478271746b23fd558",
                    "EndpointID": "",
                    "Gateway": "",
                    "IPAddress": "",
                    "IPPrefixLen": 0,
                    "IPv6Gateway": "",
                    "GlobalIPv6Address": "",
                    "GlobalIPv6PrefixLen": 0,
                    "MacAddress": "",
                    "DriverOpts": null
                }
            }
        }
    }
]
```

MichaIng, 2019-12-09:
Hmm actually I can't see any error, just some warnings about expected skipped modules. Also the status shows an exit code but no error. Not sure what is expected as hallo world output, most likely a console print, that is missing in your case?

But did you try an actual docker image/container that is not just for testing purpose?

SaturnusDJ, 2019-12-09:
@MichaIng 
Yes. I found several explanations that differ a lot. Some link it to out of memory, which sound plausible on an RPi 1B, but then the 'OOMKilled' flag should have been true, unless this is processes wrong. Other explanation is that the program finished and exited, but should that not be an exit 0 then? And indeed for hello-world I expect a console output. https://hub.docker.com/_/hello-world/

How about that `level=warning msg="mountpoint for pids not found` error?

What I saw in a test yesterday is that the memory starts to get filled until `free -m` shows 30MB free left. Then again, I remember with full certainty that similar containers as the one I was trying to run initial did run before on this Pi. I did not even force a manual update so it is extra strange that it was not possible anymore. Does DietPi do auto updating? I suspect an auto-update did something. And that commit obviously is also in a fresh install, that also does not work.

MichaIng, 2019-12-09:
Docker 139 seems to be container segmentation fault. Could be memory related or due to wrong architecture or many other reasons...

Did you take care to pull RPi or armv6hf containers/images. A usual issue is that RPi1/Zero user take regular armhf images which are not compatible with the special armv6hf that RPi1 and Zero is. Also 30 MB left memory is much too few, although hello world should never require that much. However they to increase the swapfile size in case.

SaturnusDJ, 2019-12-09:
For the arch I let it auto select. According to Docker pages it should download the right arch automatically. But...hello-world does not seem to have a `v6`.
My own images do have that and they don't work as well (original case).
And even building from source goes wrong (does not even complete), but for that I am using `alpine:3.10` so I assume Docker is smart enough to download the right arch.

MichaIng, 2019-12-09:
@SaturnusDJ 
Yeah I just know from other reports, as well on Docker repo, that indeed Docker does not detect/select well the correct arch for ARMv6 RPis. The problem is the non-standard architecture. Depending on what Docker checks:
- `uname -m` returns `armv6l`. On Debian this would mean `armel` packages, since `armhf` packages are for `armv7l`, hence not compatible with ARMv6. `armel` is wrong as well, note that hence the Debian repo is simply incompatible with ARMv6 RPi models, which was the initial reason for Raspbian, which is basically Debian compiled for this special armv6hf architecture.
- `apt-config dump | grep 'APT::Architecture'` prints `armhf`, which, when taken as ARMv7, which is correct on Debian, is as above wrong for RPi1/Zero as well.

So RPi1/Zero always need special builds, which have neither a clear uname arch print, nor a clear APT architecture. `armv6hf` is only some wording I use. The Docker images as well use different naming to clarify that it is for those RPi models.

Jep the issue you linked fits, only my state that "armhf fits always" is wrong 😅.

SaturnusDJ, 2019-12-10:
Thank you very much for the explanation!
So in the end it had nothing todo with DietPi but it was a lack of knowledge on my side. Sorry for that.

To get Alpine on an RPi 1B, you could run `docker pull arm32v6/alpine:3.10`
In case of Debian it would be a v5... `docker pull arm32v5/debian`

I guess in the case of Portainer they made an ARM image that runs on all versions. https://hub.docker.com/r/portainer/agent/tags

And I just checked for the software I used months ago on that RPi 1B, and...no surprise...it is ARM general. That's why it ran.

So there is arm32v5, arm32v6, arm32v7, arm64v8 and just 'arm'.

Opened this topic for not being able to pull arch + username: https://forums.docker.com/t/how-to-docker-pull-architecture-and-username/85629.

Offtopic:
Do you have any idea how to (Docker) build in order to get the 'general arm' image that apparently works on all (32 bit at least) ARM?
I did multiarch before (https://digilution.io/posts/multiarch-docker-builds/) but this results in images for each different ARM version.

MichaIng, 2019-12-11:
@SaturnusDJ 
The "general arm", if it shall work on all RPi models and all other ARMv7 SBCs as well, shoould be arm32v6, since ARMv7 is backwards compatible. But I would not recommend to now build or pull those everywhere, since you'd loose ARMv7 benefits on those devices. So it makes sense to use arm32v6 only on RPi models with `uname -m` == `armv6l` architecture, else arm32v7.

Aside of that, I do not use Docker myself, hence are not expert on anything that goes beyond the general architecture binary compatibility, especially the Docker-specific naming of those.

SaturnusDJ, 2019-12-11:
Thanks @MichaIng, I will stick to pulling and building specifics then. 👌

closed 2019-12-11 by SaturnusDJ