MichaIng/DietPi #3259: man-db manpages  :  man apt > No manual entry for apt (or ls or pwd etc)

smudgered, 2019-12-09:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version` 
DIETPI_VERSION_CORE=6
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
 buster
- Kernel version | `uname -a`
Linuxx DietPi 4.19.75+ #1270
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi B (arm6l)
- Power supply used | (EG: 5V 1A RAVpower)
5V 2A Generic
- SDcard used | (EG: SanDisk ultra)
Lexar High Speed (Class 10)
#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. Discover man is not installed by default. 
2. Install man-db manpages
3. Try 'man apt' (or pwd, ls, apt-get etc) 

#### Expected behaviour

- man apt (or anything and not just "man man" should display the manpages for apt

#### Actual behaviour

-Response = "No manual entry for apt" Suggested action to "See 'man 7 undocumented'" > "No manual entry fod undocumented in section 7"
Try 'sudo man apt' - same response
Try 'man man' This works! Only this. 

#### Extra details

- ...I tried the usual sources via websearch and explored manpath config to no effect. 
Also uninstalled and reinstalled man-db and manpages afresh and rebooted to no effect.

smudgered, 2019-12-09:
Workaround appears to be 'man - u' 
Prompt is '~$: What manual page do you want?' 
Submit '~$: apt'

MichaIng, 2019-12-09:
Did you probably run dietpi-cleaner once? This also has an option to remove man pages, since most users prefer to see them via web browser from desktop system. In case `apt install --reinstall apt` would recover them for apt.

smudgered, 2019-12-09:
Don't remember running diet-cleaner. Fresh dietpi install. Minimal install, so manpages very useful while installing extra software (like less etc ;)) since everything is from the console command line or ssh ( piped | less where necessary, but not necessary in man of course). 

Reinstalling all the apps, including system ones like apt to get easy manpages command would be necessary . Maybe there is a command I could run to reinstall everything except man-db and manpages? Seems a lot of work! 

But with the 'man - u' workaround which adds an interactive additional step it's no longer a big issue now I know.

Personally for above reasons I would suggest an opt-out option for man-db/manpages in the initial install and that it should default to install man-db/manpages. 

My 2¢ anyway

notDavid, 2019-12-09:
Same issue here, i never ran `dietpi-cleaner`

(DietPi v6.26.3)

smudgered, 2019-12-09:
I don't think the founders of unix/linux ever imagined man-db/manpages would ever be considered 'optional' and so never took into account the use case they might have to be installed later.. ;)

MichaIng, 2019-12-11:
@smudgered 
If you compare our image sizes with Raspbian Lite or Debian/Ubuntu server images, it is obvious that we take much more as optional :wink:. If you go consequently with what the packages are marked as:
- `man-db`: https://packages.debian.org/buster/man-db Priority: important (but not required or essential)
- `manpages`: https://packages.debian.org/buster/manpages Priority: standard
- Compare with `grep`: https://packages.debian.org/buster/grep Priority: essential, which means that other packages may use what `grep` provides without listing it as explicit dependency. Such packages must all exist on any Debian/APT system, else system scripts might call commands which do not exist or need libraries which are not present.
- Packages with priority "required" must still be listed as dependencies by other packages, when they use them, however they are usually expected present to allow proper system maintenance, e.g. repairing the system, e.g. [e2fsprogs](https://packages.debian.org/buster/e2fsprogs) which provides fsck for ext2/3/4 file systems.

We keep all essential and required packages, and those which are used by our own scripts or we think are used by very most users (sudo + nano) + firmware(WiFi)/kernel/bootloader for individual SBCs/devices. Everything else is purged.

I agree that the package man pages should remain on the system, so that when installing `man-db` one is able to access all man pages of all installed packages. I'll recheck and assure that for future images in case.

smudgered, 2019-12-11:
That would be great! 

I like DietPi from what I have seen so far. 

(And appreciate the time you've taken to clarify. In their priority labelling debian maybe didn't take account of special difficulties that might arise in man's case).

MichaIng, 2019-12-11:
@smudgered 
Of course all this is a question of individual needs and Debian package priorities finally a suggestion. Only thing that must be taken seriously is the "essential" packages, which are a must-have to assure other packages do not run into errors, at least unless you know what you're doing and how to interpret and fix potential issues. I am experimenting with purged "hostname" and "sysvinit-utils", but then some system services, cron jobs, APT installs etc produce errors due to missing "hostname" "pidof" command etc. :wink:.
Everything else is moreless optional and a question of personal preferences. And as said, we (DietPi) like to ship things with least possible overhead, least "bloated" and instead offer easy ways to install/configure additions so that all eventual dependencies are pulled automatically (dietpi-software/dietpi-config/dietpi-drive_manager/...). E.g. install Python when there is actually any software installed that requires Python, instead of having this quite large library set pre-installed on all those SBCs with in cases small SDcards etc.

man pages are for very most users not required as well, when there are resources like: https://manpages.debian.org/
However I agree that it should be easily possible to add this feature without having to reinstall every single package, hence the rephrased issue topic :slightly_smiling_face:.

smudgered, 2019-12-11:
It would be great if you can come up with a solution. I will definitely look out for it.

The beauty of man (as an extended help) is that it is both so quick and handy, and also disappears without trace on the screen afterwards.

MichaIng, 2019-12-12:
Okay, indeed we actively removed those documentation and man page files actively. I removed that from our image creation script: https://github.com/MichaIng/DietPi/commit/d198143392960f93e174a43c20f601e024ddba53
This is as well important for legal reasons: /usr/share/doc contains copyright files, which must be kept!

closed 2020-01-01 by MichaIng