MichaIng/DietPi #3260: Website Down?

Rainee4563, 2019-12-10:
Hey just wanted to ask if you guys knew the website was down, trying to download the dietpi image right now.. but for some reason I can't access the site at all just wanted to see if its just me..

MichaIng, 2019-12-10:
@Rainee4563 
Many thanks for your report.

Jep, see: https://github.com/MichaIng/DietPi/issues/3257#issuecomment-563472501

I'm not sure about the issue, since the server runs perfectly fine and it only affects certain accesses:
- I can access from here, as well from mobile network.
- I run into the same access issue when enabling "Opera for Andoid"s data savings feature on mobile phone. Requests are then done through Operas servers, hence different remote host on our server.
- Accessing ssh.dietpi.com here works fine, as well with Opera data savings, but others report an issue with this as well. This means that Cloudflare itself is not the issue.
- I tried to disable fail2ban and unblock everything, but it is not related to our server blocking anything.
- Only left idea then is the VPS provider network.

Joulinar, 2019-12-10:
I found it @MichaIng . 

It's related to IPv4. Usually my entire Network is running IPv4 only and there it is now working. Now I put my RPi directly behind my broadband router and activated IPv6. Surprise, its working again and I can run `dietpi-survey 1`. Disabling IPv6 will reintroduce the issue and `dietpi-survey 1` is failing again.

MichaIng, 2019-12-10:
@Joulinar 
Very strange, I definitely access via IPv4, especially from mobile phone since my provider mobile network is IPv4 only. It is certain IPv4 ranges obviously, that fits to some firewall, but it's not on our control.

Joulinar, 2019-12-10:
@MichaIng 
Not sure if I understood you sever setup correctly. But I assume you are using Cloudflare somehow in front of your WebServer. Because that would explain why I'm receiving the Cloudflare Always Online™ Site, reporting `dietpi.com` is offline (even I'm not using Cloudflare at all right now). 

Checking `nslookup dietpi.com` is giving me `104.27.179.199 and 104.27.178.199` as IPv4 address. 

- Is that the same IPv4 address you are using? 
- Or is that the IPv4 at Cloudflare? 
- And you web server is having an on/different IPv4 as well? 
- Could it be that there is something broken at Cloudflare side as they are not able to connect to your web server correctly?

Sorry for all the questions, but I'm trying to understand how this is going :)

MichaIng, 2019-12-10:
Testing:
```
[php7:error] [pid 8441] [client VV.XX.YY.ZZ:29424] script '/var/www/phpbb/test.php' not found or unable to stat
```
IPv4 client, matching my local network IP, not Cloudflare IP, hence the IP translation to prevent Cloudflare IPs being blocked works as expected.

Double checked that no Cloudflare IP is blocked, although access fails from certain remote IP with all IPs unblocked as well...

Raised backlog limit, lets see if this is related.

Joulinar, 2019-12-10:
for me nothing changed, still getting the error as soon as I have IPv6 disabled. Not sure if you have it already but this is Cloudflare Troubleshooting site for the reported error.

https://support.cloudflare.com/hc/en-us/articles/115003011431#522error

MichaIng, 2019-12-10:
@Joulinar 
> 104.27.179.199

That is the Cloudflare IP the is forwarded to our server. `dietpi.com` resolves here the same and connects successfully.

Since Cloudflare does not relay anything but HTTP(S) port 80+443 connections, we use `ssh.dietpi.com` for SSH/SFTP (dietpi-survey), which resolves to the server directly. Note that if your access it via HTTP on port 80, the internal HTTPS redirection redirects to `dietpi.com`, hence through Cloudflare again. So to check if there is really an issue outside of Cloudflare, `https://ssh.dietpi.com` must be used, which at least produces a warning due to non-matching SSL certificate host.

I recognised some other issus:
- Pinging ssh.dietpi.com or the IP itself directly does not work. AFAIK this was possible before. I see a bunch of incoming ICMP from MVS servers, but my own ping does not reach. dietpi.com can be pingged.
- The other way round, the server cannot ping dietpi.com and by chance I found that it fails to access or ping deb.debian.org and ftp.debian.org APT repos (the hosts themselves). ftp.de.debian.org instead works.

I'll do a server restart tonight as last resort, else open a ticket to have MVS checking what happens to those connections probably before they even reach our VPS.

Arghh, 2019-12-10:
I thought maybe this infos will help you investigate: I get a 522 error from cloudfaire from dietpi.com with pihole turned on. If i disable pihole I have no errors. I can't make out the domain what is causing the problem because I have a pretty aggressive blocklist but it could be olmprodpowerlift-cdn.azureedge.net
Do you guys use azure?

MichaIng, 2019-12-10:
@Arghh 
Many thanks for your info. Actually very strange, since Pi-hole blocks DNS resolving in the first place. `dietpi.com` is resolved to Cloudflares IP, but if it is blocked by Pi-hole, you would see raw 403 or Pi-hole blocking page, but not Cloudflare.
If you see Cloudflare, it means the hostname was resolved by Pi-hole already.

notDavid, 2019-12-10:
@MichaIng dietpi.com is not working for me either, i first noticed because i got a connection error while installing software via `dietpi-software`.

Anyways, i turned on a Vpn, and selected a vpn server in another country: Switzerland, and suddenly i can access dietpi.com again... so yea it seems like some routing/dns issue.

MichaIng, 2019-12-10:
Okay server restart approaching, will add PHP update, hence will take few minutes.

philfleck, 2019-12-11:
just my observation - reachable via vpn over france, but not from austria (showing cloudflare error over budapest)

DutchFlash, 2019-12-11:
26 hours later, Still the error page 522(cloudflare)

MichaIng, 2019-12-11:
Ticket is opened, lets see what MVS comes up with.

Joulinar, 2019-12-12:
@MichaIng 
It seems I'm able to access the website dietpi.com again. However I still have issues connection to ssh.dietpi.com. it was working for a moment but now failing again.

Arghh, 2019-12-12:
> @MichaIng
> It seems I'm able to access the website dietpi.com again. However I still have issues connection to ssh.dietpi.com. it was working for a moment but now failing again.

Can confirm. Dietpi.com is accessible from Germany.

MichaIng, 2019-12-12:
@Rainee4563 @Joulinar @Fourdee @Arghh @philfleck 
Could you guys verify its working now again from all locations? After tinkering a bid and refreshing the SSL certificate to include the testing domain (that bypasses Cloudflare) suddenly the previously failing access worked again, the VPS can access deb.debian.org and github.com domains again, ping/ICMP to VPS works etc. Since everything but the first is not related to the webserver or HTTPS certificate, it must have been coincidence 😄.

I made our SSL protocol + cipher requirements harder to fit new common intermediate standards, hence TLS1.0+1.1 are not supported anymore. Let me know if this breaks any of your clients. However all browsers since 2-3 years ago should support this without issues.

Joulinar, 2019-12-12:
@MichaIng 
I can confirm website dietpi.com as well as ssh.dietpi.com are working fine from Germany

```
root@DietPi:~# dietpi-survey 1
[  OK  ] DietPi-Survey | Connection test: ssh.dietpi.com
[  OK  ] DietPi-Survey | Successfully sent survey data
root@DietPi:~#
```

Rainee4563, 2019-12-13:
Yep I can access it from Cali as well!

philfleck, 2019-12-13:
works again from austria

Fourdee, 2019-12-13:
@MichaIng 

All working here, great work Micha :+1:

Pisgah, 2019-12-13:
It's works here from Taiwan, many thanks

MichaIng, 2019-12-13:
@Fourdee 
Had nothing to do with my work, AFAIK, but great it's solved. Was bad timing during beta phase.

Okay I mark this issue as closed then.

closed 2019-12-13 by MichaIng

johnnyasantoss, 2024-06-08:
I'm having this issue now. Not sure if something was updated but https://dnschecker.org/#A/dietpi.com returns the same ip for all regions (I'm in Brazil);

`curl -vvv -4 dietpi.com` times out after 75s. Is the region or my ip blocked?

Joulinar, 2024-06-08:
We don't do any regional or IP blocking on our web site. Probably this is some lovely with your ISP. At least from central Europe web site is working without issues.

MichaIng, 2024-06-08:
@johnnyasantoss 
Note that this issue was about `ssh.dietpi.com`, which points to our server directly, while you have issues to connect to `dietpi.com`, which connects through Cloudflare.

At which stage does it hang, DNS or actual connection? Does this work and print the IP addresses you see on DNS checker?
```sh
getent hosts dietpi.com
```
There are currently two rare issues about connections to `dietpi.com`:
- Immediate TLS failures, which do not occur always, but in probably ~50% of attempts, from a small number of certain regions/ISPs, including my home: #7022
- Long latencies, currently known only when connection from Europe via DTAG (Deutsche Telekom AG) or its daughter ISPs in oher countries: https://community.cloudflare.com/t/608356

While I am currently affected by the first, next week I'll instead be affected by the 2nd, since I am switching to DTAG ISP 😄.

Since Cloudflare has a firewall as well, I just checked the logs, and found just one blocked access from Brazil, done however with a Go HTTP library in an unusual request.

johnnyasantoss, 2024-06-08:
@MichaIng 
Running `getent hosts dietpi.com`
returned
```logs
2606:4700:20::681a:4f3 dietpi.com
2606:4700:20::681a:5f3 dietpi.com
2606:4700:20::ac43:4565 dietpi.com
```

I was trying to upgrade my dietpi and I don't think it uses Go, right? Managed to update it using a vpn

The issue was that it is taking too long to reply, leading to timeout.
```
time curl -vvv -4 dietpi.com
*   Trying 104.26.5.243:80...
* connect to 104.26.5.243 port 80 failed: Connection timed out
* Failed to connect to dietpi.com port 80: Connection timed out
* Closing connection 0
curl: (28) Failed to connect to dietpi.com port 80: Connection timed out

real    2m12.044s
user    0m0.029s
sys     0m0.101s
```
Using ipv6 it returns network unreachable
```logs
dietpi@raspbolt:~$ curl -vvv -6 dietpi.com
*   Trying 2606:4700:20::681a:5f3:80...
* Immediate connect fail for 2606:4700:20::681a:5f3: Network is unreachable
*   Trying 2606:4700:20::ac43:4565:80...
* Immediate connect fail for 2606:4700:20::ac43:4565: Network is unreachable
*   Trying 2606:4700:20::681a:4f3:80...
* Immediate connect fail for 2606:4700:20::681a:4f3: Network is unreachable
* Closing connection 0
curl: (7) Couldn't connect to server
```

Joulinar, 2024-06-09:
> Using ipv6 it returns network unreachable

I guess your SBC don't have IPv6 configured

MichaIng, 2024-06-09:
> I was trying to upgrade my dietpi and I don't think it uses Go, right?

No, this really is a dedicated little program, written in Go, which seems to be often used for harmful website crawls, so that its user agent is on the Cloudflare WAF block list.

> I guess your SBC don't have IPv6 configured

But IPv4 requests time out as well.

Does it work with `wget`?
```sh
wget --spider https://dietpi.com/
```
And can you visit our website from browser?

Joulinar, 2024-06-09:
> But IPv4 requests time out as well.

Yes, correct, IPv4 has a time out but on IPv6 network is unreachable, indicating an incorrect IPv6 connection. 😉 
But this should be unrelated as our web site is/should be reachable on IPv4, regardless of whether IPv6 has been configured.

MichaIng, 2024-06-09:
I guess both issues are related, and just have different results based on IP protocol version. At least an IPv6 route seems to exist, otherwise the error should be "no route to host". But of course can be easily tested:
```sh
curl -I6 google.com
```

johnnyasantoss, 2024-06-10:
Yeah ipv6 is still poorly configured on this machine. I've disabled it now (was only on to test).

Running wget --spider
```
wget --spider --timeout 3 dietpi.com
Spider mode enabled. Check if remote file exists.
--2024-06-10 11:33:19--  http://dietpi.com/
Resolving dietpi.com (dietpi.com)... 104.26.4.243, 104.26.5.243, 172.67.69.101, ...
Connecting to dietpi.com (dietpi.com)|104.26.4.243|:80... failed: Connection timed out.
Connecting to dietpi.com (dietpi.com)|104.26.5.243|:80... failed: Connection timed out.
Connecting to dietpi.com (dietpi.com)|172.67.69.101|:80... failed: Connection timed out.
Connecting to dietpi.com (dietpi.com)|2606:4700:20::681a:4f3|:80... failed: Cannot assign requested address.
Connecting to dietpi.com (dietpi.com)|2606:4700:20::681a:5f3|:80... failed: Cannot assign requested address.
Connecting to dietpi.com (dietpi.com)|2606:4700:20::ac43:4565|:80... failed: Cannot assign requested address.
Retrying.
...
[same]
```

I thought it was something related to Tailscale (which I have configured on this sbc) but turning it off didn't show any difference.

> And can you visit our website from browser?

On another computer on the same net, yes. On the SBC, no.

MichaIng, 2024-06-10:
But connecting to our server directly works, right?
```sh
curl -I ssh.dietpi.com
```
And what about other sites behind Cloudflare?
```sh
curl -I symfony.com
```

johnnyasantoss, 2024-06-11:
Also, yesterday I tested with Tailscale off and it didn't change a thing

MichaIng, 2024-06-11:
Can you check the route:
```sh
apt install mtr-tiny
mtr dietpi.com
mtr -4 dietpi.com
```
And compare this with what you get on other systems, e.g. on Windows with
```cmd
tracert dietpi.com
tracert -4 dietpi.com
```