MichaIng/DietPi #3261: Tor hotspot does not have internet access

blizarazu, 2019-12-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=26
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
buster . 10.2
- Kernel version | `uname -a`
Linux DietPi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi 3 Model B+ (armv7l)
- Power supply used | (EG: 5V 1A RAVpower)
5V 2.1A
- SDcard used | (EG: SanDisk ultra)
Samsung EVO Plus 32GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
Tor Hotspot
- Was the software title installed freshly or updated/migrated?
Installed freshly
- Can this issue be replicated on a fresh installation of DietPi?
Yes

#### Steps to reproduce
1. Install Tor Hotspot application
2. Connect to the hotspot with another device

#### Expected behaviour
- You should be able to navigate using Tor.

#### Actual behaviour
- The device has no internet connection.

#### Extra details
- If I install the Wifi Hotspot only, there is internet connection through the hotspot. But once you install the Tor Hotspot you can't navigate anymore.

MichaIng, 2019-12-11:
@blizarazu 
Many thanks for your report. Since I didn't use Tor for a while, I need to do some test install. Please keep us updated if you find some solution or additional debug info meanwhile.

blizarazu, 2019-12-11:
I think I found the solution in this tutorial:
https://3os.org/raspberryPi/TOR-Pi/

As the user **Apocalypse1508** says in the comments, the config entries "TransListenAddress" and "DNSListenAddress" seem to be deprecated, so I added the IP to "DNSPort" and "TransPort" and removed the "TransListenAddress" and "DNSListenAddress" entries.

blizarazu, 2019-12-11:
I also have to reload tor after restoring the iptables rules on each boot.

MichaIng, 2019-12-12:
@blizarazu 
Okay, indeed those two settings are deprecated since at least Stretch, were still supported on Stretch but not anymore on Buster (respectively the related/shipped Tor version).
Do you agree to skip file logging and instead keep logs in journal, available via `journalctl -u tor` or `journalctl` to have other probably related syslogs aside? I aim to remove file-based logging for software, where possible, since journal is the default and always present on Debian anyway. Otherwise e.g. logs from the service/systemd unit are in journal, while binary logs are in file, which makes debugging more difficult when one always needs to check both.

closed 2020-01-01 by MichaIng