MichaIng/DietPi #3262: about crontab

hifitime, 2019-12-10:
I want to run a script like: */10 * * * * * bash /var/abc.sh

Which file should I add? /etc/crontab ？ or  crontab -e 

 It is also not lost when dietpi is upgraded

Please help me, thank you

Joulinar, 2019-12-11:
@hifitime 

you could try to copy your script to `/etc/cron.minutely/` . Afterwards run `dietpi-cron` to enable and configure this feature. It allows to run scripts every chosen amount of minutes.

MichaIng, 2019-12-11:
Jep, little addition:
- Files in `/etc/cron.*/` are not allowed to have file endings, hence rename `abc.sh` to `abc` as well: `mv /var/abc.sh /etc/cron.minutely/abc`
- Take care it has the shebang `#!/bin/bash` (or whichever shell works), as those files are not explicitly executed with a specific shell.
- Take care it is executable: `chmod +x /etc/cron.minutely/abc`
- Check if it would be executed: `run-parts --test /etc/cron.minutely/`

Just for reference, as the bid strict rules of `run-parts`, used to execute the files in those dirs, I link the man page: https://manpages.debian.org/run-parts

hifitime, 2019-12-12:
I want to run my script every 10 seconds. What should I do? Is there any good solution?
 thank you

MichaIng, 2019-12-12:
@hifitime 
Cron only allows to run jobs on minutely basis, hence every 10 seconds is not natively possible. In this case I suggest you create a systemd unit that starts your script and do a 10 second loop within this script:
```
while:
do
   <your_code>
   sleep 10
done
```
Create and enable + start the systemd unit:
```
cat << _EOF_ > /etc/systemd/system/<your_service_name>.service
[Unit]
Description=<your_service_name>

[Service]
ExecStart=/path/to/your/script

[Install]
WantedBy=multi-user.target
_EOF_
systemctl daemon-reload
systemctl enable --now <your_service_name>
```

hifitime, 2019-12-12:
Thank you. Finally, which directory should I put my own files in? Like a script and so on... Any suggestions?
I want to be more standard and clear

MichaIng, 2019-12-12:
Basically you can choose freely. E.g. use /mnt/dietpi_userdata/<name>.sh to have it together with other DietPi userdata and settings. Or /opt/<name>.sh for "optional" programs or within your users home dir, although as such systemd unit it runs for all users/system-wide.

hifitime, 2019-12-12:
Thank you very much for your patient answer, which has helped me a lot.

closed 2019-12-12 by hifitime