MichaIng/DietPi #3263: Abnormal start of gmrender does not work normally。。。。Please help

fnsnyc, 2019-12-12:
### Creating a bug report/issue

Service doesn't start correctly. I'm using SofaPlay on my Mac as a client for streamming. It shows DietPi but when I press play ... nothing happens.

#### Required Information
- DietPi version | DietPi v6.26.3
- Distro version | buster
- Kernel version | Linux DietPi 4.19.75-v7l+ #1270
- SBC device | RPi4
- Power supply used | (Ciaudio 5V)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

root@DietPi:~# systemctl status gmrender -l
● gmrender.service - GMediaRender (DietPi)
   Loaded: loaded (/etc/systemd/system/gmrender.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-12-12 17:07:35 -03; 26min ago
  Process: 644 ExecStartPre=/usr/bin/touch /var/log/gmrender.log (code=exited, status=0/SUCCESS)
  Process: 645 ExecStartPre=/bin/chown gmrender:dietpi /var/log/gmrender.log (code=exited, status=0/SUCCESS)
 Main PID: 646 (dash)
    Tasks: 21 (limit: 4915)
   Memory: 156.1M
   CGroup: /system.slice/gmrender.service
           ├─646 /bin/dash -c gmediarender -u "20fc1862-ecff-4773-b5a6-033ef6ab6f8c" -f "DietPi" --gstout-audiodevice=sysdefault --gstout-initial-volume-db=-1 --logfile=/var/log/gmrender.log -I
 "$(sed -n 4p /DietPi/dietpi/.network)"
           └─649 gmediarender -u 20fc1862-ecff-4773-b5a6-033ef6ab6f8c -f DietPi --gstout-audiodevice=sysdefault --gstout-initial-volume-db=-1 --logfile=/var/log/gmrender.log -I 192.168.1.109

Dec 12 17:07:48 DietPi gmediarender[649]: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed
Dec 12 17:07:48 DietPi gmediarender[649]: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (src)' failed
Dec 12 17:07:48 DietPi gmediarender[649]: gst_element_get_static_pad: assertion 'GST_IS_ELEMENT (element)' failed
Dec 12 17:07:48 DietPi gmediarender[649]: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (dest)' failed
Dec 12 17:07:48 DietPi gmediarender[649]: 
                                          Trying to dispose object "mixer", but it still has a parent "(null)".
                                          You need to let the parent manage the object instead of unreffing the object directly.
Dec 12 17:07:48 DietPi gmediarender[649]: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed
Dec 12 17:07:48 DietPi gmediarender[649]: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed
Dec 12 17:07:48 DietPi gmediarender[649]: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (src)' failed
Dec 12 17:07:48 DietPi gmediarender[649]: gst_element_get_static_pad: assertion 'GST_IS_ELEMENT (element)' failed
Dec 12 17:07:48 DietPi dash[646]: Ready for rendering.

#### Extra details

- ...

fnsnyc, 2019-12-12:
systemctl cat gmrender:

# /etc/systemd/system/gmrender.service
[Unit]
Description=GMediaRender (DietPi)
After=network.target dietpi-boot.service

[Service]
User=gmrender
Group=dietpi
ExecStartPre=+/usr/bin/touch /var/log/gmrender.log
ExecStartPre=+/bin/chown gmrender:dietpi /var/log/gmrender.log
ExecStart=/bin/dash -c 'gmediarender -u "20fc1862-ecff-4773-b5a6-033ef6ab6f8c" -f "DietPi" --gstout-audiodevice=sysdefault --gstout-initial-volume-db=-1 --logfile=/var/log/gmrender.log -I "$(sed -n 4p /Di
etPi/dietpi/.network)"'

[Install]
WantedBy=multi-user.target

fnsnyc, 2019-12-12:
sed -n 4p /DietPi/dietpi/.network
192.168.1.109

fnsnyc, 2019-12-13:
Please help.

MichaIng, 2019-12-13:
@fnsnyc 
> gmediarender -u 20fc1862-ecff-4773-b5a6-033ef6ab6f8c -f DietPi --gstout-audiodevice=sysdefault --gstout-initial-volume-db=-1 --logfile=/var/log/gmrender.log -I 192.168.1.109

Ah sorry that log entry already shows a correct IP value. Hmm could you please do the following:
```
systemctl restart gmrender
# Then try to play something
journalctl -u gmrender # Paste the output of this command here
```

fnsnyc, 2019-12-13:
Here it goes Michael

-- Logs begin at Thu 2019-02-14 10:11:58 GMT, end at Fri 2019-12-13 17:46:17 GMT. --
Dec 13 12:02:43 DietPi systemd[1]: Starting GMediaRender (DietPi)...
Dec 13 12:02:43 DietPi systemd[1]: Started GMediaRender (DietPi).
Dec 13 12:03:15 DietPi gmediarender[690]: file gstdtlsagent.c: line 192 (gst_dtls_agent_init): should not be reached
Dec 13 12:03:15 DietPi gmediarender[690]: gst_dtls_agent_set_property: assertion 'self->priv->ssl_context' failed
Dec 13 12:03:15 DietPi gmediarender[690]: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed
Dec 13 12:03:15 DietPi gmediarender[690]: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed
Dec 13 12:03:15 DietPi gmediarender[690]: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (src)' failed
Dec 13 12:03:15 DietPi gmediarender[690]: gst_element_get_static_pad: assertion 'GST_IS_ELEMENT (element)' failed
Dec 13 12:03:15 DietPi gmediarender[690]: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (dest)' failed
Dec 13 12:03:15 DietPi gmediarender[690]: 
                                          Trying to dispose object "mixer", but it still has a parent "(null)".
                                          You need to let the parent manage the object instead of unreffing the object directly.
Dec 13 12:03:15 DietPi gmediarender[690]: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed
Dec 13 12:03:15 DietPi gmediarender[690]: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed
Dec 13 12:03:15 DietPi gmediarender[690]: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (src)' failed
Dec 13 12:03:15 DietPi gmediarender[690]: gst_element_get_static_pad: assertion 'GST_IS_ELEMENT (element)' failed
Dec 13 12:03:16 DietPi dash[687]: Ready for rendering.
Dec 13 17:46:14 DietPi systemd[1]: Stopping GMediaRender (DietPi)...
Dec 13 17:46:14 DietPi systemd[1]: gmrender.service: Main process exited, code=killed, status=15/TERM
Dec 13 17:46:14 DietPi systemd[1]: gmrender.service: Succeeded.
Dec 13 17:46:14 DietPi systemd[1]: Stopped GMediaRender (DietPi).
Dec 13 17:46:14 DietPi systemd[1]: Starting GMediaRender (DietPi)...
Dec 13 17:46:14 DietPi systemd[1]: Started GMediaRender (DietPi).
Dec 13 17:46:16 DietPi gmediarender[4933]: file gstdtlsagent.c: line 192 (gst_dtls_agent_init): should not be reached

Dec 13 17:46:16 DietPi gmediarender[4933]: gst_dtls_agent_set_property: assertion 'self->priv->ssl_context' failed
Dec 13 17:46:17 DietPi gmediarender[4933]: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed
Dec 13 17:46:17 DietPi gmediarender[4933]: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed
Dec 13 17:46:17 DietPi gmediarender[4933]: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (src)' failed
Dec 13 17:46:17 DietPi gmediarender[4933]: gst_element_get_static_pad: assertion 'GST_IS_ELEMENT (element)' failed
Dec 13 17:46:17 DietPi gmediarender[4933]: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed
Dec 13 17:46:17 DietPi gmediarender[4933]: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed
Dec 13 17:46:17 DietPi gmediarender[4933]: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (src)' failed
Dec 13 17:46:17 DietPi gmediarender[4933]: gst_element_get_static_pad: assertion 'GST_IS_ELEMENT (element)' failed
Dec 13 17:46:17 DietPi gmediarender[4933]: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (dest)' failed
Dec 13 17:46:17 DietPi gmediarender[4933]: 
                                           Trying to dispose object "mixer", but it still has a parent "(null)".
                                           You need to let the parent manage the object instead of unreffing the object directly.
Dec 13 17:46:17 DietPi dash[4931]: Ready for rendering.

MichaIng, 2019-12-14:
Hmm I'm thinking if gmrender has issues with active mixer. Did you try to disable the mixer via dietpi-config > Audio options?

fnsnyc, 2019-12-14:
Sorry but I can't find this option to disable the mixer. It don't show any menu under Audio Options.

MichaIng, 2019-12-14:
Ah I meant the equalizer :wink:.

fnsnyc, 2019-12-14:
EQ is Off already.

fnsnyc, 2019-12-14:
Do you know if it's an issue with Rpi4 and I should downgrade to Rpi3 ?

My NanoPi works flawlessly. But I just bought the Allo Digione (SPDIF board) to use with Rpi4. When I test it with RoonBridge, it works fine and plays thru SPDIF. But I need the DLNA working :-(

MichaIng, 2019-12-14:
Hmm just found that gstreamer issue reported here: https://gitlab.freedesktop.org/gstreamer/gst-plugins-bad/issues/811

Log entries fit yours, but not sure if this is the actual reason that makes the service stop or this one:
```
Dec 13 17:46:17 DietPi gmediarender[4933]:
Trying to dispose object "mixer", but it still has a parent "(null)".
You need to let the parent manage the object instead of unreffing the object directly.
```

Also I see that we assign audio device sysdefault via cmd option/argument, while we name the PCM in /etc/asound.conf "default" only....

Have to test myself later.

fnsnyc, 2019-12-15:
Well ... removed gmrender thru dietpi-config and tried to install manually "gmrender-resurrect" following the instructions at https://github.com/hzeller/gmrender-resurrect/blob/master/INSTALL.md

Same thing. At the end, didn't run, same errors.

Got another SD Card and installed Raspbian Buster Lite and then edited /boot/config.txt  to enable DigiOne Allo board. And then followed the instructions on gmrender-resurrect Install. 

Now it's working perfectly under Raspbian. I would prefer to run over DietPi because I love the tiny OS .... will keep the other SD card and wait for a solution if there is any.

MichaIng, 2019-12-15:
Many thanks for testing. On DietPi the compiling went through without error as well?

Also did you enable the dtoverlay on DietPi the same way? Through dietpi-config, it should be offered, leading to the same /DietPi/config.txt entry (note that config.txt is in RAMdisk on DietPi why the RAMdisk location needs to be edited.

And could you paste the following from the Raspbian system: `cat /etc/asound.conf`

fnsnyc, 2019-12-15:
Didn't enable dtoverlay on Dietpi... The only thing that I did on Dietpi was menu >> Audio >> Select DigioneAllo ... that's it. 

I had to edit boot/config.txt on Raspbian because my sound card was not showing when aplay -l was entered...

Here are the results under Raspbian :

sudo nano /etc/asound.conf  .... empty file.

pi@raspberrypi:~ $ systemctl cat gmrender-resurrect
# /etc/systemd/system/gmrender-resurrect.service
[Unit]
Description=gmrender-resurrect service
After=network.target sound.target

[Service]
Environment="UPNP_DEVICE_NAME=Allo-DigiOne"
ExecStartPre=/bin/sh -c "/bin/systemctl set-environment UPNP_UUID=`ip link show | awk '/ether/ {print \"salt:)-\" $2}' | head -1 | md5sum | awk '{print $1

ExecStart=/usr/local/bin/gmediarender -f "$UPNP_DEVICE_NAME" -u "$UPNP_UUID" \
                                      --gstout-audiosink=alsasink --gstout-audiodevice=sysdefault \
                                      --logfile=/tmp/gmediarenderer.log --gstout-initial-volume-db=-10
Restart=always

[Install]
WantedBy=multi-user.target

MichaIng, 2019-12-15:
Okay only difference then is the asound.conf.

Could you paste: `aplay -l`
The second part of our entry is suggested on gmrender install instructions for Raspbian as well, the first part should define sound card + device only, which should be 0,0 which should match the default as well.

fnsnyc, 2019-12-15:
~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndallodigione [snd_allo_digione], device 0: Allo DigiOne HiFi wm8804-spdif-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0

MichaIng, 2019-12-15:
@fnsnyc 
Okay, card 0 device 0 as expected, so all this should be correct on DietPi as will. I'll run some test to check the influence of the network related options we add with our systemd unit, especially since there are some error messages about that in your service log, and the default systemd unit does not contain any of them.

MichaIng, 2019-12-16:
I made some update: https://github.com/MichaIng/DietPi/commit/32c5e492f4f8fa259133c66f9a11092ce164d24c

It potentially solves your issue, if its due to permissions indeed. I observed in some cases, where running a systemd unit as specific group ("Group=dietpi" in our case), supplementary user groups were overridden. Since GMediaRender does not require dietpi group permissions for anything that I can think of, I just removed that, hence it will run as gmrender:gmrender now, with supplementary "audio" and "render" group permissions assured. When you find time, would be great if you could test the following:
```
cat << _EOF_ > /etc/systemd/system/gmrender.service
[Unit]
Description=GMediaRender (DietPi)
Wants=network-online.target
After=network-online.target dietpi-boot.service

[Service]
User=gmrender
ExecStartPre=+/bin/dash -c 'systemctl set-environment ACTIVE_IP=\$(mawk 'NR==4' /DietPi/dietpi/.network)'
ExecStart=gmediarender -u '$(mawk 'NR==5' /DietPi/dietpi/.hw_model)' -f '$(</etc/hostname)' --gstout-audiosink=alsasink --gstout-audiodevice=sysdefault --gstout-initial-volume-db=-1 --logfile=stdout -I "\$ACTIVE_IP"

[Install]
WantedBy=multi-user.target
_EOF_
systemctl daemon-reload
systemctl restart gmrender
```

With this, debugging is easier as well, as all logs are available via `journalctl -u gmrender` without having binary logs split. Especially in this case, file logging is a bad thing since the gstreamer modules as well still log to STDOUT, which makes it difficult to read everything in correct order...

fnsnyc, 2019-12-16:
```
systemctl status gmrender -l
● gmrender.service - GMediaRender (DietPi)
   Loaded: loaded (/etc/systemd/system/gmrender.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2019-12-16 11:10:47 GMT; 4s ago
  Process: 791 ExecStartPre=/bin/dash -c systemctl set-environment ACTIVE_IP=$(mawk NR==4 /DietPi/dietpi/.network) (code=exited, status=0/SUCCESS)
  Process: 794 ExecStart=/usr/bin/gmediarender -u 5a84151a-60e7-4772-b6ed-b2c73e125201 -f DietPi --gstout-audiosink=alsasink --gstout-audiodevice=sysdef
ault --gstout-initial-volume-db=-1 --logfile=stdout -I $ACTIVE_IP (code=exited, status=1/FAILURE)
 Main PID: 794 (code=exited, status=1/FAILURE)

Dec 16 11:10:42 DietPi gmediarender[794]: ERROR [2019-12-16 11:10:42.486423 | upnp] UpnpInit(ip=fe80::dea6:32ff:fe4f:218f, port=49494) Error: UPNP_E_SOC
KET_ERROR (-208). Retrying... (5s)
Dec 16 11:10:43 DietPi gmediarender[794]: ERROR [2019-12-16 11:10:43.489009 | upnp] UpnpInit(ip=fe80::dea6:32ff:fe4f:218f, port=49494) Error: UPNP_E_SOC
KET_ERROR (-208). Retrying... (4s)
Dec 16 11:10:44 DietPi gmediarender[794]: ERROR [2019-12-16 11:10:44.492973 | upnp] UpnpInit(ip=fe80::dea6:32ff:fe4f:218f, port=49494) Error: UPNP_E_SOC
KET_ERROR (-208). Retrying... (3s)
Dec 16 11:10:45 DietPi gmediarender[794]: ERROR [2019-12-16 11:10:45.495584 | upnp] UpnpInit(ip=fe80::dea6:32ff:fe4f:218f, port=49494) Error: UPNP_E_SOC
KET_ERROR (-208). Retrying... (2s)
Dec 16 11:10:46 DietPi gmediarender[794]: ERROR [2019-12-16 11:10:46.498324 | upnp] UpnpInit(ip=fe80::dea6:32ff:fe4f:218f, port=49494) Error: UPNP_E_SOC
KET_ERROR (-208). Retrying... (1s)
Dec 16 11:10:47 DietPi gmediarender[794]: ERROR [2019-12-16 11:10:47.501184 | upnp] UpnpInit(ip=fe80::dea6:32ff:fe4f:218f, port=49494) Error: UPNP_E_SOC
KET_ERROR (-208). Retrying... (0s)
Dec 16 11:10:47 DietPi gmediarender[794]: ERROR [2019-12-16 11:10:47.503769 | upnp] UpnpInit(ip=fe80::dea6:32ff:fe4f:218f, port=49494) Error: UPNP_E_SOC
KET_ERROR (-208). Giving up.
Dec 16 11:10:47 DietPi gmediarender[794]: ERROR [2019-12-16 11:10:47.503825 | main] ERROR: Failed to initialize UPnP device
Dec 16 11:10:47 DietPi systemd[1]: gmrender.service: Main process exited, code=exited, status=1/FAILURE
Dec 16 11:10:47 DietPi systemd[1]: gmrender.service: Failed with result 'exit-code'.

```

MichaIng, 2019-12-16:
@fnsnyc 
At least we have full module logging now:
> UpnpInit(ip=fe80::dea6:32ff:fe4f:218f, port=49494)

LibUPnP tries to bind socket to IPv6 address, while "`mawk NR==4 /DietPi/dietpi/.network`" should be an IPv4 address, as long as there is one attached to the network adapter. I verified this method of `ACTIVE_IP` variable creation on all systems and it works very well, so this should not be an issue.

You could try to disable IPv6 (dietpi-config > Network Options: Adapters). As long as your RPi is behind a NAT/router in local network, IPv6 should be pretty useless overhead anyway.

fnsnyc, 2019-12-16:
Turned IPv6 `off`

Now it plays... but little bit strange because does not show the minutes progress bar, and does not allow me to forward or rewind ...

```
systemctl status gmrender -l
● gmrender.service - GMediaRender (DietPi)
   Loaded: loaded (/etc/systemd/system/gmrender.service; disabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-12-16 12:26:42 GMT; 4min 23s ago
  Process: 708 ExecStartPre=/bin/dash -c systemctl set-environment ACTIVE_IP=$(mawk NR==4 /DietPi/dietpi/.network) (code=exited, status=0/SUCCESS)
 Main PID: 711 (gmediarender)
    Tasks: 19 (limit: 4915)
   Memory: 61.8M
   CGroup: /system.slice/gmrender.service
           └─711 /usr/bin/gmediarender -u 5a84151a-60e7-4772-b6ed-b2c73e125201 -f DietPi --gstout-audiosink=alsasink --gstout-audiodevice=sysdefault --gstout-initial-volume-d
b=-1 --logfile=stdout -I 192.168.1.109

Dec 16 12:31:01 DietPi gmediarender[711]: <CurrentTransportActions val="PLAY,SEEK"></CurrentTransportActions>
Dec 16 12:31:01 DietPi gmediarender[711]: </InstanceID>
Dec 16 12:31:01 DietPi gmediarender[711]: </Event>
Dec 16 12:31:01 DietPi gmediarender[711]: ERROR [2019-12-16 12:31:01.935722 | gstreamer] source: Error: Not Found (Debug: gstsouphttpsrc.c(1426): gst_soup_http_src_parse_stat
us (): /GstPlayBin:play/GstURIDecodeBin:uridecodebin2/GstSoupHTTPSrc:source:
Dec 16 12:31:01 DietPi gmediarender[711]: Not Found (404), URL: http://192.168.1.110:60060/Test.flac, Redirect to: (NULL))
Dec 16 12:31:01 DietPi gmediarender[711]: ERROR [2019-12-16 12:31:01.936191 | gstreamer] source: Error: Internal data stream error. (Debug: gstbasesrc.c(3055): gst_base_src_l
oop (): /GstPlayBin:play/GstURIDecodeBin:uridecodebin2/GstSoupHTTPSrc:source:
Dec 16 12:31:01 DietPi gmediarender[711]: streaming stopped, reason error (-5))
Dec 16 12:31:01 DietPi gmediarender[711]: ERROR [2019-12-16 12:31:01.936383 | gstreamer] typefindelement7: Error: Stream doesn't contain enough data. (Debug: gsttypefindeleme
nt.c(991): gst_type_find_element_chain_do_typefinding (): /GstPlayBin:play/GstURIDecodeBin:uridecodebin2/GstTypeFindElement:typefindelement7:
Dec 16 12:31:01 DietPi gmediarender[711]: Can't typefind stream)
Dec 16 12:31:03 DietPi gmediarender[711]: INFO  [2019-12-16 12:31:03.912179 | control] cmd_obtain_variable: CurrentMute for instance 0
```

fnsnyc, 2019-12-16:
It's not playing thru Audirvana ... it doen't allow me to select the render. It shows but when I select it ... is automatically unselect.

MichaIng, 2019-12-16:
So far so good. Okay so now I see:
> Not Found (404), URL: http://192.168.1.110:60060/Test.flac, Redirect to: (NULL))

Generally about any issues, best to debug is to replicate the issue to force fresh logs, then run `journalctl -u gmrender` and check all logs from matching timestamp til end.

The following additional gstreamer plugins which allow wider range of A/V formats: `G_AGI gstreamer1.0-libav`
Could be related if some standard decoder can play the format but not read it very well to know current playback time, forward or rewind etc.
_________
Btw I just recognised the same error logs your printed above on my VM as well:
```
...
Dec 16 13:28:54 VM-Buster gmediarender[5769]: INFO  [2019-12-16 13:28:54.755466 | connmgr] Registering support for 'application/x-rtcp'
Dec 16 13:28:54 VM-Buster gmediarender[5769]: file gstdtlsagent.c: line 192 (gst_dtls_agent_init): should not be reached
Dec 16 13:28:54 VM-Buster gmediarender[5769]: gst_dtls_agent_set_property: assertion 'self->priv->ssl_context' failed
Dec 16 13:28:54 VM-Buster gmediarender[5769]: INFO  [2019-12-16 13:28:54.841817 | connmgr] Registering support for 'application/x-dtls'
...
Dec 16 13:28:54 VM-Buster gmediarender[5769]: INFO  [2019-12-16 13:28:54.939538 | connmgr] Registering support for 'application/x-subtitle-vtt'
Dec 16 13:28:54 VM-Buster gmediarender[5769]: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed
Dec 16 13:28:54 VM-Buster gmediarender[5769]: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed
Dec 16 13:28:54 VM-Buster gmediarender[5769]: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (src)' failed
Dec 16 13:28:54 VM-Buster gmediarender[5769]: gst_element_get_static_pad: assertion 'GST_IS_ELEMENT (element)' failed
Dec 16 13:28:54 VM-Buster gmediarender[5769]: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (dest)' failed
Dec 16 13:28:54 VM-Buster gmediarender[5769]: Trying to dispose object "mixer", but it still has a parent "(null)".
                                              You need to let the parent manage the object instead of unreffing the object directly.
Dec 16 13:28:54 VM-Buster gmediarender[5769]: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed
Dec 16 13:28:54 VM-Buster gmediarender[5769]: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed
Dec 16 13:28:54 VM-Buster gmediarender[5769]: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (src)' failed
Dec 16 13:28:54 VM-Buster gmediarender[5769]: gst_element_get_static_pad: assertion 'GST_IS_ELEMENT (element)' failed
Dec 16 13:28:54 VM-Buster gmediarender[5769]: INFO  [2019-12-16 13:28:54.961966 | connmgr] Registering support for 'application/vnd.ms-sstr+xml'
...
```
But these do not break start. Since module logs were missing, we were unable to see what really made the process terminate.

fnsnyc, 2019-12-16:
It doen't work on Audirvana. Can I remove `gmrender` thru menu and then install `gmrender-resurrect` manually ?

MichaIng, 2019-12-16:
@fnsnyc 
Jep of course. DietPi is based on standard Raspbian, you can install any software outside of dietpi-software by following usual guides for Debian/Raspbian.

When you do, would be nice if you could test the repo package first (`apt install gmediarender gstreamer1.0-alsa`), before doing a source build, and just use the sysvinit service is ships + default configs. Would be interesting if this fails the same way, as we install this package as well. AFAIK it pulls too few dependencies, at least `gstreamer1.0-plugins-good` seems to be required additionally, `gstreamer1.0-alsa` definitely is.

fnsnyc, 2019-12-16:
Now it's working perfectly! Just installed `gmrender-resurrect`. Tested the src after `make` and before `install`. At the end copied the `gmrender-resurrect.service` to `/etc/systemd/system/gmrender-resurrect.service`.

Thanks so much!

MichaIng, 2019-12-16:
@fnsnyc 
Great. Yeah as I mentioned above, I see the same errors, but they do not break the service. Logging is done to `/tmp/gmediarenderer.log` now, hence again you do not see the (successful/informational) gstreamer module logs via `journalctl`/`systemctl status` that are before, between and after those errors until "Ready for rendering.".

So the question is now if the Debian/Raspbian build on Buster has issues on RPi or if our systemd unit is (still) the issue, e.g. on RPi still some permissions are missing for user "`gmrender`". The service now runs with `root` permissions, which I don't like to ship as default, not because of mistrust against gmrender-resurrect, but as a general security strategy on UNIX systems, no grant any program not more permissions as it requires, which as well protects against accidents, bugs, misuse, etc.

If you're in mood you could test it:
```
# Create/edit user to assure it has all permissions, usually required for media playback
usercmd='useradd -rMU'
getent passwd gmrender &> /dev/null && usercmd='usermod -a'
$usercmd -G audio,render -s $(command -v nologin) gmrender
# Create systemd unit, note it does not override your current one, as of different file name
cat << _EOF_ > /etc/systemd/system/gmrender.service
[Unit]
Description=GMediaRender (DietPi)
Wants=network-online.target
After=network-online.target dietpi-boot.service

[Service]
User=gmrender
ExecStartPre=+/bin/dash -c 'systemctl set-environment ACTIVE_IP=\$(mawk 'NR==4' /DietPi/dietpi/.network)'
ExecStart=$(command -v gmediarender) -u '$(mawk 'NR==5' /DietPi/dietpi/.hw_model)' -f '$(</etc/hostname)' --gstout-audiosink=alsasink --gstout-audiodevice=sysdefault --gstout-initial-volume-db=-1 --logfile=stdout -I "\$ACTIVE_IP"

[Install]
WantedBy=multi-user.target
_EOF_
systemctl daemon-reload
# Stop functional service
systemctl stop gmrender-resurrect
# Start ours
systemctl start gmrender
journalctl -u gmrender
# If it fails again, check if running it as root solves it (highly expected then)
sed -i '/^User=/d' /etc/systemd/system/gmrender.service
systemctl daemon-reload
systemctl restart gmrender
journalctl -u gmrender
```
The run user is the only relevant difference, everything else is only optical, like UUID and name of the server for clients. The IP is only relevant, if one has more then one network adapter, or more then one IP attached to it, like in your previous case the IPv6. I guess IPv6 was also the reason why your previous source build attempt did not work, but we were not able to see it due to split log to file 😉.

Btw, GMediaRender can be used for video files as well, right? Seems to be a rare use case, but as far as I can see possible. Hence it makes sense to add the user to `video` group as well, to allow framebuffer and monitor access, in case, not sure if some GPU decoder features require this group as well or all this is outsourced to new `render` group 🤔.

fnsnyc, 2019-12-17:
You are right. It's not recommended running as `root`. I did your suggestion and is running perfect. I'm going to use `gmrender` for Audio only.
Also  removed the `gmrender-resurrect.service` file from `systemctl`

MichaIng, 2019-12-17:
@fnsync
Without removing the gmrender user from our systemd unit? 
So then it is indeed an issue with the binary shipped by Raspbian Buster. Damn thing. I'll see if I can build own binaries then for ARMv6+7, new version 0.0.8 should be beneficial as well.

MichaIng, 2019-12-18:
Changelog: https://github.com/MichaIng/DietPi/commit/a5b68d17e446cae701ec4b44ea5c9500514ba0ec

But ahh, while the made changes resolve the potential permissions issue, the faulty Raspbian binary is still an issue...

fnsnyc, 2019-12-19:
MichaIng, rebooted my Rpi4 and something happened with the system ... had to reinstall DietPi again. But now having error installing `raspotify` and it's on loop : 

```
[ INFO ] DietPi-Software | APT update, please wait...
Get:1 https://dtcooper.github.io/raspotify raspotify InRelease [1,706 B]
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Err:1 https://dtcooper.github.io/raspotify raspotify InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 09146F2F7953A455
Hit:3 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
W: GPG error: https://dtcooper.github.io/raspotify raspotify InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 09146F2F7953A455
E: The repository 'https://dtcooper.github.io/raspotify raspotify InRelease' is not signed.
[FAILED] DietPi-Software | G_AGUP
```

MichaIng, 2019-12-28:
@fnsnyc 
Could you please test the following:
```
[[ -f '/usr/local/bin/gmediarender' ]] && mv /usr/local/bin/gmediarender /usr/local/bin/gmediarender_bak
wget https://dietpi.com/downloads/binaries/buster/gmediarender_armv6l -O /usr/local/bin/gmediarender
systemctl restart gmrender
```

fnsnyc, 2019-12-28:
Opps .. I guess I did something wrong. I was under `root`

```
Dec 28 14:13:35 DietPi systemd[1]: gmrender.service: Current command vanished from the unit file, execution of the command
 list won't be resumed.
Dec 28 14:13:38 DietPi gmediarender[713]: INFO  [2019-12-28 14:13:38.174431 | main] Exiting.
Dec 28 14:13:38 DietPi systemd[1]: Stopping GMediaRender (DietPi)...
Dec 28 14:13:38 DietPi systemd[1]: gmrender.service: Succeeded.
Dec 28 14:13:38 DietPi systemd[1]: Stopped GMediaRender (DietPi).
Dec 28 14:13:38 DietPi systemd[1]: Starting GMediaRender (DietPi)...
Dec 28 14:13:38 DietPi systemd[1]: Started GMediaRender (DietPi).
Dec 28 14:13:38 DietPi systemd[17822]: gmrender.service: Failed to execute command: Permission denied
Dec 28 14:13:38 DietPi systemd[17822]: gmrender.service: Failed at step EXEC spawning /usr/local/bin/gmediarender: Permiss
ion denied
Dec 28 14:13:38 DietPi systemd[1]: gmrender.service: Main process exited, code=exited, status=203/EXEC
Dec 28 14:13:38 DietPi systemd[1]: gmrender.service: Failed with result 'exit-code'.

```

MichaIng, 2019-12-28:
@fnsnyc 
Sorry, I forgot the execute permissions: `chmod +x /usr/local/bin/gmediarender`

fnsnyc, 2019-12-28:
Yeap, now working flawlessly.

MichaIng, 2019-12-28:
Many thanks for testing. Great, I'm currently compiling binaries for all architectures and all distro versions, luckily this is going pretty fast 😃.

MichaIng, 2019-12-29:
Issue resolved with: f70a008faa48687977a62ec4f8644610872d6500
Changelog: https://github.com/MichaIng/DietPi/commit/3f95e2e294bcd29d53ebc677aa286fd31aff27a1

MichaIng, 2019-12-30:
@fnsnyc 
Last question: There are two icons the binary expects/the build installs:
- `/usr/local/share/gmediarender/grender-128x128.png`
- `/usr/local/share/gmediarender/grender-64x64.png`

On startup it complains if those are missing, although just as a warning. Since I have not much experience with GMediaRender, do you know where those are used and if its bad when missing?

fnsnyc, 2019-12-31:
I guess is nothing major, those icons is shown on devices like TVs... when listing renderers in your network.

MichaIng, 2019-12-31:
Okay indeed nothing urgent then, although to not confuse users with error messages I'll add those files.

hifitime, 2020-01-01:
It doesn't matter. At least I don't see any impact

closed 2020-01-01 by MichaIng