MichaIng/DietPi #3264: DietPi-Software | Allo Web Gui - Upgrade from 6.18 to 6.26 => "Login failed"

Heroldgray, 2019-12-15:
- Flashed Sparky SBC ALLO Web Gui image  downloaded from: https://dietpi.com/downloads/images/DietPi_SparkySBC-ARMv7-Stretch_AlloGUI.7z
- Upgrade process (6.18 > 6.26)
- Result: "Login failed" in ALLO Web GUI login page.

#### Extra details
Using SSH console (dietpi-config etc ..) under 6.26 is OK as a workaround but, sadly, Web GUI unusable. Thank's for your work.

MichaIng, 2019-12-15:
@Heroldgray 
Many thanks for your report. I'll have a look to get this fixed. Since the Allo web GUI was worked on by @Fourdee I cannot guarantee to be able to fix it for v6.27 already, as long as it needs to be fixed in its own code.

MichaIng, 2019-12-17:
Issues solved with: https://github.com/MichaIng/DietPi/commit/61ccd94abb77e06fe18689081dd1d061f79a89a6

To solve now:
```
chown -R www-data:www-data /var/www/allo
chmod -R 770 /var/www/allo
```

Heroldgray, 2019-12-17:
> To solve now:
> chown -R www-data:www-data /var/www/allo
> chmod -R 770 /var/www/allo

Thank's @MichaIng !
I've tried your quick fix (chown / chmod) under 6.26.3 but unfortunately still "login failed" on web GUI after reboot.

- Some logs to monitor ?
- Would you advise me to reinstall Allo Gui (160) ?

MichaIng, 2019-12-17:
@Heroldgray 
Okay, so it was not just me. Please try:
```
chpasswd <<< 'allo:allo'
```

The user is definitely created with this password, not sure if the web GUI changes something about that somehow, when failing 🤔.

Heroldgray, 2019-12-17:
> Please try:
> chpasswd <<< 'allo:allo'

Bingo !
Thank's @MichaIng

MichaIng, 2019-12-18:
@Heroldgray 
Great, thanks for testing!

closed 2020-01-01 by MichaIng