MichaIng/DietPi #3265: shutdown -r +5 - restart is immediate instead of 5 min delay

bbsixzz, 2019-12-15:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2019-12-15:
@bbsixzz 
Many thanks for your report.

For time delay on shutdown and reboot you need to enable systemd-logind:
```
systemctl unmask systemd-logind
G_AGI libpam-systemd
```

bbsixzz, 2019-12-15:
G_AGI dbus did the trick

Is the lack of functionality a deliberate choice, or is this something you plan on adding later?

MichaIng, 2019-12-16:
@bbsixzz 
It was a deliberate choice indeed. But we need to collect + communicate these cases better, I think, especially since `shutdown` command is present by default, does not give a clear error message or info about `dbus` need for time-based shutdown or reboot schedules, e.g. when running `shutdown --help`.