MichaIng/DietPi #3267: Transmission uses sonarr favicon in Safari 13.0.4 (macOS 10.15.2)

bbsixzz, 2019-12-17:
Creating a bug report/issue

Required Information

DietPi version |
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=26
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

Distro version | buster 10.2

Kernel version | 
Linux DietPi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux

SBC device | RPi 4 Model B (armv7l)

Power supply used | 3A Official

SDcard used | SanDisk ultra

Extra details

sonarr:
[image]

Transmission:
[image]

MichaIng, 2019-12-17:
@bbsixzz
Many thanks for your report.

I'm petty sure it's sure to Safari browser caching. Like:
- You access Sonarr
- Safari stores favicon in browser cache on domain/IP-basis
- You access Transmission on same domain/IP
- Safari loads Sonarr favicon from browser cache

You can try to clear browser cache via command+F5 (or whatever the CTRL button on Apple keyboards is :wink:) when having Transmission page opened.
It should then show the correct favicon, if Transmission even has one :thinking:.

bbsixzz, 2019-12-17:
Well that doesn't work.

I cleared cache by closing Safari and deleting everything here:

`~/Library/Safari/Favicon Cache/`

This works but only resets the same sonarr favicon.

Transmission has the favicon here:

```
root@DietPi:/usr/share/transmission/web/images
-rw-r--r-- 1 root root 1.4K Jan  1  2019 favicon.ico
-rw-r--r-- 1 root root  693 Jan  1  2019 favicon.png
-rw-r--r-- 1 root root  21K Jan  1  2019 webclip-icon.png
```

sonarr has this in the html code:
```
<link rel="apple-touch-icon" href="/Content/Images/touch/57.png?h=cYW4iAoiF8Umm1DHSzinJA"/>
<link rel="apple-touch-icon" sizes="72x72" href="/Content/Images/touch/72.png?h=8iKZ1Ip3YAXGoD7q/n82Uw"/>
<link rel="apple-touch-icon" sizes="114x114" href="/Content/Images/touch/114.png?h=9tDnXMrTFbU2OW5+X8wTkw"/>
<link rel="apple-touch-icon" sizes="144x144" href="/Content/Images/touch/144.png?h=UFS9RLRUFxgLXaIGUYrQEg"/>
<link rel="mask-icon" href="/Content/Images/safari/logo.svg?h=DncijoBjtiMYRYcHOs1JQQ" color="#35c5f4">
<link rel="icon" type="image/ico" href="/Content/Images/favicon.ico?h=hnQH4IZszkSOw1Uy6Oe9Aw"/>
```

transmission has this in the html code:
```
<link href="./images/favicon.ico" rel="icon" />
<link href="./images/favicon.png" rel="shortcut icon" />
<link rel="apple-touch-icon" href="./images/webclip-icon.png"/>
```

It doesn't matter if I load transmission first, sonarr always "wins".

Either way, shouldn't each have their separate icon even on the same domain?

MichaIng, 2019-12-17:
@bbsixzz 
> Either way, shouldn't each have their separate icon even on the same domain?

They should of course. If you clear cache completely and then open Transmission only, it still shows the Sonarr favicon? That is very strange indeed as Sonarr runs on a totally different port and should not be requests at all. Also they have their code in totally dedicated directories... Did you test this with other browsers?

Not sure if on apple there is some special mechanism, related to these "`apple-touch-icon`" definitions. But is still strange since Safari should not access the port where Sonarr runs on, whatever these definitions then, when called, override, obviously IP/domain-wide.... You should definitely report this to Sonarr devs, as this is IMO quite critical, thinking about access permissions etc. But the "bug" is on Safari-side then, when it for whatever reason accesses to an additional dedicated webserver (Sonarr) to pull resources for another webserver (Transmission). Interesting what so widely beloved Apple comes around with...

bbsixzz, 2019-12-17:
Well that wasn't much help.

I did a test on iOS 13.3 and it seems there is something fishy indeed with Transmission:

[image]

You can see Transmission has no favicon while sonarr does.

Maybe someone with web expertise can figure this out, I'm out of my depth.

bbsixzz, 2019-12-18:
I also found these errors in console:

```
[Error] Failed to load resource: the server responded with a status of 409 (Conflict) (rpc, line 0) http://10.0.1.2:9091/transmission/rpc

[Error] Failed to load resource: the server responded with a status of 404 (Not Found) (jquery.ui-contextmenu.min.js.map, line 0) http://10.0.1.2:9091/transmission/web/javascript/jquery/jquery.ui-contextmenu.min.js.map
```

MichaIng, 2019-12-18:
@bbsixzz 
Okay, so:
- When you load Sonarr first (and only then!), Sonarr favicon is loaded on Transmission as well, overriding the Transmission favicon, permanently until clearing cache.
- I was looking wrong yesterday, since Transmission and Sonarr seem to have the favicon both implemented the same classic way: `<link href="path/to/favicon.ico" rel="icon" />`
  - Sonarr does not have it stored in its webroot as well, does it? This this is what would be used domain-wide, AFAIK.
- And indeed Safari caches the Sonarr favicon on IP/domain basis, obviously not respecting different ports, which is simply a bug, that should be reported to Apple/Safari devs.

bbsixzz, 2019-12-18:
First of all, in typical Apple fashion [Creating Pinned Tab Icons](https://developer.apple.com/library/archive/documentation/AppleApplications/Reference/SafariWebContent/pinnedTabs/pinnedTabs.html).

These are not your grandfather's favicons from 1999 :)

It seems if Safari can't find any, it will just use the same for the whole domain if one is provided.

That's retarded.

Here's a dirty workaround, empty this folder and lock it:

`~/Library/Safari/Template Icons`

[image]

The result:

[image]

Maybe I'll report this to Apple, but frankly I couldn't care less at this point.

I'm sure they're not as helpful as the DietPi community and @MichaIng :)

MichaIng, 2019-12-19:
@bbsixzz 
Yeah, the issue is quite minor, but pinned tabs are kinda useless when the icons do not match 😄. Great to have a workaround for now, since at least with DietPi multi-webserver systems are quite an intended use case.
Would be interesting to know how quickly Apple devs are reacting to user reports, I only know it from Microsoft. My wife has a MacBook, probably I can use her Apple ID for a report.

closed 2021-02-14 by MichaIng