MichaIng/DietPi #3271: rTorrent missing DarkBetter theme

bbsixzz, 2019-12-19:
Creating a bug report/issue

Required Information

DietPi version |
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=26
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

Distro version | buster 10.2

Kernel version | 
Linux DietPi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux

SBC device | RPi 4 Model B (armv7l)

Power supply used | 3A Official

SDcard used | SanDisk ultra

Extra details

This folder is empty:

`/var/www/rutorrent/plugins/theme/themes/DarkBetter`

Should contain these files [DarkBetter](https://github.com/chocolatkey/DarkBetter)

I imported them manually, shame it's not included by default the theme is great.

MichaIng, 2019-12-19:
@bbsixzz 
Many thanks for your report.

DarkBetter is invoked as repo link on the rutorrent project: https://github.com/Novik/ruTorrent/tree/master/plugins/theme/themes
Since we download the archive instead of using Git clone, this is missing. I'll add code to have it loaded manually and moved in place for dietpi-software installs.

MichaIng, 2019-12-20:
Fixed with: https://github.com/MichaIng/DietPi/commit/dcb2e90b94d4a2a6fae0d035699a3163cb163959
Changelog: https://github.com/MichaIng/DietPi/commit/6f2e51d70ea10018f0d13a2ad6d7e2a0760453c5

closed 2020-01-01 by MichaIng